espressif/arduino-esp32 #588: The connected () function in wificlient.cpp does not work properly

codywon, 2017-08-20:
I use the tcpclient connection server and use connected () to check if the TCP connection is disconnected. I want to reconnect when the TCP connection is broken. Actually, when the server is disconnected, the value returned by connected () is always 1

lonerzzz, 2017-08-21:
Could you provide a code example to better explain the exact situation that you are encountering? Is this a situation of one connection being down while another is still up with the result being that TCP still indicates that it is connected? Also, does it help to you check your Wifi connection status for your use case?

codywon, 2017-08-22:
Thanks! The problem has been solved. After opening the keeptcpalive function, it will work properly!

nbs2017, 2017-08-23:
Pl z could any body help us ...whenever I try to connect WiFi after uploading sketch, it get connected but when AP disconnected by chance ESP32 never recover the connection until Hard Reset!!
Or If AP On after ESP32  WiFi, It does not get AP and its showing like in searching Mode (..................)

lonerzzz, 2017-08-25:
@nbs2017 It is not a good idea to append to an existing issue unless the symptoms seem very much the same. Please raise a separate issue for your problem and provide more detail with some code.

lonerzzz, 2017-08-25:
Closing since the original issue was resolved.

closed 2017-08-25 by lonerzzz