espressif/arduino-esp32 #593: Webserver - How respond to HTTP POST request?

DeepSheth01, 2017-08-24:
I want to make a web server on my ESP32 that can host a HTML form and can submit data using POST method. Can anybody give me any idea how to do that with some sample code?

copercini, 2017-08-24:
Like this?
https://github.com/bbx10/WebServer_tng/tree/master/examples

DeepSheth01, 2017-08-25:
Yes thanks! Got it working with [WebServer_tng](https://github.com/bbx10/WebServer_tng/tree/master/examples) library, but I guess there is no way of doing the same with official Webserver library.

closed 2017-08-25 by copercini

bot1131357, 2018-11-23:
Not sure if anyone is still looking, but POST data can be retrieved with the String arg(String name) method as below:
```
void handlePost() {
	Serial.print("Data: ");
	Serial.println(server->arg("myData1"));
	Serial.print("Data: ");
	Serial.println(server->arg("myData2"));
}
```
Hope this helps.

pcbtborges, 2019-10-15:
bot1131357, what library are you talking about?
Thanks

bot1131357, 2019-10-20:
Just the WebServer from arduino-esp32. Here's a [link](https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples) to the examples.