espressif/arduino-esp32 #598: [ QUESTION ] Is there a way to send custom data using SmartConfig

Raienryu97, 2017-08-30:
I'd like to know if there is a way to modify the existing smart config to not only receive just network configuration using the Mobile App, but if possible, even other small data too. Like adding maybe two more text fields or so. 

Thanks

Raienryu97, 2017-08-30:
Ah, okay, thanks. I wanted to try out connection less mechanism, heard that smart config does that. So maybe I'll try looking into UDP , or is there any better suggestion ? 
I just want to sniff the packets in the air without actually connecting and get some data from that.

me-no-dev, 2017-08-30:
Surely there are hacky things that you can find with a simple search in github :)
You can also start looking into this also: https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/esp_wifi.h#L576-L645

Raienryu97, 2017-08-30:
I am just now learning about the promisc mode. Seems to be very interesting but kind of complicated to get data. As of now I can get packet length,RSSI,etc.  I will go ahead and spend some time on this.
Will keep this issue open for a short while more in case I run into any trouble :)
Thank you very much for the pointer.

closed 2017-09-01 by Raienryu97