espressif/arduino-esp32 #599: WiFi.h conflicts with official Arduino WiFi lib

Testato, 2017-08-30:
```
Multiple libraries were found for "WiFi.h"
 Used: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\espressif\esp32\libraries\WiFi
 Not used: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.8.4\libraries\WiFi
exit status 1
Error compiling for board ESP32 Dev Module.
```

lonerzzz, 2017-08-30:
@Testato Can you provide more of the error output. What you are providing is not an error, it is just telling you which file was chosen when multiple options were present.

Testato, 2017-08-30:
but the Wi.Fi.h not used is the Official Arduino ones, so you always have it on your PC.
It block the compilation, i temporary deleted the Arduino wifi lib.

lonerzzz, 2017-08-30:
I am not sure what you mean about the WiFi.h not being used. Please clarify. We will need more information from the error output that you are seeing in order to help.

copercini, 2017-08-30:
Same as https://github.com/espressif/arduino-esp32/issues/20

copercini, 2017-08-30:
> Multiple libraries were found for "WiFi.h"
>  Used: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\espressif\esp32\libraries\WiFi
>  Not used: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.8.4\libraries\WiFi

@Testato the used library is the correct one for ESP32.

as @lonerzzz say, it's something else causing the error, so please provide more output logs

Testato, 2017-08-30:
i tried to recreate the problem but it now work.
I do not know what is happened, but the message that i attached stop the compilation, i cannot go ahead.
If you want can close this issue, or can leave opened for some time, i it reappear i will post the Verbose consolle output

copercini, 2017-08-30:
I also get this problem rarely ( maybe 1 every 500 compiles), but it is more likely that it is something from arduino IDE than ESP32 core

closed 2017-08-30 by copercini

pandadom, 2018-05-05:
i had the same problem, but solved it by not including #include <Arduino.h> in the sketch

aspiringguru, 2018-07-06:
I found this while experiencing similar problem.

my code

#include <NTPClient.h>
#include <WiFi.h>
#include <WiFiUdp.h>

#define NTP_OFFSET  -3  * 60 * 60 // In seconds
#define NTP_INTERVAL 60 * 1000    // In miliseconds
#define NTP_ADDRESS  "europe.pool.ntp.org"

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, NTP_ADDRESS, NTP_OFFSET, NTP_INTERVAL);

void setup()
{
  Serial.begin(115200);
  timeClient.begin();
}

my error log

uild options changed, rebuilding all
Archiving built core (caching) in: C:\Users\username\AppData\Local\Temp\arduino_cache_169483\core\core_esp32_esp32_esp32_PartitionScheme_default,FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_1d403d56f8c5bdcedfc8df0e02585b69.a
C:\Users\username\AppData\Local\Temp\arduino_build_395865/arduino.ar(main.cpp.o):(.literal._Z8loopTaskPv+0x0): undefined reference to `loop()'

C:\Users\username\AppData\Local\Temp\arduino_build_395865/arduino.ar(main.cpp.o): In function `loopTask(void*)':

C:\Users\username\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.0-rc2\cores\esp32/main.cpp:15: undefined reference to `loop()'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\username\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.0-rc2\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
Error compiling for board ESP32 Dev Module.
Sketch uses 216808 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13768 bytes (4%) of dynamic memory, leaving 313912 bytes for local variables. Maximum is 327680 bytes.
Archiving built core (caching) in: C:\Users\username\AppData\Local\Temp\arduino_cache_169483\core\core_esp32_esp32_esp32_PartitionScheme_default,FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_1d403d56f8c5bdcedfc8df0e02585b69.a

nb: I had to install library NTPClient for above 

by comparison, this sketch below (also using Wifi.h) does not experience any problems during loading.
newb to arduino, actually confused by this. plus lost too much time 

#include <SPI.h>
#include <WiFi.h>

char ssid[] = "foo";      // your network SSID (name)
char pass[] = "bar";         // your network password
int keyIndex = 0;                 // your network key Index number (needed only for WEP)

int status = WL_IDLE_STATUS;

WiFiServer server(80);

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // attempt to connect to Wifi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, pass);

    // wait 3 seconds for connection:
    delay(3000);
  }
  server.begin();
  // you're connected now, so print out the status:
  printWifiStatus();
}

void loop() {
  // listen for incoming clients
  Serial.println("inside loop");
  delay(500);
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}

katrinmate, 2021-02-22:
Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

i had this error, but i have installed the esp32 kit