espressif/arduino-esp32 #602: Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.

vyruz1986, 2017-08-31:
### Hardware:
Board:							Wemos Lolin32
Core Installation/update date:	        on commit a7108969a3b638da250374b60cd483fd05d3c2e9 from master
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						921600

### Description:
Hi,
I've been busy over the past days to port an existing project I did on an Arduino Pro Mini to an ESP32.
I'm using the wemos Lolin32 board.

I've been getting Guru meditation errors and can't track down what is causing them.

I get the following error in the serial monitor:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8
Ready!
Started at: 208Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40086bba  PS      : 0x00060933  A0      : 0x80082416  A1      : 0x3ffccf10  
A2      : 0x3ffc6d30  A3      : 0x00000010  A4      : 0x3ffc8118  A5      : 0x000003ff  
A6      : 0x7ff00000  A7      : 0x40590000  A8      : 0x0000007c  A9      : 0xfffc81e8  
A10     : 0x00000001  A11     : 0x3ffc8118  A12     : 0x0000007c  A13     : 0x3ffc6d5c  
A14     : 0x00000001  A15     : 0x3ffc812b  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfffc81e8  LBEG    : 0x400d1178  LEND    : 0x400d1188  LCOUNT  : 0x00000000  

Backtrace: 0x40086bba:0x3ffccf10 0x40082413:0x3ffccf30 0x400826f9:0x3ffccf50 0x4000beaf:0x3ffccf70 0x400d2785:0x3ffccf90 0x400d27d0:0x3ffccfb0 0x400d27f1:0x3ffccfd0 0x400d287d:0x3ffccff0 0x400d289e:0x3ffcd010 0x400d0fab:0x3ffcd030 0x400d0c3b:0x3ffcd060 0x400f211d:0x3ffcd080

Rebooting...
```

When I translate this with EspExceptionDecoder I get the following stack trace:
```
0x40086bba: is_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap.c line 91
:  (inlined by) multi_heap_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap.c line 305
0x400d1178: LightsControllerClass::HandleStartSequence() at C:\Users\Alex\AppData\Local\Temp\arduino_build_314089\sketch/LightsController.cpp line 146
0x400d1188: LightsControllerClass::HandleStartSequence() at C:\Users\Alex\AppData\Local\Temp\arduino_build_314089\sketch/LightsController.cpp line 153
0x40086bba: is_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap.c line 91
:  (inlined by) multi_heap_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap.c line 305
0x40082413: heap_caps_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c line 123
0x400826f9: _malloc_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c line 28
0x400d2785: String::changeBuffer(unsigned int) at D:\OneDrive\Projects\Arduino\hardware\espressif\esp32\cores\esp32/WString.cpp line 739 (discriminator 2)
0x400d27d0: String::reserve(unsigned int) at D:\OneDrive\Projects\Arduino\hardware\espressif\esp32\cores\esp32/WString.cpp line 739 (discriminator 2)
0x400d27f1: String::copy(char const*, unsigned int) at D:\OneDrive\Projects\Arduino\hardware\espressif\esp32\cores\esp32/WString.cpp line 739 (discriminator 2)
0x400d287d: String::operator=(String const&) at D:\OneDrive\Projects\Arduino\hardware\espressif\esp32\cores\esp32/WString.cpp line 739 (discriminator 2)
0x400d289e: String::String(String const&) at D:\OneDrive\Projects\Arduino\hardware\espressif\esp32\cores\esp32/WString.cpp line 739 (discriminator 2)
0x400d0fab: LCDControllerClass::Main() at C:\Users\Alex\AppData\Local\Temp\arduino_build_314089\sketch/LCDController.cpp line 77 (discriminator 2)
0x400d0c3b: loop() at D:\OneDrive\Projects\FlyballETS-Software/FlyballETS-Software.ino line 213
0x400f211d: loopTask(void*) at D:\OneDrive\Projects\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 18 (discriminator 1)
```

From what I read here it seems a regular string declaration, more specifically this one at [LCDController.cpp:103](https://github.com/vyruz1986/FlyballETS-Software/blob/065ae4a017850418ed70108512f368284a724df2/LCDController.cpp#L103)

Now this is where my knowledge starts to fade, I have no idea how to debug this problem which seems to be coming from the ESP32 arduino core.

### Sketch:
The code that I currently upload to the ESP32 is this code:
https://github.com/vyruz1986/FlyballETS-Software/tree/065ae4a017850418ed70108512f368284a724df2

jeffeb3, 2017-10-22:
I had a similar error message (which is why google brought me here). I was spawning a task, and I didn't give it a large enough stack size. Since you're stuck in a malloc, that might be a place to look. If this is just in the loop, you should know that they implemented it as a task, so it has a max stack size of 8192 (https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp).

If that is your problem, you could put your function on a new task, with a bigger stack size, or move some things to the heap (or just generally do things that require less stack).

If that's not your problem, then can you blame me? I have no idea what I'm doing here either! But I'm having fun.

eighthree, 2018-01-14:
Getting a similar exception, here's my serial monitor output:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d2afc  PS      : 0x00060e30  A0      : 0x800d2ba4  A1      : 0x3ffd1c00  
A2      : 0xfefefefe  A3      : 0x3ffdf73c  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d2980  A9      : 0x3ffd1bc0  
A10     : 0x3ffdf73c  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfefefefe  LBEG    : 0x4000142d  LEND    : 0x4000143a  LCOUNT  : 0xfffffffc  

Backtrace: 0x400d2afc:0x3ffd1c00 0x400d2ba1:0x3ffd1c20 0x400d312c:0x3ffd1c70 0x400d2782:0x3ffd1c90 0x400d23d4:0x3ffd1cb0 0x4012cfb0:0x3ffd1cf0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c
```

```
Decoding 6 results
0x400d2afc: AdafruitIO_Group::call(AdafruitIO_Data*) at C:\Users\eight\Documents\Arduino\libraries\Adafruit_IO_Arduino\src/AdafruitIO_Group.cpp line 320
0x400d2ba1: AdafruitIO_Group::subCallback(char*, unsigned short) at C:\Users\eight\Documents\Arduino\libraries\Adafruit_IO_Arduino\src/AdafruitIO_Group.cpp line 320
0x400d312c: Adafruit_MQTT::processPackets(short) at C:\Users\eight\Documents\Arduino\libraries\Adafruit_MQTT_Library/Adafruit_MQTT.cpp line 827
0x400d2782: AdafruitIO::run(unsigned short) at C:\Users\eight\Documents\Arduino\libraries\Adafruit_IO_Arduino\src/AdafruitIO.cpp line 164
0x400d23d4: loop() at C:\Users\eight\Documents\Arduino\sketch_jan13a/sketch_jan13a.ino line 80
0x4012cfb0: loopTask(void*) at C:\Users\eight\Documents\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 18 (discriminator 1)

```

stickbreaker, 2018-01-14:
@eighthree It looks to me that your Adafruit_MQTT call back address is invalid. The Function Address you passed when you registered your subscribe function is not correct.  Verify you passed the correct function.

This is just a guess because you did not post any code.

Chuck.

dpharris, 2018-01-14:
Quick comment: the passed value is simply the name of the function.  No
need to use &.

borolaci, 2018-02-01:
I had the same issue with win7, arduino 1.8.2.
My final solution was to entirely uninstall arduino and then do the whole installation procedure again (install arduino 1.8.5 actually, install python 2.7 and git as shown ín the installation instructions).
After this searched for esp32 ön board manager,installed and then the problematic (unchanged) code successfully compiled and run.
Hope this helps.

closed 2018-04-19 by everslick