espressif/arduino-esp32 #603: Ethernet + Wifi

SuperFra, 2017-09-01:
I have an EVB-ESP32 Olimex, the wifi interface is connected to a wifi network and the eth interface is connected to another network by means of a switch. The 2 network are insulated, I don't want no routing and not even bridging  between them. For example if I need to do a request to a website I want to address the request to the wifi interface, on other hand if I want to do a request to an internal server on my lan I need to use the eth interface.  How I can specify for a request the interface I want to use? 
TIA

me-no-dev, 2017-09-01:
if both are not in the same subnet, the packet will go out from the interface that can reach the target. Since your server is local to your ETH, it will go out from there

SuperFra, 2017-09-01:
The 2 interfaces are not in the same subnet. At this moment the request are sent always on the ethernet, but the lan is not connected to internet so the request fails.

me-no-dev, 2017-09-01:
There is [this](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/lwip/lwip/netif.h#L372) but I do not think you have a way to get the ```netif``` for a given interface, but I am soon to merge such way in IDF so you would be able to select which interface to use.

SuperFra, 2017-09-01:
Ok. Thanks to much for your answer. What do you think about, how long time it take? In the while can you suggest me any workaround?

me-no-dev, 2017-09-02:
I can but you will need to use Arduino as IDF component :) are you ready for that? See the readme for instructions and once you have it running I'll give you a small patch to integrate so you can have access to the netif

SuperFra, 2017-09-02:
YES! I'm ready. Currently I created a new project in IDF using Arduino as component and I done the porting of my sketch in it. Everythings is working. What is the next step? 
TIA

me-no-dev, 2017-09-03:
OK :) first edit ```components/tcpip_adapter/include/tcpip_adapter.h``` and add at the bottom:
```cpp
/**
 * @brief  Get the LwIP netif* that is assigned to the interface
 *
 * @param[in]   tcpip_if: the interface which we will get the hostname
 * @param[out]  struct netif * netif: resulting interface
 *
 * @return ESP_OK:success
 *         ESP_ERR_TCPIP_ADAPTER_IF_NOT_READY:interface status error
 *         ESP_ERR_TCPIP_ADAPTER_INVALID_PARAMS:parameter error
 */
esp_err_t tcpip_adapter_get_netif(tcpip_adapter_if_t tcpip_if, struct netif ** netif);
```
Then edit ```components/tcpip_adapter/tcpip_adapter_lwip.c``` and add at the bottom:
```cpp
esp_err_t tcpip_adapter_get_netif(tcpip_adapter_if_t tcpip_if, struct netif ** netif)
{
    if (tcpip_if >= TCPIP_ADAPTER_IF_MAX) {
        return ESP_ERR_TCPIP_ADAPTER_INVALID_PARAMS;
    }

    *netif = esp_netif[tcpip_if];

    if (*netif == NULL) {
        return ESP_ERR_TCPIP_ADAPTER_IF_NOT_READY;
    }
    return ESP_OK;
}
```

Then in your sketch something like:
```cpp
bool setDefaultIf(tcpip_adapter_if_t tcpip_if){
  struct netif * _netif;
  if(tcpip_adapter_get_netif(tcpip_if, &_netif)){
    //error
    return false;
  }
  netif_set_default(_netif);
  return true;
}

//to use it
setDefaultIf(TCPIP_ADAPTER_IF_STA);
```

Keep in mind this is only theory on my side :D I hope that it will work for you ;)

ghostzero2000, 2017-09-04:
Hello. I'm trying to set up esp32 as an access point. Make a bridge between Ethernet LAN8720 and wifi. I looked through the Internet and didn`t found ready solutions.
Prompt please the decision or where to search for the decision.

Thank you.

SuperFra, 2017-09-05:
Thanks a lot for your help. It works!!! Anyway, at first I received the error:"_'netif_set_default' was not declared in this scope_" but I solved using #include "lwip/netif.h" in my sketch.
I have just a question:
Can I use directly this function 
```
esp_err_t tcpip_adapter_set_netif(tcpip_adapter_if_t tcpip_if)
{
    if (tcpip_if >= TCPIP_ADAPTER_IF_MAX) {
        return ESP_ERR_TCPIP_ADAPTER_INVALID_PARAMS;
    }
	netif_set_default(esp_netif[tcpip_if]);
    
    return ESP_OK;
}
```
and call it in the sketch? Is it an error?
Thanks again.

me-no-dev, 2017-09-05:
I don't think that you have access to ```esp_netif[]``` from the sketch

SuperFra, 2017-09-05:
Sorry, my mistake. I did mean to write the function tcpip_adapter_set_netif in the file tcpip_adapter_lwip.c.

me-no-dev, 2017-09-05:
there is no need ;) the function that I asked you to add will actually be added to IDF in near future, because it's needed by other components. So you adding your own function would just complicate things.

SuperFra, 2017-09-07:
Thanks a lot for your help ;)

closed 2017-09-07 by SuperFra

rammichael1, 2017-10-18:
here i am trying to connect USR-TS232t2 LAN to my ESP32. how to connect to wifi through LAN Please he me with the code

me-no-dev, 2017-10-18:
what is that? and what do you mean by "connect to wifi through LAN"?

rammichael1, 2017-10-20:
Actually, USR TCP232T2 is ethernet converter http://www.sunrom.com/p/serial-uart-to-ethernet-convertertcpip-module-usr-tcp232-t
i want to send and receive data through this. But i am confused how to do that, Please share any examples thanks in advance

sauttefk, 2017-10-20:
@rammichael1:
* You are posting to an issue tracker.
* This is not a help forum
* You are hijacking this issue with a off-topic problem of yours.

Your device is a serial port server https://en.wikipedia.org/wiki/Serial_over_LAN this is no SLIP device https://de.wikipedia.org/wiki/Serial_Line_Internet_Protocol

yanzixiang, 2019-08-01:
> Hello. I'm trying to set up esp32 as an access point. Make a bridge between Ethernet LAN8720 and wifi. I looked through the Internet and didn`t found ready solutions.
> Prompt please the decision or where to search for the decision.
> 
> Thank you.

Have u find a solution to do this ?