espressif/arduino-esp32 #604: Compiles, but crashes when adding NULL to char array, empty or not.

mitchsf, 2017-09-01:
### Hardware:
Board:	ESP32 DevKitC						
Core Installation/update date: 8/15?		
IDE name:	Arduino	IDE					
Flash Frequency:	80Mhz			
Upload Speed:		921600				

### Description:
This compiles, but crashes when run. Sketch below demonstrates the issue. With my larger program, I was also getting errors in WString.h, line 47.

void setup() {
char string[50];
strcat(string1, '\0');
}

void loop() {

}

Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x4000c52a  PS      : 0x00060030  A0      : 0x800d0599  A1      : 0x3ffc7650  
A2      : 0x3ffc7670  A3      : 0x00000000  A4      : 0x00000020  A5      : 0x80000020  
A6      : 0x00000003  A7      : 0x00060e23  A8      : 0x00000000  A9      : 0x3ffc7670  
A10     : 0x00000000  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x3ffc3240  
A14     : 0x00000003  A15     : 0x00060e23  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4000c52a:0x3ffc7650 0x400d0599:0x3ffc7670 0x400d93ae:0x3ffc76d0

CPU halted.

Decoding 6 results
0x400d05a8: setup() at C:\Users\Mitchell\SkyDrive\Arduino\sketch_sep01a/sketch_sep01a.ino line 8
0x400d93be: loopTask(void*) at C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 17 (discriminator 1)

mitchsf, 2017-09-01:
Should be char string1[50];

devyte, 2017-09-01:
Also, your string array is not initialized.
string[0] = 0;

mitchsf, 2017-09-01:
It still crashes when run.

void setup() {
char string1[50];
string1[0] = 'A';
strcat(string1, '\0');
}

void loop() {
}

devyte, 2017-09-01:
Your string has to be null terminated, this is basic C programming. Strcat
will search for the first null, and concatenate after that.
string[0] = 'A';
string[1] = 0;

mitchsf, 2017-09-08:
Maybe I was not clear. My point, with these quick examples, was that I don't believe a program should crash if a string is initialized with NULL. If that assumption is incorrect, then I'm wrong.

me-no-dev, 2017-09-08:
It could be the fact that you are trying to strcat 0 to the string? What is the purpose of the code above?

mitchsf, 2017-09-08:
Nothing, really. I was just trying to show a condition in a 3000 line program, that I've since corrected. I just thought that it should not have crashed the program.

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

(hint: you're use of strcat is wrong. strcat takes a pointer to a string as second parameter (char *) not char.)

closed 2018-04-19 by everslick