espressif/arduino-esp32 #607: Preferences does not save data

bbasil2012, 2017-09-04:
### Hardware:
Board:							LOLIN32 v1.0.0
Core Installation/update date:			01.09.2017
IDE name:							Arduino IDE 1/8/4
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:

Preferences does not save data

### Sketch:

#include <Preferences.h>
typedef struct _newCh {
  int value[16]={-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1}; 
  long tp[16];                                                     
  byte color[3];                                                   
  byte group;                                                      
  byte type;                                                       
  byte inv;                                                         
  char name[32]="";                                                   
  byte gpio;
  int freq;
  byte bits;
  
};
_newCh newCh[16];

void setup() {
  Preferences preferences;
  Serial.begin(115200);
  preferences.begin("test", false);
  
  String ch_name;
  ch_name="name ";
  for (byte i=0;i<16;i++){
    ch_name+=String(i);
    ch_name.toCharArray(newCh[i].name,32);
  }
  Serial.println("Start saving...");
  preferences.putBytes("name",newCh,sizeof(newCh));
  Serial.println("Done");
  preferences.end();
  
}

void loop() {}
### Debug Messages:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:708
load:0x40078000,len:0
load:0x40078000,len:11460
entry 0x400789f4
Start saving...
[E][Preferences.cpp:272] putBytes(): nvs_set_blob fail: name k
Done

copercini, 2017-09-09:
@bbasil2012 why not use `preferences.putString("test",ch_name);`  ?

lonerzzz, 2017-09-28:
@bbasil2012 Any update on this issue? Any further details so you can receive assistance?

copercini, 2017-10-20:
Closed due no response =/

closed 2017-10-20 by copercini