espressif/arduino-esp32 #610: client.write sending time is too long

ShawnXuan, 2017-09-05:
### Hardware:
Board:							?ESP32S?
IDE name:							?Arduino IDE?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
I found it may take very long time to send a message using `client.write`. sometime more than 1 second. especially when the model is on moving status or be waved, but it is very close to the router(within 5m)

### Sketch:
here is test code. the sending time(`delt_t`) which is less than `1` is ignored.
```cpp
  int delt_t;
  delt_t = millis();
  int res = client.write(message,sizeof(message));
  delt_t = millis() - delt_t;
  if(delt_t>1){
    Serial.print("client.write time: ");
    Serial.println(delt_t);
  }
```

### Debug Messages:
the numbers is not start with 'client.write time:' is the main loop frequency. 
```
154
client.write time: 1059
27
118
153
154
154
154
154
154
154
154
153
154
client.write time: 29
150
154
154
153
client.write time: 1196
99
25
154
154
153
154
154
154
154
client.write time: 1201
104
18
154
154
154
client.write time: 1176
34
93
154
154
153
154
154
client.write time: 25
client.write time: 8
149
154
153
154
```
Thanks!

me-no-dev, 2017-09-05:
Just to clarify, you are saying that you have the device on a moving platform and when it's moving, you experience long time to send a packet. What is the size of the message? What is the model of the router? Have you checked RSSI? Maybe noise that comes from the other electronics/motors is creating the issue? Maybe the ESP32 is somewhat restricted?

ShawnXuan, 2017-09-06:
- Yes, it's a moving platform, the ESP32 connects MPU9250 only via I2C. 
- the newest router is **NETGEAR R8000 AC3200M.**
- The size of message is **83 bytes**(raw data)
- normally, I send 100 packets per second(**100Hz**). and I changed from 10Hz, 20Hz and 50Hz, no improvement.
- I will check RSSI later, the distance between ESP32 and the router is no more than 10 meter indoor open area.
- without moving, the throughput rate can keep still. the movement is the main reason

Thanks!