espressif/arduino-esp32 #611: Help with preferences

sansillusion, 2017-09-06:
### Hardware:
Board:							NodeMCU-32S
Core Installation/update date:			today
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I would like to store Dallas temperature devices address in preferences (so I can manipulate them in web app)

    DeviceAddress Probe01 = { 0x28, 0xFF, 0xF0, 0x73, 0x01, 0x17, 0x04, 0x86 }; 
    DeviceAddress Probe02 = { 0x28, 0xFF, 0x58, 0x54, 0x01, 0x17, 0x04, 0x7D };
    DeviceAddress Probe03 = { 0x28, 0xFF, 0x3A, 0x36, 0x01, 0x17, 0x05, 0x26 };
    DeviceAddress Probe04 = { 0x28, 0xFF, 0x2E, 0x34, 0x01, 0x17, 0x05, 0x42 };
    DeviceAddress Probe05 = { 0x28, 0xFF, 0xC9, 0x79, 0x01, 0x17, 0x04, 0xEA };
    DeviceAddress Probe06 = { 0x28, 0xFF, 0xF5, 0x2F, 0x01, 0x17, 0x05, 0xC0 };

I I try to use a String like so :  

    String str = "0x28, 0xFF, 0xF0, 0x73, 0x01, 0x17, 0x04, 0x86";
    DeviceAddress Probe01 = { str }; 

I get "Cannot convert 'String' to 'uint8_t {aka unsigned char}' in initialization"
I am verry confused with all the variable types in C. What variable type should I use and I need it to be compatible with "Preferences.h" like : 

    int ahi = preferences.getInt("add1", "0x28, 0xFF, 0xF0, 0x73, 0x01, 0x17, 0x04, 0x86");

???
I am lost in all those variables types and the different way to use them any help would be welcome.

tetious, 2017-09-07:
Looks like DeviceAddress is just an 8 byte array:
```c++
typedef uint8_t DeviceAddress[8];
```
Therefore you'll want to use putBytes and getBytes like this:
```c++
// to save
preferences.putBytes("Probe01", &Probe01, 8);

// to load
DeviceAddress Probe01; // declare the array first!
preferences.getBytes("Probe01", &Probe01, 8);
```

sansillusion, 2017-09-07:
I went another way and instead of using/storing addresses I used the sensors.getAddress() method
    sensors.getAddress(Probe0, 0);
that lets me use a simple int value.
Working on it here : https://github.com/sansillusion/esp32GrowSystem

But thanks a lot for your help, I was almost there with my tests I was missing the 8 in : 
    preferences.getBytes("Probe01", &Probe01, 8);

closed 2017-09-07 by sansillusion