espressif/arduino-esp32 #612: Broken OTA updates since 0.10.0

TheSimpleZ, 2017-09-06:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
Core Installation/update date:			5/sep/2017
IDE name:							Platform.io
Flash Frequency:					default?
Upload Speed:						115200

### Description:
Update 0.10.0 broke OTA updates. I can flash the BasicOTA sketch through the serial port, but when i try to flash OTA i get the result 1436 and then the error 1436. Below is a picture of what's happening:
[image]

When i monitor the device through the serial port i get the following messages:
[image]

If I set the platform dependency to 0.9.1 everything works perfectly. 

### Sketch:
BasicOTA example found bundled in the package.

copercini, 2017-09-06:
After that, it runs the new firmware or the old one?

TheSimpleZ, 2017-09-06:
To clarify: If the device is running 0.10.0 i get those error messages no matter what version i'm trying to send through OTA. If the device is running any older firmware, it works.

copercini, 2017-09-06:
On arduino IDE this number "1436" is the size of the new sketch, it show as error but the update works fine and after reboot the updated firmware is running

My suspect it's occurring in PIO too

me-no-dev, 2017-09-06:
What is 0.10.0 and 0.9.1? Where can I see the difference?

TheSimpleZ, 2017-09-06:
That is the version number of th Esp32 package when downloaded through PlatformIO. Or is this the wrong repo?

me-no-dev, 2017-09-06:
Try the staging version ;) I just tested OTA with the latest repo and it works.

TheSimpleZ, 2017-09-06:
Ok, thanks!

beegee-tokyo, 2017-09-23:
Still get OTA update failures, even using the staging version :(
platformio.ini:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp32dev
framework = arduino
upload_port = ESP32-Test-10FCFC3F.local
```
Output on console:
```
Uploading .pioenvs\esp32dev\firmware.bin
10:58:57 [DEBUG]: Options: {'esp_ip': 'ESP32-Test-10FCFC3F.local', 'host_port': 49353, 'image': '.pioenvs\\esp32dev\\firm
ware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 3232, 'spiffs': False, 'debug': True, 'progress': True}
10:58:57 [INFO]: Starting on 0.0.0.0:49353
10:58:57 [INFO]: Upload size: 636784
10:58:57 [INFO]: Sending invitation to: ESP32-Test-10FCFC3F.local

10:58:58 [INFO]: Waiting for device...
Uploading: [============================================================] 100% Done...
 
10:59:26 [INFO]: Waiting for result...
10:59:26 [INFO]: Result: 43614361436143614361436143614361
 
10:59:26 [ERROR]: 43614361436143614361436143614361
*** [upload] Error 1
============================================== [ERROR] Took 40.64 seconds ==============================================
```
Output on Serial debug:
```
Start updating sketch
[E][WiFiClient.cpp:223] write(): 104
[E][Updater.cpp:245] end(): premature end: res:0, pos:630784/636784

Error[4]: End Failed
```
Same with 
`platform = https://github.com/platformio/platform-espressif32.git#develop`
and
`platform = espressif32`

Looking on the debug output, it seems that the transfer was stopped before the complete file was sent:
[E][Updater.cpp:245] end(): premature end: res:0, pos:**630784**/636784

me-no-dev, 2017-09-25:
```https://github.com/platformio/platform-espressif32.git``` <<< Not latest staging! Please check PIO documentation how to use the latest staging which is pulled directly from this repository.

beegee-tokyo, 2017-09-25:
@me-no-dev 
I got the information how to use staging from [Using Arduino Framework with Staging version](http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version) and I am not using `https://github.com/platformio/platform-espressif32.git` but `https://github.com/platformio/platform-espressif32.git#feature/stage` as stated in my comment.

If this is wrong, please teach me the correct URL.

Thank you.

me-no-dev, 2017-09-25:
@ivankravets can you please make sure that this is correct? Thank you :)

ivankravets, 2017-09-25:
@me-no-dev , @beegee-tokyo 

It should be 
```
[env:...]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
```

P.S: Please note that this is an upcoming feature using a platform as VCS url. You need the latest development version of PIO Core 3.5.0

If you use PIO Core < 3.5.0, you need 
1. `pio platform install https://github.com/platformio/platform-espressif32.git#feature/stage`
2. Use in `platformio.ini`
```
[env:...]
platform = espressif32_stage
```

You can check your PIO Core version via `pio --version` command.

loa, 2017-09-25:
I tried back and forth but sadly get stuck in the same issue.

These are commands ran just now:

```
$ pio --version
PlatformIO, version 3.5.0a16
```

```
[env:nodemcu-32s]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = nodemcu-32s
framework = arduino
upload_port = 192.168.0.107
```

```
Checking program size
text	   data	    bss	    dec	    hex	filename
413557	  73600	  28960	 516117	  7e015	.pioenvs/nodemcu-32s/firmware.elf
Building .pioenvs/nodemcu-32s/firmware.bin
esptool.py v2.0-beta1
Looking for upload port...
Use manually specified: 192.168.0.107
Uploading .pioenvs/nodemcu-32s/firmware.bin
21:48:56 [DEBUG]: Options: {'esp_ip': '192.168.0.107', 'host_port': 35743, 'image': '.pioenvs/nodemcu-32s/firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 3232, 'spiffs': False, 'debug': True, 'progress': True}
21:48:56 [INFO]: Starting on 0.0.0.0:35743
21:48:56 [INFO]: Upload size: 610832
21:48:56 [INFO]: Sending invitation to: 192.168.0.107
21:48:57 [INFO]: Waiting for device...
Uploading: [============================================================] 100% Done...

21:49:14 [INFO]: Waiting for result...
21:49:14 [INFO]: Result: 14601460

21:49:14 [ERROR]: 14601460
*** [upload] Error 1
```

ivankravets, 2017-09-25:
@loa could you run in PIO IDE Terminal `pio update` and try again?

P.S: I've just updated esptool.py and espota.py to the latest versions.

me-no-dev, 2017-09-26:
yes, espota.py must be old because 1460 is the old packet size. In the new one 1024 is used

bill-orange, 2017-09-26:
I just tried the OTA example code with no success at all.  Something appears broken in the latest changes.
Here's what I get:

 (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:572
load:0x40078000,len:0
load:0x40078000,len:9880
entry 0x400789d8
⸮

me-no-dev, 2017-09-26:
@bill-orange Nope ;) OTA is just fine. Here is the output of BasicOTA:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
Booting
[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 7 - STA_GOT_IP
Ready
IP address: 192.168.42.190
[D][Updater.cpp:123] begin(): OTA Partition: app1
Start updating sketch
Progress: 100%
End
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
Booting
[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 7 - STA_GOT_IP
Ready
IP address: 192.168.42.190
```

me-no-dev, 2017-09-26:
@beegee-tokyo that error with number does not necessarily mean that upload failed. it's some weird way that the python get's the result... will eventually figure out how to make it work and not show fail when it actually worked.

beegee-tokyo, 2017-09-26:
@me-no-dev 
I can confirm after **`pio update`** the OTA works, but PlatformIO sometimes still gives error message at the end.
What I found so far is that if I **clean** the project and then use **`platformio run --target upload`** I get successful OTA update and no error message. If I start **`platformio run --target upload`** again (without rebuilding the project) OTA works but I get an error message on PlatformIO.

@ivankravets 
I think there is something wrong on the PlatformIO part of the ESP32 package/board/???
I always need to run **`platformio run --target upload`** from the terminal. The buttons/menu entries for `build`, `upload `and `clean `are all doing a build, but no clean and no upload.

bill-orange, 2017-09-26:
Okay, I should have done more testing before my last post.  I retested and the OTA example ran fine.  It failed when added to my sketch with the repeating string of boot gibberish.  I started adding the #includes from my sketch into the running example.  The inclusion of

 #include "esp_system.h"

into the OTA example appears to be the cause of the crash.  Can someone confirm?

loa, 2017-09-26:
@ivankravets I did update and it works for me now! Several OTA updates without any error!

@bill-orange I don't have `esp_system.h` included yet in my small example.

beegee-tokyo, 2017-09-26:
@bill-orange 
I included **`esp_system.h`** and use some functions (_`esp_chip_info(&espInfo);`_, _`esp_read_mac(baseMac, ESP_MAC_WIFI_STA);`_) and have no problems. No crash, no problem with OTA.

bill-orange, 2017-09-26:
That is weird. All I have to do is put in the include statement and OTA crashes. I don't even have to use any functions.  The mere presence of the #include line is enough.  I am using Arduino IDE 1.8.4 on a Lolin32.  Is your setup very different in some way?

me-no-dev, 2017-09-27:
@bill-orange this really makes no sense.... must be something else :) including a header would not break the app. As you can see from the comments, others have no such issue :)

bill-orange, 2017-09-27:
I am not sure what’s going on. Perhaps it is an upload problem after all. I have not had problems uploading other sketches. Unfortunately I have only one ESP32 board and is now occupied serving up JSON strings. I will order a spare and do more testing.

beegee-tokyo, 2017-09-27:
@bill-orange 
I am using PlatformIO Version:` Home 0.3.0·Core 3.5.0a16`
Selected Board: `Espressif ESP32 Dev` Module ID: `esp32dev`
Selected Platform: `Espressif 32 (Stage)` Version: `aba1d7d`
platformio.ini:
```
[env:esp32dev]
platform = espressif32_stage
board = esp32dev
framework = arduino
upload_port = 192.168.0.111
```
I tried the original OTA sample code with and without `#include <esp_system.h>` and it always works.

bill-orange, 2017-09-27:
Interesting. I am compiling on an entirely different platform. I will continue testing at a later date. I only own one ESP32 and is busy serving up JSON strings right now.

bill-orange, 2017-09-27:
You have got me.  When I am able to test again, I will use esptool to erase the entire EEPROM before I start testing.  I have the Arduino IDE installed on a second machine.  I can test on it to see if my IDE is hosed up.  I will need to order another board from China so I do not have to take my only  ESP32 away from its server duties. I will report back later on this thread if it is still open.

bill-orange, 2017-10-15:
Well, my issue is solved. The reason for my OTA compile time crashes was lack of program memory.

closed 2017-10-20 by copercini

marbalon, 2017-10-28:
Is there any solution for this ? I have the same issue, no problem when uploading from Arduino, and errors when uploading from platformio:

Platformio + Atom and all tools installed yesterday on fresh Windows 10.

`
19:08:34 [INFO]: Waiting for result...
19:08:34 [INFO]: Result: 14361436
 
19:08:34 [ERROR]: 14361436
*** [upload] Error 1`

Sometimes different values as result:
`19:11:19 [INFO]: Waiting for result...
19:11:19 [INFO]: Result: 1460
 
19:11:19 [ERROR]: 1460
*** [upload] Error 1
 [ERROR] Took 11.34 seconds`

me-no-dev, 2017-10-30:
@marbalon you seem to be using an old copy of espota.py ;) we no longer send 1460 bytes packets.

proddy, 2017-12-28:
I have the same problem (espressif32 with OTA) after I ported my project & code from ESP8266 to ESP32. The upload hits 100% but returns an error code in the result. The firmware has been successfully uploaded though. Tried with VS Code and plain ole command line. Using latest PlatformIO 3.5.0, esptool.py v2.1

ivankravets, 2017-12-28:
Do you use http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version ?

```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp32dev
framework = arduino
upload_port = ***********
```

proddy, 2017-12-29:
thanks @ivankravets for the quick response - confirmed OTA is now working using the latest staging version (https://github.com/platformio/platform-espressif32/tree/feature/stage)

usmanshahid001, 2018-03-05:
hey i am ubdating device from web server but due to rtos functions its being crashed ..
any advice for me...i am using IDE

williamesp2015, 2018-03-30:
I can not program ESP32 (OTABasic.ino) using PlatformIO in Windows 10 with staging or developed version:
PlatformIO IDE and COre and all library updated. 
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
;platform = espressif32
board = esp32dev
framework = arduino
upload_port = 192.168.1.6
Error is:
18:51:02 [INFO]: Upload size: 599600
Sending invitation to 192.168.1.6 ...
18:51:05 [INFO]: Waiting for device...
18:51:15 [ERROR]: No response from device

beegee-tokyo, 2018-03-30:
@williamesp2015 can you reach the ESP32 with `ping 192.168.1.6`?
Are you sure that the IP address is correct?
Are you calling `ArduinoOTA.handle()` inside `loop()`?