espressif/arduino-esp32 #613: SD.begin() crashes the board if called again after failing to mount

tetious, 2017-09-07:
### Hardware:
Board:							ESP32-DevKitC
Core Installation/update date:			9/6/2017
IDE name:							Tested with platformio stage and "stable"

### Description:

If the first attempt mounting an SD card via SD.begin() fails for some reason (such as the SPI bus is disconnected), subsequent calls crash the board. I suspect something is either not getting cleaned up or is getting cleaned up incorrectly when the mount fails.

If the card mounts successfully, I can run SD.end() and then SD.begin() as many times as I like without issue.

### Sketch:
You can reproduce this by running the sketch below **while not having an SD card reader attached.**

```cpp
#include <Arduino.h>
#include <SD.h>

void setup() {  
    Serial.begin(115200);
}

void init_sd() {
    if(!SD.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
}

void loop() {
    Serial.println("First attempt, will fail.");
    init_sd();
    delay(5000);
    Serial.println("Second attempt, will crash the board.");
    init_sd(); // boom
}
```

### Debug Messages:
```
First attempt, will fail.
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
Card Mount Failed
Second attempt, will crash the board.
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1792 (vQueueDelete)- assert failed!
abort() was called at PC 0x40084a97 on core 1

Backtrace: 0x40087010:0x3ffccb40 0x4008710f:0x3ffccb60 0x40084a97:0x3ffccb80 0x400fbf21:0x3ffccba0 0x400fa7c2:0x3ffccbc0 0x400f134a:0x3ffccc00 0x400f079d:0x3ffccc30 0x400d0a39:0x3ffccc60 0x400d0a6b:0x3ffccc80 0x400f21d5:0x3ffccca0

Rebooting...
ets Jun  8 2016 00:22:57

```
### Stacktrace:
No extra info seems to show up with debug on, but I've attached a decoded but not particularly helpful stacktrace:
```
0x00000bac: ?? ??:0
0x400fa7c2: f_mount at ff.c:4443
0x3ffccc80: ?? ??:0

```

Thanks for all your awesome work and let me know if I can provide any additional information.

luc-github, 2017-10-11:
I can reproduce this issue also with SD card reader attached but no SD Card present
Using spi.end(); neither SD.end(); after mount failed help to avoid issue

Aleksandrovas, 2017-11-02:
Soliution is verry simple, open SD.c and in 39 line replace sdcard_uninit(_pdrv); to sdcard_unmount(_pdrv);
Espressif please fix it.

me-no-dev, 2017-11-06:
done ✅

closed 2017-11-06 by me-no-dev

psatya111, 2018-01-17:
I have replace sdcard_uninit(_pdrv); to sdcard_unmount(_pdrv); but still sd card mount failed in 2nd time onwards. first time it is working fine.please help me.

MikeyMoMo, 2018-03-02:
Just did this.  From total fail to totally working.  Miracle fix!