espressif/arduino-esp32 #614: WIFI wont connect to AP anymore (worked for at least a month)

sansillusion, 2017-09-08:
### Hardware:
Board:							WEMOS LOLIN32
Core Installation/update date:			??
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
My esp32 will no longer connect to any AP, it used to work for at least a month and one day I rebooted the router and it would keep connecting with the wrong IP (should have been 192.168.0.8 from DHCP server but used 192.168.0.80)
I kept rebooting the device and router to try to get it to play nice again (like it used to without issues) and then it just kept getting it STA mode (used wifimanager).

I did some test and it goes in STA mode with no issues but will no longer connect to any AP (tried 3 that all works with other esp32 devices and used to work with this one). even the simple example sketch just keeps trying to connect with no success.

### Sketch:

https://github.com/sansillusion/espLightServer

This is the sketch that was running on it for a long time but even the example sketch does not connect anymore :

/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <WiFi.h>

const char* ssid     = "MYAP";
const char* password = "MYPASSKEY";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

int value = 0;

void loop()
{
    delay(5000);
    ++value;

    Serial.print("connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;
    const int httpPort = 80;
    if (!client.connect(host, httpPort)) {
        Serial.println("connection failed");
        return;
    }

    // We now create a URI for the request
    String url = "/input/";
    url += streamId;
    url += "?private_key=";
    url += privateKey;
    url += "&value=";
    url += value;

    Serial.print("Requesting URL: ");
    Serial.println(url);

    // This will send the request to the server
    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            Serial.println(">>> Client Timeout !");
            client.stop();
            return;
        }
    }

    // Read all the lines of the reply from server and print them to Serial
    while(client.available()) {
        String line = client.readStringUntil('\r');
        Serial.print(line);
    }

    Serial.println();
    Serial.println("closing connection");
}

sansillusion, 2017-09-08:
I have also tried to run : 
    python ~/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM6 --baud 115200 --before default_reset --after hard_reset erase_flash

to make sure flash memory was ok nothing changed

sansillusion, 2017-09-08:
I must add it stopped connecting to AP all together (not just getting wrong IP) after I pressed the boot button 3 times in a row.

I can still program it and use it in STA mode, wifimanager does show the AP's available with strong signal.

BudBundi, 2017-09-08:
Last time this happened to me adding
WiFi.disconnect()
before 
WiFi.begin()
solved my problem.

sansillusion, 2017-09-08:
@BudBundi WOW Thank you !
That totaly worked !
Why would one need to send WiFi.disconnect() even after a  full power cycle ?

sansillusion, 2017-09-08:
I may have spoken too soon, it became verry buggy, wifimanager still wont work anymore I get : 

    *WM: WiFi save
    *WM: Sent wifi save page
    *WM: Connecting to new AP
    *WM: Connecting as wifi client...
    *WM: Connection result: 
    *WM: 6
    *WM: Failed to connect.

sansillusion, 2017-09-08:
WiFi.disconnect()

No longer works ! even in a simple test sketch, weird that it did work once.

agcaahmet, 2017-09-08:
If you are booting/starting from serial port, have you tried pressing reset button after?

sansillusion, 2017-09-08:
Yes I did, I also tried to connect to my laptop wifi and after a few tries it started to work again on my laptop but would not on my router.
I then played with the dhcp and mac fitering in my router since it was connecting to a wrong ip before failing, it started connecting with a wrong IP again so I just rebooted my router another time and all started to work again.

So it seems that a bad reboot of my router has corrupted the wifi connection of my esp32 (needed a few reset before it worked on my laptop).

Anyway I am happy to know it was probably related to my crappy router but it's weird that only this device stopped working not the others I have going (one NodeMCU-32S and a NodeMCU-8266, phones, laptops and PCs).

closed 2017-09-12 by sansillusion

dMEK, 2018-08-24:
just a wee tip, because this tripped me up recently using my Laptop as a mobile hotspot, make sure the band you are connecting to is 2.4GHz and not 5GHz

wisze, 2019-03-04:
I am having similar problems. I assume these boards have a really bad service life. The wifi connections just stops working after a few months. I used wemos lolin boards for a home server which switches on the lighting automatically and through a web interface. I've replaced the wemos board twice now,