espressif/arduino-esp32 #615: esp-idf commit 547ef83 breaks second serial initialization

forthlightning, 2017-09-08:
I'm using arduino as component, initializing HardwareSerial( UART_NUM_1 ). Code worked up until this commit.

I communicate with RS485 transceiver linked to UART1.

I tried to enable the uart with periph_module_enable(PERIPH_UART1_MODULE); after Serial1.begin(...), but it uses the wrong baudrate.

If I enable uart1 before hardwareserial, then it hits a wdt reset like:

ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:4
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:11360
entry 0x40078cb0
W (283) boot: PRO CPU has been reset by WDT.
W (283) boot: WDT reset info: PRO CPU PC=0x40083a4a
0x40083a4a: _uart_isr at /Users/xx/xx/xx/components/arduino-esp32/cores/esp32/esp32-hal-uart.c:335

W (284) boot: WDT reset info: APP CPU PC=0xae483a1
It looks like the arduino library should be setting these bits so that the peripheral timer starts.

me-no-dev, 2017-09-10:
Fixed in the latest commit :)

forthlightning, 2017-09-11:
Ok, works for that commit, walked up to 4f812eb, and now I'm hitting a wdt for a line which doesn't exist woohoo.

```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:4384
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:11360
entry 0x40078cb0
W (283) boot: PRO CPU has been reset by WDT.
W (283) boot: WDT reset info: PRO CPU PC=0x40083a10
0x40083a10: _uart_isr at /Users/............./components/arduino-esp32/cores/esp32/esp32-hal-uart.c:341
```

forthlightning, 2017-09-12:
with the same commit, the above problem persists, even with a single uart transmission

me-no-dev, 2017-09-12:
can you try now? I just updated everything

forthlightning, 2017-09-13:
awesome this fixed it!

forthlightning, 2017-09-13:
wait not really! It just made the bug more intermittent, almost to the point of working... :D 

Still seeing wdt timeouts at uartGetBaudRate. I have been initializing serial the same way for a couple months now so not sure what the issue is. 

Something new is that serial garbage after "mode:DIO, cloc"

not sure if this should be a new issue at this point, lemme know and I can make a new one.

```
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, cloc�x��x�~������x��xx�xx����f���怘����x��xx�xx����f���`怘����x�����f���������x�����f������x���f��`��f������怘��`��~�x�~f����~�f�������~f~ff�xfxf~f��f�~�����f��f���W (212) boot: PRO CPU has been reset by WDT.
W (212) boot: WDT reset info: PRO CPU PC=0x40083b4a
0x40083b4a: _uart_isr at /Users/maxfieldmorrison/NSR/solsense32/Firmware/solsense_esp32/components/arduino-esp32/cores/esp32/esp32-hal-uart.c:341

W (212) boot: WDT reset info: APP CPU PC=0x74f6b94b
```

me-no-dev, 2017-09-14:
WDT in uartGetBaudRate? that method is so short... this is rather strange. Cables all good? Maybe a bit of delay before initialization, though it makes no sense...

forthlightning, 2017-09-14:
it seems to come in bunches... 

I will see a couple wdt resets, in a row, and then it will be stable for a relatively long period of time.

It ran for 21 hours polling pretty consistently, and then restarted 5 times shortly after flashing, then stabilized.

forthlightning, 2017-09-14:
The code seems to run fine when there is a slave connected and responding. Hits the bug when there is no response, and retry logic starts kicking in. I can work around this in my code.... but this is something that has run for some time already, so it may be something that was introduced recently.

me-no-dev, 2017-09-14:
OK, can I please ask you to try something? Edit ```esp32-hal-uart.c``` and change [_uart_isr](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L70-L121) to:
```cpp
static void IRAM_ATTR _uart_isr(void *arg)
{
    uint8_t i, c;
    BaseType_t xHigherPriorityTaskWoken;
    uart_t* uart;

    for(i=0;i<3;i++){
        uart = &_uart_bus_array[i];
        uart->dev->int_clr.rxfifo_full = 1;
        uart->dev->int_clr.frm_err = 1;
        uart->dev->int_clr.rxfifo_tout = 1;
        while(uart->dev->status.rxfifo_cnt) {
            c = uart->dev->fifo.rw_byte;
            if(uart->queue != NULL && !xQueueIsQueueFullFromISR(uart->queue)) {
                xQueueSendFromISR(uart->queue, &c, &xHigherPriorityTaskWoken);
               if (xHigherPriorityTaskWoken) {
                    portYIELD_FROM_ISR();
                }
            }
        }
    }
}
```

forthlightning, 2017-09-14:
hard to tell if this made a difference. maybe slightly better, but I am still getting the same timeout behavior. Happens more often on first reading over UART, happens more often when no lines are connected, but still happens in normal configuration.

closed 2019-08-15 by stale[bot]