espressif/arduino-esp32 #616: AsyncTCP.h: No such file or directory

nardev, 2017-09-08:
I'm trying to compile one project which is for ESP8266 initially but i get this... 
could it be a path issue?

src/../lib/ESPAsyncWebServer/src/ESPAsyncWebServer.h:33:22: fatal error: AsyncTCP.h: No such file or directory

copercini, 2017-09-09:
I think you should get these files from here: https://github.com/me-no-dev/AsyncTCP

nardev, 2017-09-09:
Great but now an issue with SPIFFS

```
lib/ESPAsyncWebServer/src/SPIFFSEditor.h:6:20: fatal error: SPIFFS.h: No such file or directory
```

:(

copercini, 2017-09-09:
This is very recent ( few hours ago) @me-no-dev is still adjusting the things to release officially...

Soon everything will work fine =)

nardev, 2017-09-09:
@copercini thnx

me-no-dev, 2017-09-09:
you can get it going without SPIFFS if you edit one file ;)
comment ```#include "SPIFFS.h"``` and remove ```=SPIFFS``` from the constructor [here](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.h). Feel free to substitute with "SD" or "SD_MMC" also

me-no-dev, 2017-09-09:
I actually adjusted the code so it no longer requires SPIFFS at all.
To enable the editor, on ESP32 you need to give the FileSystem as first argument:
```cpp
//SPIFFS
server.addHandler(new SPIFFSEditor(SPIFFS, "user","password"));
//SD
server.addHandler(new SPIFFSEditor(SD, "user","password"));
//SD_MMC
server.addHandler(new SPIFFSEditor(SD_MMC, "user","password"));
```

copercini, 2017-09-16:
@nardev with last git version everything should work fine now =)

closed 2017-09-16 by copercini