espressif/arduino-esp32 #623: Hi, I am porting a gateway application currently running on Arduino (and WeMos D1 mini) and using a RFM69 SPI transceiver to a WeMos Lolin32. This one is connected using the SPI pins (SS, MOSI, MISo, SCK) and is using A16 as Interrupt. It looks working find, however I have random SPI_FAST_FLASH_BOOT during data exchange (on heavy traffic once per minute 1 of 3 times and then back to normal) . I spend already few days to insert yield() commands in several while loops and critical data transfer functions without success. I wonder if somebody could help or to give me hints to investigate this issue. Thanks  Robert  CODE: SELECT ALL rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) ets Jun  8 2016 00:22:57  rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) configsip: 0, SPIWP:0xee clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00 mode:DIO, clock div:1 load:0x3fff0010,len:4 load:0x3fff0014,len:588 load:0x40078000,len:0 load:0x40078000,len:10472 entry 0x40078a28

rrobinet, 2017-09-11:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

me-no-dev, 2017-09-11:
First: Please do not write your full issue in the title :)
Second: When filling an issue you were presented with template that you were supposed to fill. In that template there was a link to a tool to let you debug that exact issue.
https://github.com/me-no-dev/EspExceptionDecoder

rrobinet, 2017-09-11:
Sorry about that ... it was a mistake

closed 2017-09-11 by rrobinet