espressif/arduino-esp32 #624: Wemos Lolin32 Watchdog reset using RFM69 Library

rrobinet, 2017-09-11:
### Hardware:
Board:							WeMos LOLIN V1.0.0
Core Installation/update date:			11/jul/2017 (https://github.com/espressif/arduino-esp32)
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Hi,
I am porting a gateway application currently running on Arduino (and WeMos D1 mini) and using a RFM69 SPI transceiver to a WeMos Lolin32.
This one is connected using the SPI pins (SS, MOSI, MISO, SCK) and is using A16 as Interrupt.
It looks working find, however I have random SPI_FAST_FLASH_BOOT during data exchange (on heavy traffic once per minute 1 of 3 times and then back to normal) .
I spend already few days to insert yield() commands in several while loops and critical data transfer functions without success.
I wonder if somebody could help or to give me hints to investigate this issue.
Thanks 
Robert

### Debug Messages:
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:10472
entry 0x40078a28 
```

me-no-dev, 2017-09-11:
can you provide some code? Log shows nothing that we could go on by :)
yield is not needed on esp32 :)

rrobinet, 2017-09-11:
Please find attached the scripts with some explanation.
Again these scripts runs perfectly on a Wemos D1 mini (no yield()). 
Regarding the yield(), In deed I have tested long delays on WeMos LOLIN and I never at the need to use yield such as for the ESP8266.
The scripts used are derived from the RFM69 Library (see https://github.com/LowPowerLab/RFM69) example (Struct_receive.ino and Struct_send.ino) and adapted for the particularities of WeMos (no flash memory test,  integer replaced by long integer ,  normalised structure,...)
In this case only RFM69.cpp and RFM69.h is used (no RFM_ATC, no RFM_OTA) 
RFM69W transceiver (IS_RFM69HW is set to false) is connected on the WEMOS LOLIN SPI bus and is using external interrupt (in my case A16 is used).
The error described above is obtained by running the RFM69_Struct_receive.ino **on the WeMos Lolin**. The sender is a Moteino standard model that includes an RFM69W transceiver (IS_RFM69HW is set to false), but the sketch is adapted to be compatible width the data structure of the WeMos.
See the two scripts in attachment (LOLIN_RFM69_Struct_receive.ino (the one for WeMo) and Moteino_for_LOLIN_Struct_send.ino

[Wemos_RFM69_Sketches.zip](https://github.com/espressif/arduino-esp32/files/1293117/Wemos_RFM69_Sketches.zip)
You should see from the Wemos Serial interface:
```
[99]  [RX_RSSI:-29] nodeId=99 uptime=122200 temp=91.23 - ACK sent.
[98]  [RX_RSSI:-28] nodeId=98 uptime=35700 temp=91.23ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8

Listening at 433 Mhz...
[99]  [RX_RSSI:-28] nodeId=99 uptime=123639 temp=91.23 - ACK sent. Pinging node 99 - ACK...nothing
[99]  [RX_RSSI:-29] nodeId=99 uptime=123700 temp=91.23 - ACK sent.
[98]  [RX_RSSI:-29] nodeId=98 uptime=37200 temp=91.23 - ACK sent.
[99]  [RX_RSSI:-11] nodeId=99 uptime=123800 temp=91.23 - ACK sent. Pinging node 99 - ACK...ok!
[99]  [RX_RSSI:-29] nodeId=99 uptime=123900 temp=91.23 - ACK sent.
[99]  [RX_RSSI:-29] nodeId=99 uptime=124000 temp=91.23 - ACK sent.
[98]  [RX_RSSI:-29] nodeId=98 uptime=37500 temp=91.23 - ACK sent. Pinging node 98 - ACK...nothing
[99]  [RX_RSSI:-11] nodeId=99 uptime=124100 temp=91.23 - ACK sent.
[99]  [RX_RSSI:-28] nodeId=99 uptime=124200 temp=91.23 - ACK sent.
[99]  [RX_RSSI:-29] nodeId=99 uptime=124300 temp=91.23 - ACK sent. Pinging node 99 - ACK...ok!
[98]  [RX_RSSI:-28] nodeId=98 uptime=37800 temp=91.23 - ACK sent.
[99]  [RX_RSSI:-11] nodeId=99 uptime=124400 temp=91.23 - ACK sent.
[99]  [RX_RSSI:-28] nodeId=99 uptime=124500 temp=91.23 - ACK sent. Pinging node 99 - ACK...nothing
[99]  [RX_RSSI:-29] nodeId=99 uptime=124600 temp=91.23 - ACK sent.
[98]  [RX_RSSI:-28] nodeId=98 uptime=38100 temp=91.23 - ACK sent.
[99]  [RX_RSSI:-11] nodeId=99 uptime=124700 temp=91.23 - ACK sent. Pinging node 99 - ACK...ok!
[99]  [RX_RSSI:-29] nodeId=99 uptime=124800 temp=91.23 - ACK sent.
[99]  [RX_RSSI:-29] nodeId=99 uptime=124900 temp=91.23 - ACK sent.
[98]  [RX_RSSI:-28] nodeId=98 uptime=38400 temp=91.23 - ACK sent. Pinging node 98 - ACK...nothing
[99]  [RX_RSSI:-12] nodeId=99 uptime=125000 temp=91.23ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8

Listening at 433 Mhz...
[99]  [RX_RSSI:-28] nodeId=99 uptime=126374 temp=91.23 - ACK sent. Pinging node 99 - ACK...nothing
[98]  [RX_RSSI:-15] nodeId=98 uptime=39900 temp=91.23 - ACK sent.
[99]  [RX_RSSI:-28] nodeId=99 uptime=126476 temp=91.23 - ACK sent.
[99]  [RX_RSSI:-29] nodeId=99 uptime=126566 temp=91.23 - ACK sent. Pinging node 99 - ACK...ok!
[99]  [RX_RSSI:-28] nodeId=99 uptime=126603 temp=91.23 - ACK sent.
```
Note they the `ACK...nothing` is also another symptom because it should not occur with the RSSI level.

me-no-dev, 2017-09-11:
I would say the issue is in the interrupt handler. It's doing way too many things. How often do the interrupts happen? maybe you can adapt the lib a bit, set a flag in the interrupt and handle in the loop.

rrobinet, 2017-09-12:
It is very hard to debug because I can't actually see where it is happening.
Interrupts are asynchronous in the receiving part and as far I could see it the issue happens only during a reply to a received message or directly when sending a message after a message is received (if I disable the ACKReply or the SendWithRetry functions everything works fine). I ran the same script where the WeMos is sending data and the Moteino is receiving and everything looks also OK (a least for a few hours of the duration of the test).

However I also believe that the issue is related to the interrupt management which has an effect on the watchdog timers
Now on AVR the SPI library was modified sometimes ago for SPI_TRANSACTION, which is also the case for the WEMOS ESP8266 & ESP32 with the difference that the `SPIClass::usingInterrupt(uint8_t interruptNumber)` is not. 
To be conform with the SPI transaction structure I did already modify the RFM69 library with the `SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE0));` and   `SPI.endTransaction(); `without success. 
So there must be something different in the way ESP32 manage asynchronous interrupts vs AVR and ESP8266.

Is it a way to have a better log info reason why TG0WDT and TG1WDT expires simultaneously?
Robert

me-no-dev, 2017-09-12:
there seems to be something unclear :) not a SPI issue ;) you spend too much time in the interrupt. That is the problem. It reads lots of data and does many things, while it should have just done the minimal. Sine you reply in the interrupt, you spend too much time and trigger WDT. That is all that is happening. You can not spend that mush time there :) Lib needs to be adapted to work on ESP32

rrobinet, 2017-09-12:
OK, I will look to see if I can reduce the Interrupt code.
I will give you a feedback.

rrobinet, 2017-09-16:
hi @me-no-dev 
The issue is not related to the time spent in the `interruptHandler` function but in the SPI transfer handling.
With my last test, the Wemos LOLIN is sending data only and for each frame sent it is expecting an acknowledgement from the receiver.
The acknowledgement comes through an interrupt and the interruptHandler function is activated.

I have extremely simplified the interruptHandler function including one SPI transfer as follow.

```
void RFM69::interruptHandler() {
  // noInterrupts();
  SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE0));
  digitalWrite(_slaveSelectPin, LOW);
  SPI.transfer(0x28 & 0x7F);   // Read RFM69 register
  uint8_t regval = SPI.transfer(0);
  digitalWrite(_slaveSelectPin, HIGH);
  SPI.endTransaction(); 
  // interrupts(); 
}
```

In this case the time-out occurs quite often, nearly each time the interrupt is raised. Disabling and enabling the interrupt during the SPI transaction doesn’t help.
```
ets Jun  8 2016 00:22:57
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
t:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8

Transmitting at 433 Mhz...
Sending struct (12 bytes) ... ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8

Transmitting at 433 Mhz...
Sending struct (12 bytes) ...  nothing...
Sending struct (12 bytes) ... ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8

```

copercini, 2017-09-16:
@rrobinet Maybe I have a clue of what is happening...
Could you test the interruption in another pin, like A17 and see if the problem persists?

rrobinet, 2017-09-16:
@copercini 
I did the same test with A16 and 16 as interrupt pins but the issue is still the same

rrobinet, 2017-09-16:
Note that the same SPI transaction outside de interrupt works fine

rrobinet, 2017-09-16:
In fact:
```
void RFM69::interruptHandler() {
  SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE0));
  SPI.endTransaction(); 
}
```
is sufficient to have the problem

copercini, 2017-09-16:
oh, it's can be because SPI uses interrupt un-safe FreeRTOS APIs here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.c#L61-L62

rrobinet, 2017-09-16:
Sorry, what do you mean by that? 
Is it hopeless or should I use another module?

copercini, 2017-09-16:
In simple terms: SPI was not planned to run inside an interruption....

the simple solution:
make a flag triggered by the interruption and run SPI stuff outside it, like this:

```cpp
bool spiflag = false;

void RFM69::interruptHandler() {
    spiflag = true;
}

void loop{
    if(spiflag){
        //do SPI stuff here
        SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE0));
        SPI.endTransaction(); 
        spiflag = false; 
   }
}

```

rrobinet, 2017-09-16:
Well this is the drastic solution and is a major change to a library (RFM69)  that is used on several platform (AVR) Arduino, Moteino and  WeMos D1 and probably others. 
Because the RFM transceiver is using SPI with interrupt mode (like an Ethernet controller) , most of the received frames have to be processed as fast as possible, not in a loop (I will verify that)
Anyway this is not a global solution, but just a workaround for ESP32.
So the only thing I can conclude is that currently RFM69 library will not work as such on WEMOS LOLIN.
 It's a pity because a lot of House automation systems are using RFM69 transceivers.
Anyway thanks for your help.
Robert

copercini, 2017-09-16:
So, we can try the complex solution:

try edit these lines: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.c#L61-L62

replace by this:
```cpp
#define SPI_MUTEX_LOCK()    do {} while (xSemaphoreTakeFromISR(spi->lock, portMAX_DELAY) != pdPASS)
#define SPI_MUTEX_UNLOCK() xSemaphoreGiveFromISR(spi->lock)
```
After that close arduino IDE (or what IDE are you using), open it again and flash the sketch

copercini, 2017-09-16:
So, we can try the complex solution:

try edit these lines: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.c#L61-L62

replace by this:
```cpp
#define SPI_MUTEX_LOCK()    do {} while (xSemaphoreTakeFromISR(spi->lock, NULL) != pdPASS) //this while also can cause interrupt problems...
#define SPI_MUTEX_UNLOCK() xSemaphoreGiveFromISR(spi->lock, NULL)
```
After that close arduino IDE (or the IDE you are using), open it again and flash the sketch

rrobinet, 2017-09-16:
Yes this looks a complex one; the result is:
```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8

Transmitting at 433 Mhz...
Sending struct (12 bytes) ...  ok!
Sending struct (12 bytes) ...  ok!
Sending struct (12 bytes) ... /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1613 (xQueueReceiveFromISR)- assert failed!
abort() was?called at PC"Pets Jun  8 2016 00:22:57

```

**Please note that I will have to stop to investigate this issue for at least one week!**
I will continue when I will be back from abroad.
Thanks for your support
Robert

me-no-dev, 2017-09-18:
Was just gonna say it's not ISR safe :D but we need to find a better solution ;) 
```do {} while (xSemaphoreTakeFromISR(spi->lock, NULL) != pdPASS)``` in ISR is not a good idea.

rrobinet, 2017-09-26:
I am back again,
Before I continue to this issue, I did load the last version of esp32-Arduino and it looks to have an improved version of the esp32-hal-spi.c. However I still have random reset (much less than before):

```
Sending struct (12 bytes) ... ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, ⸮⸮⸮⸮⸮⸮⸮x⸮⸮x⸮~⸮⸮⸮⸮⸮⸮⸮x⸮⸮xx⸮xx⸮⸮⸮⸮⸮f⸮⸮⸮怘⸮⸮⸮⸮⸮x⸮⸮xx⸮xx⸮⸮⸮⸮⸮⸮f⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮ ⸮⸮⸮⸮⸮f⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮ ⸮⸮⸮⸮⸮f⸮⸮⸮⸮⸮⸮⸮⸮f⸮⸮`⸮⸮f ⸮⸮⸮⸮x⸮⸮⸮⸮
Transmitting at 433 Mhz...
Sending struct (12 bytes) ...  nothing...
```
I don't know it this one is related to the same problem of SPI running in interrupt mode.
On my side I will try to modify the interrupt routine to move the SPI transfer handling to a specific procedure and see if this is the right way to proceed (I agree that too much data handling in the interrupt routine is not an ideal configuration).
Robert

rrobinet, 2017-10-03:
@me-no-dev 
I have posted on https://github.com/rrobinet/RFM69X_Library  an updated version tested with WeMoS LOLIN32, expecting to be compatible with all ESP-32 processors.
I use the letter 'X' as the ESP-32 extension
The RFM69X is just a modification of the last RFM69 library (july 2014).
All SPI transactions are moved from the `interruptHandler` routine to the `interruptHandling` function which is activated for each `receiveDone`.
See also #https://github.com/LowPowerLab/RFM69/issues/81
Robert

closed 2017-10-03 by rrobinet

me-no-dev, 2017-10-04:
good stuff :) maybe we can figure an easy way out for such situation in the future... i’ll think about it :)

tferrin, 2017-12-22:
The root cause of this issue is trying to fit the hardware abstraction layer of the relatively simple AVR processor onto the much more complex ESP32 processor architecture.  The ESP32 has two cores that share a common memory (symmetric multiprocessing -- SMP). Each core executes code independently from the other and each core can process hardware interrupts independently.  To handle this complexity Espressif uses the FreeRTOS operating system where the fundamental "processing unit" is a task and there's a scheduler that executes tasks based on their priority.  This is a major change to the traditional AVR hardware abstraction layer.  For example, calls to interrupts()/noInterrupts() on the ESP32 currently do nothing because there is no easy way to do the right thing in all cases. So if you look at RFM69.cpp, it is using interrupts()/noInterrupts() to block stuff from happening during critical code sections, but on the ESP32 nothing is getting blocked and this leads to the various errors noted above.
The ultimate fix here is to treat the different functional code blocks in RFM69.cpp as tasks. For example, if a receiveDone interrupt needs to send an ACK then the code needs to queue a task whose job it is the send the ACK. Queueing the task can happen during interrupt processing, but actually sending the ACK needs to happen after interrupt processing has completed. Also critical sections of code that should not be interrupted or preempted by another task need to use mutual exclusion (mutex) locking functions. FreeRTOS supports all of this functionality and it just needs to be incorporated into RFM69.cpp.  I'm working on a modified version of RFM69.cpp for the ESP32 that I'll post as a pull request after I do a bunch more testing.

rrobinet, 2017-12-22:
Tom,

I am happy to read that you are taking care of this issue, because as I said above RFM69 transceivers are widely used for House automation and I think a RFM69 shield for ESP32 shall really be appreciated by hobbyists.
My opinion is that adding RF capabilities (probably LoRa in the future) to the ESP32 will ensure a full coverage to control remote devices for Machine to Machine data exchanges.
Now I am not a programming guru nor a microprocessor specialist  and my contribution was just to find a workaround to the issue above, however I must say that I am not  quite satisfied about my workaround, so I expect that you solution will be the real one.
Thanks to take care about this issue, you may count on me for testing the new library.
Robert

tferrin, 2018-01-03:
Thanks @Vorms. My version of RFM69.cpp appears to be working well now, although I still want to do more testing. I will be ordering more Moteino modules so that I can test with a larger network of radios all sending to my gateway simultaneously. And I plan to add one or two additional devices to the SPI bus on the gateway and having those running at the same time as I/O to the radio. If that testing all goes well I'll post a pull request over in the LowPowerLab repository with a pointer here.

Vorms, 2018-01-03:
Many thanks for your help.

I am a bit confused with GPIO pinout and SPI bus so if you have a link I
will be very intersted for..

For me I worked on a modified version with the ESP8266 Wemos d1 mini.
I had different problem and a subtile one was the noise..
It's appear that there is a lot of nice near my computer.
I have a gateway and an other device.

I lot of time was spend in the function canSend():

I do a script to display the level of reception when no emeter run and I
replace the constant CSMA_LIMIT with a variable.
I add a function to define the noise level.

If you are interested I can send my source.

Best regards

Thierry Vorms

tferrin, 2018-01-03:
I'm not experiencing any of those kinds of problems and my hardware is also completely different from yours. Also my goal was to make the fewest changes to the RFM69 library necessary to get it working on the ESP32, and those changes have only involved the logic for handling interrupts from the radio module and how SPI transactions are handled.

Vorms, 2018-01-04:
Hello Tom,
Thank you again for your mail.
Please keep me in touch if you discover some misfunctions when adding
partner in the SPI bus.

Thanks again,

Best regards

Thierry

Vorms, 2018-01-05:
Hello,
I try your library with ESP32 Lolin and... it's work perfectly!
The noise is less than with ESP8266.

It's just work and I want to thanks a lot for your work !

Best regards
Thierry

2018-01-03 21:57 GMT-05:00 Thierry Vorms <vormsty@gmail.com>:

> Hello Tom,
> Thank you again for your mail.
> Please keep me in touch if you discover some misfunctions when adding
> partner in the SPI bus.
>
> Thanks again,
>
> Best regards
>
> Thierry
>
>
>> I'm not experiencing any of those kinds of problems and my hardware is
>> also completely different from yours. Also my goal was to make the fewest
>> changes to the RFM69 library necessary to get it working on the ESP32, and
>> those changes have only involved the logic for handling interrupts from the
>> radio module and how SPI transactions are handled.
>>
>>
>
>
>
> Thierry Vorms
> Québec Canada

rrobinet, 2018-01-05:
@Vorms Hello Thierry 
Can you be more precise about the library you are using?
Robert

Vorms, 2018-01-06:
Hello,
I use the Arduino ide and developp my own library for my project.
I think you need the version of the Arduino library for ESP32, but I don't
know where to find it.

Many thanks for your help and please keep me in touch!

Best regards

Thierry

rrobinet, 2018-01-06:
Sorry Thierry  @Vorms ,
I thought you was talking about RFM69 library on ESP32, the one I have modified ( https://github.com/rrobinet/RFM69X_Library ) or a test version of Tom @tferrin.
For the ESP32 one I use https://github.com/espressif/arduino-esp32 which is regularly updated and improved
Robert

euquiq, 2018-11-07:
Hi guys! By any chance, is there a way of getting your RFM69 library version for ESP32, @tferrin  ? Until now, I've been using the one that @rrobinet forked from the original one, but I am encountering some problems that I think are related to SPI and interrupts ... I just posted it as an Issue, in chance that Rob was interested on it:

https://github.com/rrobinet/RFM69X_Library/issues/1

thanks,

Enrique

rrobinet, 2018-11-07:
Enrique,
See my reply at 
https://github.com/rrobinet/RFM69X_Library/issues/1
Robert

jLynx, 2022-07-27:
Hey @rrobinet I was trying to take a look at your fork but its no longer available. Have you got any update around this?

rrobinet, 2022-07-29:
@jLynx 
Well this code was obsoleted by new versions of the RFM69 library which supports RFM69 on ESP32 (need of course  to connect the SPI interface according to the ESP32).
Basically it corrects the way the Interrupt is configured for ESP8266 and a ESP32 and the management of the interrupt is not done in the interrupt routine itself  has it was originally done. I expect always to improve the code to fit the ESP32 architecture by moving the receive routine and interrupt  into CPU0 for freeing  CPU1 for customized code but this requires a better knowledge than I have now of the ESP32 architecture