espressif/arduino-esp32 #625: Change Arduino SDK

urbanze, 2017-09-11:
Hi! I using idf-component to use this mcu (to change sdk configs in arduino ide) and I can't find how to read temperature sensor. How can I do this ? 

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 DoIt Nodemcu
Core Installation/update date:			?11/jul/2017?
IDE name:							IDF component with Arduino boot
Flash Frequency:					80Mhz
Upload Speed:						115200

copercini, 2017-09-16:
@urbanze what temperature sensor?

urbanze, 2017-09-16:
[image]

This sensor, from technical_reference datasheet.

copercini, 2017-09-16:
There are some examples from @pcbreflux here: https://github.com/pcbreflux/espressif/blob/master/esp32/arduino/sketchbook/ESP32_int_temp_sensor/ESP32_int_temp_sensor.ino

Also a video explaining the things: https://www.youtube.com/watch?v=d57LKkHCVgI

urbanze, 2017-09-16:
Oh, the secret of temperature sensor is "extern "C" "........ I forgot this -_-. Thank you for reply, see you later :D

closed 2017-09-16 by urbanze

copercini, 2017-09-18:
See you =)