espressif/arduino-esp32 #626: Can't install toolchain on a Mac

mundodisco8, 2017-09-11:
### Hardware:
Board:							Adafruit Huzzah32, but irrelevant.
Core Installation/update date:			11/sept/2017?
IDE name:							Arduino IDE 1.8.4
Flash Frequency:					80MHz
Upload Speed:						921600

### Description:
I can't compile my code because the "get.py" script doesn't generate the xtensa-esp32-elf" folder.
I have a Mac running the latest OS version, with Python 2.7. When I follow the instructions, there are no errors and everything seems to be fine. When I open the Arduino IDE, I can see a list of ESP32 boards that weren't there before, but compiling anything fails as the IDE can't find the xtensa g++ compiler. When I follow the path, I can see some folders on Hardware/Tools/... directory, but not the "xtensa-esp32-elf" one. I tried again on another Mac computer, that is also running the latest Mac Os version and python 2.7, and it works there. I tried deleting the whole git repository and reinstalling the Arduino IDE, but the result is the same. Any ideas why I'm not getting the toolchain, and why is the script not complaining about any errors? Is there any way to run it verbose?

### Debug Messages:
```
fork/exec /Users/....../Hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: no such file or directory
Error compiling for board ESP32 dev Module
```

[image]

And the ouput from the installation script:

[image]

me-no-dev, 2017-09-11:
how did you run get.py?

mundodisco8, 2017-09-11:
Both copying and pasting the code in the page and by itself with python
get.py. both of them outputted the same message.

I tried in another Mac computer, where it worked. Are there any requisites
to run the code?

Raienryu97, 2017-09-12:
@me-no-dev @mundodisco8 
I see the errors in the console also say 
`ImportError: No module named serial`
`ImportError: No module named SCons.Script`
In linux we use
```Bash
wget https://bootstrap.pypa.io/get-pip.py && \
sudo python get-pip.py && \
sudo pip install pyserial
```
to install pyserial. Maybe your mac needs it too along with SCons.Script?

me-no-dev, 2017-09-12:
@Raienryu97 unrelated in Arduino IDE ;) get.py does not require those pip modules. It's bad permissions and they need to be fixed. He is either not the owner of the folder or something else

mundodisco8, 2017-09-12:
I just checked and I have read and write permissions on that folder, is
there anything else I could check?

tobozo, 2017-09-12:
the owner (shouldn't be root)

fuseps, 2017-09-12:
I'm having the same issue. I found on my mac it is installing everything in the correct place except for the esp folder. For some reason it's dropping that in a folder named esp on my root, not in docs, and there is another folder in the root called esp32, which is empty. when I copy paste the contents of esp into the relevant folder that the arduino IDE is looking for, I get closer, but now get the following issue

`fork/exec /Users/USERNAME/Documents/Arduino/hardware/espressif/esp32/tools/esptool: no such file or directory
Error compiling for board ESP32 Dev Module.
`

Trying to run the esptool.py brings up this error in the terminal

`Traceback (most recent call last):
  File "esptool.py", line 25, in <module>
    import serial
ImportError: No module named serial`

help?

me-no-dev, 2017-09-12:
Run get.py it will install it

fuseps, 2017-09-12:
Tried that, it doesn't work. What should it be creating/where could it have ended up?

fuseps, 2017-09-12:
apologies to be clear, the esptool.py is in the directory, so I'm not sure what the issue is

me-no-dev, 2017-09-12:
[image]

fuseps, 2017-09-13:
Ok so I have those files in the right place, but arduino IDE just isn't finding them when trying to compile. I don't understand why

me-no-dev, 2017-09-13:
but you said you are seeing the boards in the menu... is it possible that you have it installed somewhere else as well?

mundodisco8, 2017-09-13:
So the thing is that the script is not installing everything were it should. Yes, the boards are on the boards menu, and there are some files in '''/Documents/Arduino/Hardware/espressif''', but the boards appear on the IDE because I cloned the git repo, and that copies the '''boards.txt''' file. The script installed some files in the user home, under a folder called '''ESP'''. Copying those files to the expected location ('''/Documents/Arduino/hardware/espressif/esp32/tools''') allowed me to at least compile, but I still get the other error.

So to sum up: ran the script on a Mac, everything worked. Ran it again in a different mac, didn't work, couldn't compile, although boards appear on the board manager in the IDE. Run the script multiple times, same issue. Found out that the files were somewhere else, copied them to the expected path, can compile but get error about "no esptool".

fuseps, 2017-09-13:
[image]
[image]

me-no-dev, 2017-09-13:
esptool is missing in your screenshot... I have no clue why. Can you show me the log from get.py and also show what is in tools/dist folder

fuseps, 2017-09-13:
dist is empty

`sudo python get.py
Password:
System: Darwin, Info: Darwin-16.7.0-x86_64-i386-32bit
Platform: i386-apple-darwin
Done
`

mundodisco8, 2017-09-13:
Is there any way to get a verbose output for get.py? Does it store a log
somewhere?

fuseps, 2017-09-14:
I don’t see one anywhere in my system…

Paul Sohi
Fusion 360 Evangelist/Product Designer

M: +447971 823348

Twitter<http://twitter.com/fuseps>
Facebook <http://facebook.com/fuseps>
[ignature_992800167]

me-no-dev, 2017-09-14:
aaaaa why sudo? sudo installs it as root and you get bad permissions after...

fuseps, 2017-09-14:
it wasn't doing anything without sudo at all, and so I did it with sudo too. I'll run it again as standard but it won't change anything, I've tried it before

fuseps, 2017-09-14:
just tried it without sudo 

`tools paulSohilocal$ python get.py
System: Darwin, Info: Darwin-16.7.0-x86_64-i386-32bit
Platform: i386-apple-darwin
Done`

nothing's changed...

me-no-dev, 2017-09-14:
what do you see inside your tools/dist folder? there should be some archives there

fuseps, 2017-09-14:
it was empty. I deleted everything and @mundodisco8 provided me with everything from his working version as a zip and now I can compile. I think maybe something is up with the git clone? I am gonna try the the instruction from git on a different mac later and report back!

keefo149, 2017-11-13:
I'm having the same issue, did you ever find out what the issue was?
If so could you please post it. 
cheers

mundodisco8, 2017-11-13:
never found out what the problem was. We were having the issue on a mac, but it worked on a Windows machine. Tried on a different Mac, and it worked, so we copied the files over to the problematic Mac.

closed 2018-04-19 by everslick