espressif/arduino-esp32 #630: WiFi & Interrupt Causing Execution to Halt

Diablofan, 2017-09-12:
### Hardware:
Board: Adafruit Feather Huzzah32
Core Installation/update date: 2017/09/11
IDE name: Arduino IDE
Flash Frequency: 80MHz
Upload Speed: 115200

### Description:
While working on a project requiring interrupts and wifi, I came across an interesting bug. If you set pin 5 to INPUT mode and then attach an interrupt to it, any calls to the WiFi object will cause the board to stop execution without any sort of debug output even on the verbose logging setting. This appears to only happen on INPUT mode for Pin 5, as INPUT_PULLUP causes execution to happen like normal.

Inserting a debug line after the call to `WiFi.begin("Demo", "");` will show that execution stops somewhere after `WiFi.begin` is called.

Below is a sketch that replicates the setup along with expected and actual output.

### Sketch:
```cpp
#include <WiFi.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  constexpr int pin = 5;
  pinMode(pin, INPUT);
  attachInterrupt(digitalPinToInterrupt(pin), [](){/* Dummy Callback */}, RISING);
  Serial.println("Connecting to WiFi");
  WiFi.begin("Demo", "");
  while(WiFi.status() != WL_CONNECTED) {
    delay(100);
  }
}

void loop() { Serial.println("This will not get printed."); }
```
### Debug Messages:
```

```
### Expected Output
```
Connecting to WiFi
This will not get printed.
This will not get printed.
This will not get printed.
...
```
### Actual Output
```
Connecting to WiFi
```

copercini, 2017-09-12:
>   pinMode(pin, INPUT);
  attachInterrupt(digitalPinToInterrupt(pin), [](){/* Dummy Callback */}, RISING);

Or you declare it as input or as attachInterrupt, but not both ;)

me-no-dev, 2017-09-18:
```[](){/* Dummy Callback */}``` will not work ;) have a function and reference it instead

Diablofan, 2017-09-18:
@me-no-dev I can resoundingly tell you that, yes, `[](){}` [does in fact work](https://godbolt.org/g/3mCr82). For extra proof, here is a [screenshot](https://i.imgur.com/U7QvJXM.png) of the arduino ide compiling it, the ide obtained via the Windows Store.

The reason this works is that the bundled version of avr-gcc has C++ 11 *language* features such as lambdas and basic constexpr enabled (the -std=c++0x flag). So, with that in mind, lets take a look at the lambda's defined in both the sample code above and my screenshot/compiler explorer links. Since the lambda's capture group, the square brackets, are empty, that allows the compiler to promote the lambda to a plain old function pointer.

So long as our lambda's signature then matches what we're looking to pass through, in the case of both code samples, a void function with no arguments, then we can use a lambda in place of regular function instead, as I have just shown.

But no, go ahead and try to detract from the issue at hand with my use of C++ 11 language features. I'm pretty sure you didn't notice the constexpr I threw in the sample code either. Guess what, that [compiles cleanly as well](https://i.imgur.com/DpIhG9M.png).

me-no-dev, 2017-09-18:
How about you try ```WiFi.begin("Demo", NULL);``` then. I'm stuck on ESP8266 mindset for lambdas, so you might be correct. Another thing I can offer you is to enable Debug (if you have selected ESP32 dev board it has a menu). That will tell you what events are received

Diablofan, 2017-09-18:
I'm going to close the issue. Trying it once again with `WiFi.begin("Demo", "")`, `WiFi.begin("Demo");` and with both the Adafruit Feather Huzzah32 and the ESP32 Dev Module options. It appears that somewhere between 0bce98e72ccbd2f7b38a23c79eefb7a36d3baa74 and 79ee46ead23fd91669ac4fd60b8ac8a8e482dd90 (most recent version I pulled) something fixed the issue.

By the way, you mentioned/implied that the ESP8266 doesn't like lambdas, [about that...](https://i.imgur.com/P60PcFG.png)

closed 2017-09-18 by Diablofan

me-no-dev, 2017-09-19:
I did not say that it does not like them :) we use lambdas on many places :)