espressif/arduino-esp32 #632: Blink example fails to build when arduino-esp32 added as component

csann, 2017-09-13:
### Hardware:
Board:							        ESP-WROVER-32 Dev Board
Core Installation/update date:			jul/2017
IDE name:							IDF component
Flash Frequency:					        40Mhz
Upload Speed:						115200

### Description:
When I build the blink example project (~/esp/esp-idf/examples/get-started/blink) using IDF, it builds fine by running make.

However when I add a components folder and then add arduino-esp32 to it using the instructions from this site, it does not build. Note that I have not changed the blink project at all except by only adding the arduino component. The error is:

CXX build/arduino/libraries/SPIFFS/src//SPIFFS.o
/Users/clark/esp/apps/blink/components/arduino/libraries/SPIFFS/src//SPIFFS.cpp:21:24: fatal error: esp_spiffs.h: No such file or directory
compilation terminated.
make[1]: *** [libraries/SPIFFS/src//SPIFFS.o] Error 1
make: *** [component-arduino-build] Error 2

Also note that if I place the file in the SPIFF folder, the missing file will be found but make then immediately errors on another missing file. Maybe a path is not set correctly somewhere??

### Sketch:
 No sketch prepared. The problem is the arduino component won't build due to a missing file. 

### Debug Messages:

CXX build/arduino/libraries/SPIFFS/src//SPIFFS.o
/Users/clark/esp/apps/blink/components/arduino/libraries/SPIFFS/src//SPIFFS.cpp:21:24: fatal error: esp_spiffs.h: No such file or directory
compilation terminated.
make[1]: *** [libraries/SPIFFS/src//SPIFFS.o] Error 1
make: *** [component-arduino-build] Error 2

Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
Not using Arduino IDE, instead I am building using IDF.

me-no-dev, 2017-09-13:
This should do :)
```bash
cd $IDF_PATH
git checkout master
git pull
git submodule update --init
```

roamingthings, 2017-09-13:
Thank you very much!
I can confirm that this helped in my case.

csann, 2017-09-13:
Well, I no longer get the missing esp-spiffs.h error message. Thank you for that!

But unfortunately, now it complains about a missing bta_api.h file. I have done make clean, make menuconfig, then make.

roamingthings, 2017-09-13:
You have to do the following steps:
make clean
make menuconfig
Activate Bluetooth
make

Now it should work

csann, 2017-09-14:
Thank you all for your help. 

I can confirm that me-no-dev's instructions fixed the spiffs problem. I can also confirm the roamingthings instructions fixed the bluetooth problem. After I set Bluetooth on in menuconfig, I was able to make. After that I turned Bluetooth off, and I was still able to make. Seems odd to me that you should have to turn on a feature you don't want in order to build, but I suppose this is just a workaround and that  espressif will get this fixed. 

Thanks again everyone!

closed 2017-09-14 by csann