espressif/arduino-esp32 #633: Compilation error: fatal error: esp_sleep.h: No such file or directory

kidproquo, 2017-09-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:						IDF component
Flash Frequency:					40Mhz
Upload Speed:						115200

OS: Windows 10

### Description:
Hello, followed the instructions to use as a component of ESP-IDF. Trying to setup with the esp-idf-template (https://github.com/espressif/esp-idf-template)

`make menuconfig` completes successfully.

However, running `make` gives the following error:

```
wirop_arduino/components/arduino/cores/esp32/Esp.cpp:23:23: fatal error: esp_sleep.h: No such file or directory
compilation terminated.
```

copercini, 2017-09-13:
esp_deep_sleep.h was [deprecated in IDF](https://github.com/espressif/esp-idf/commit/d2acf1ce77a2883e519ef57da5ff3e45bb0d6bbe#diff-d559a707898ac2219e52770f63377ce0R26) in favor of esp_sleep.h which has [already been updated in Arduino](https://github.com/espressif/arduino-esp32/commit/39fb8c30440a4abd5fe0e2c87609ba6798ae8013)

So my guess is that you're running and updated version of Arduino core on old IDF...

kidproquo, 2017-09-13:
I have checked out the latest release v2.1 on the IDF. And, it's not `esp_deep_sleep.h` that can't be found. It's `esp_sleep.h`, which actually is in `arduino\tools\sdk\include\esp32\esp_sleep.h`. Maybe I need to update the include path?

me-no-dev, 2017-09-13:
not release ;) latest GIT is what you need

kidproquo, 2017-09-13:
No luck. Switched to master branch in esp-idf, did `git pull` and `git submodule update --init --recursive`. New error message:

```
wirop_arduino/components/arduino/libraries/SimpleBLE/src/SimpleBLE.cpp:19:21: fatal error: bta_api.h: No such file or directory
```

me-no-dev, 2017-09-13:
make sure you make clean :)

kidproquo, 2017-09-13:
Started a new repo, based on esp-idf-template. Followed the instructions, menuconfig, followed by make. Same error:
```
/test/components/arduino/libraries/SimpleBLE/src/SimpleBLE.cpp:19:21: fatal error: bta_api.h: No such file or directory
compilation terminated.
```

me-no-dev, 2017-09-13:
That is rather strange. [here](https://github.com/espressif/esp-idf/blob/master/components/bt/bluedroid/bta/include/bta_api.h) is the header in esp-idf and I just compiled it under ArduinoIDE

kidproquo, 2017-09-13:
I know. I have the header as well. It's just not seeing it. I think it's an include path issue.

me-no-dev, 2017-09-13:
Still interesting... the config file with includes have not been touched in 6 months. And that include dir is the first in the list: https://github.com/espressif/esp-idf/blob/master/components/bt/component.mk#L25

csann, 2017-09-13:
I'm having essentially the same issue that I reported in issue #632. When I add Arduino core to the blink project, I get an error that it cannot find esp-spiffs.h. Just adding Arduino core as a component causes the error. I will try it again later today using the esp-idf-template, but I bet I'll get the same result. I can't explain why we are all getting different missing h files though. Hoping I was only missing the one file, I copied that file into the SPIFF folder. Then it errored on another missing file...the same BLE file mentioned by kidproquo above. It sure seems like a path is not set correctly or maybe the folder structure is wrong.

me-no-dev, 2017-09-13:
for esp-spiffs.h you probably need to run ```git submodule update --init``` in IDF's folder

timbartsch, 2017-09-15:
Thanks. `git checkout master && git pull && git submodule update --init` did the trick for me.

Had also the following issue:
`/Users/tim/esp/esp-idf/components/spiffs/./esp_spiffs.c:16:20: fatal error: spiffs.h: No such file or directory`

dadosch, 2017-09-16:
I'm having this problem with esp_sleep.h as well. I can't change to esp-idf master branch, because bluetooth is not functional in there currently [https://github.com/espressif/esp-idf/issues/857](https://github.com/espressif/esp-idf/issues/857).

How can I get the arduino core compatible with esp-idf release V2.1?

me-no-dev, 2017-09-18:
```esp_sleep.h``` is the new name of ```esp_deep_sleep.h``` and it also removes ```_deep``` from most functions, so you can maybe add ```esp_sleep.h``` file in your project and make that include ```esp_deep_sleep.h``` and then use define to translate the renamed methods. Example:
```cpp
#define esp_sleep_enable_ulp_wakeup(a) esp_deep_sleep_enable_ulp_wakeup(a)
```

kidproquo, 2017-09-20:
Can anyone provide any help? How do I specify an include path to see if the problem goes away?

me-no-dev, 2017-09-20:
make sure you are not just running make, but actually ```make clean``` in that project. Maybe you forgot so some old config is lurking there (you can also just delete the ```build``` folder in the project root)

kidproquo, 2017-09-20:
I started with a fresh copy of the esp-idf-template and followed the instructions. So nothing to clean.

I tried building without the arduino component and that works. So, I think the arduino component is somehow messing up the include paths.

me-no-dev, 2017-09-20:
not possible ;) many people are using it that way (me included) so it is really strange what is going on

pathob, 2017-09-21:
I came to this issue, because I also had the problem above. Now, I updated esp-idf to 9a26296 (as mentioned in ba929be and also tried the master branch (and in both cases I updated and initialized the submodules). The error above is gone but now I'm getting:

```bash
CXX build/arduino/libraries/SimpleBLE/src//SimpleBLE.o
.../arduino/components/arduino/libraries/SimpleBLE/src//SimpleBLE.cpp:19:21: fatal error: bta_api.h: No such file or directory
```

> not possible ;)

How can you be sure? I don't see any automated tests for compiling as a esp-idf project component. And I don't see any instructions which esp-idf release (or version) should be used.

Please also do a `make clean` and if it works, tell me your esp-idf and arduino-esp32 commits you tried with. Thank you!

me-no-dev, 2017-09-21:
do you have BT enabled in menuconfig?

me-no-dev, 2017-09-21:
BTW I usually run the very latest IDF ;)

pathob, 2017-09-21:
> do you have BT enabled in menuconfig?

I didn't, now I enabled it, cleaned and it works. But it's not documented, right? Thanks...

kidproquo, 2017-09-21:
Bingo! Enabled BT and it worked. Now I am getting a new error:

```
fatal error: esp_bt_defs.h: No such file or directory
compilation terminated.
```

csann, 2017-09-21:
I have found that after you have enabled BT and gotten it to build, you can then disable it. Very goofy.

me-no-dev, 2017-09-21:
i'll put a guard to not compile BT code unless enabled. Should fix that issue.

> fatal error: esp_bt_defs.h: No such file or directory
compilation terminated.

IDF error. Something else ;)

> Thanks...

You are welcome...

pathob, 2017-09-21:
@kidproquo 

`make clean` once again!

me-no-dev, 2017-09-21:
For all BT disablers (and non-enablers): https://github.com/espressif/arduino-esp32/commit/c66a020e8067f657e730d018700e7de5b23175d4

closed 2017-09-21 by me-no-dev

csann, 2017-09-21:
Thank you for the improvement! One less bug...more happiness.