espressif/arduino-esp32 #634: No HSPI communication possible with Arduino-ESP32 now

mgo-tec, 2017-09-13:
### Hardware:
Board:							ESP32-DevKitC
Core Installation/update date:			13/sep/2017
IDE name:							Arduino IDE 1.8.3
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
In the Arduino core for ESP32 which is currently released, HSPI communication has become impossible.
The one before it moves without problems.
why?

jestebanes, 2017-09-13:
Is the same for me. I need to back to version 10/9/2017

I've got an external flash connected to HSPI and I lost it with the last update to IDF 9a26296

me-no-dev, 2017-09-13:
OK, I might have a clue ;)

me-no-dev, 2017-09-13:
Can you guys pull the latest master?

mgo-tec, 2017-09-13:
Great!
It worked without problems.
Thank you very much for your quick response.

closed 2017-09-13 by mgo-tec

jestebanes, 2017-09-13:
Ok! it works again
Thank you @me-no-dev

me-no-dev, 2017-09-13:
Thank you guys for reporting it :D