espressif/arduino-esp32 #636: I need help making an Arduino app using IDF

csann, 2017-09-16:
### Hardware:
Board:							   ESP-WROVER-32 
Core Installation/update date:		   11/jul/2017
IDE name:					           IDF component
Flash Frequency:					   40Mhz
Upload Speed:					   115200

### Description:
With a lot of help from @me-no-dev and others I have been able to squash the errors preventing me from building Arduino Core. Now I am working on getting make to make the rest of my program. I am getting a number of error messages that I do not understand. 

Here are the error messages I get after running make.

```
make
LD build/app-template.elf
Somewhere/absStepperTest/build/main/libmain.a(ABS_Stepper.o):(.bss.currentAngle+0x0): multiple definition of `currentAngle'
Somewhere/absStepperTest/build/main/libmain.a(app_main.o):(.bss.currentAngle+0x0): first defined here
Somewhere/absStepperTest/build/main/libmain.a(ABS_Stepper.o):(.bss.dir+0x0): multiple definition of `dir'
Somewhere/absStepperTest/build/main/libmain.a(app_main.o):(.bss.dir+0x0): first defined here
Somewhere/absStepperTest/build/main/libmain.a(ABS_Stepper.o):(.bss.numSteps+0x0): multiple definition of `numSteps'
Somewhere/absStepperTest/build/main/libmain.a(app_main.o):(.bss.numSteps+0x0): first defined here
Somewhere/absStepperTest/build/main/libmain.a(Adafruit_MotorShield.o):(.literal._ZN20Adafruit_MotorShield5beginEt+0x8): undefined reference to `Adafruit_MS_PWMServoDriver::begin()'
Somewhere/absStepperTest/build/main/libmain.a(Adafruit_MotorShield.o):(.literal._ZN20Adafruit_MotorShield5beginEt+0xc): undefined reference to `Adafruit_MS_PWMServoDriver::setPWMFreq(float)'
Somewhere/absStepperTest/build/main/libmain.a(Adafruit_MotorShield.o):(.literal._ZN20Adafruit_MotorShield5beginEt+0x10): undefined reference to `Adafruit_MS_PWMServoDriver::setPWM(unsigned char, unsigned short, unsigned short)'
Somewhere/absStepperTest/build/main/libmain.a(Adafruit_MotorShield.o):(.literal._ZN20Adafruit_MotorShieldC2Eh+0x0): undefined reference to `Adafruit_MS_PWMServoDriver::Adafruit_MS_PWMServoDriver(unsigned char)'
Somewhere/absStepperTest/build/main/libmain.a(Adafruit_MotorShield.o): In function `Adafruit_MotorShield::begin(unsigned short)':
Somewhere/absStepperTest/main/./Adafruit_MotorShield.cpp:151: undefined reference to `Adafruit_MS_PWMServoDriver::begin()'
Somewhere/absStepperTest/main/./Adafruit_MotorShield.cpp:151: undefined reference to `Adafruit_MS_PWMServoDriver::setPWMFreq(float)'
Somewhere/absStepperTest/main/./Adafruit_MotorShield.cpp:151: undefined reference to `Adafruit_MS_PWMServoDriver::setPWM(unsigned char, unsigned short, unsigned short)'
Somewhere/absStepperTest/build/main/libmain.a(Adafruit_MotorShield.o): In function `Adafruit_MotorShield::setPWM(unsigned char, unsigned short)':
Somewhere/absStepperTest/main/./Adafruit_MotorShield.cpp:151: undefined reference to `Adafruit_MS_PWMServoDriver::setPWM(unsigned char, unsigned short, unsigned short)'
Somewhere/absStepperTest/main/./Adafruit_MotorShield.cpp:151: undefined reference to `Adafruit_MS_PWMServoDriver::setPWM(unsigned char, unsigned short, unsigned short)'
Somewhere/absStepperTest/build/main/libmain.a(Adafruit_MotorShield.o): In function `Adafruit_MotorShield::setPin(unsigned char, bool)':
Somewhere/absStepperTest/main/./Adafruit_MotorShield.cpp:151: undefined reference to `Adafruit_MS_PWMServoDriver::setPWM(unsigned char, unsigned short, unsigned short)'
Somewhere/absStepperTest/main/./Adafruit_MotorShield.cpp:151: undefined reference to `Adafruit_MS_PWMServoDriver::setPWM(unsigned char, unsigned short, unsigned short)'
Somewhere/absStepperTest/build/main/libmain.a(Adafruit_MotorShield.o): In function `Adafruit_MotorShield::Adafruit_MotorShield(unsigned char)':
Somewhere/absStepperTest/main/./Adafruit_MotorShield.cpp:151: undefined reference to `Adafruit_MS_PWMServoDriver::Adafruit_MS_PWMServoDriver(unsigned char)'
Somewhere/absStepperTest/main/./Adafruit_MotorShield.cpp:151: undefined reference to `Adafruit_MS_PWMServoDriver::Adafruit_MS_PWMServoDriver(unsigned char)'
collect2: error: ld returned 1 exit status
make: *** [Somewhere/absStepperTest/build/app-template.elf] Error 1
```
First of all, I can't figure out why I am getting the three "multiple definition of ..." errors. I have carefully checked my two class files and they do not have duplicates. I have even searched the .h and .cpp files....no duplicates. 

Next I can't fathom why I am getting so many "undefined reference to..." errors. Those errors are in Adafruit created classes and I know they build on Arduinos.  

HOWEVER, I am new to C++ so I am hoping someone will look at this and say "Hey dummy, this is why this happens...".

me-no-dev, 2017-09-18:
maybe you are trying to include CPP files in a C file?

csann, 2017-09-18:
I don't think so because I don't have any C files. I only have C++ files (app_main.cpp, Adafruit_MotorShield.h, Adafruit_MotorShield.cpp, ABS_Stepper.h, ABS_Stepper.cpp). 

I do have these declarations in the ABS_Stepper.h. 

extern "C" void taskStepperWrapper(void* parm);
extern "C" void homeDetectedWrapper(void* parm);

And I have the implementation in ABS_Stepper.cpp...

extern "C" void taskStepperWrapper(void* parm) {  // wrapper is task pointer in freeRTOS task
    (static_cast<ABS_StepperMotor*>(parm))->taskStepper();
}

extern "C" void homeDetectedWrapper(void* parm) {  // wrapper is task pointer to ISR
    (static_cast<ABS_StepperMotor*>(parm))->homeSwitchDetected();
}

These are wrappers to allow C++ methods to be run as freeRTOS tasks and as ISRs. These are the only C parts of my code. Could these be causing the problem?

closed 2018-04-19 by everslick