espressif/arduino-esp32 #637: Anyone have any success with using Eclipse with Arduino-esp32?

csann, 2017-09-16:
### Hardware:
Board:							ESP-WROVER-32 Dev Module
Core Installation/update date:		11/jul/2017
IDE name:						IDF component
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:

I am developing an app using Arduino-esp32, some Adafruit Stepper classes, and a class I developed which overrides some of the Adafruit classes. I am currently in the process of getting it to build using IDF, but having some problems.

In the mean time, I would like to start developing using Eclipse. I have setup the IDF toolchain in Eclipse so I am able to develop and debug using JTAG, an IDF project that does not use Arduino-esp32. I would like to extend Eclipse to use Arduino-esp32.

Can I somehow modify the Eclipse configuration to use the Arduino component I have already created and built?

Or must I use make or PlatformIO? If I must use one of these, can anyone give me a recommendation as to which one is easier to use and can support JTAG debugging?

me-no-dev, 2017-09-18:
I am using "sloeber" which is Eclipse with some Arduino addons. I have basically IDF projects and pure Arduino projects inside. Should do what you are asking ;)

csann, 2017-09-18:
Thanks for this, MND!

closed 2017-09-22 by copercini