espressif/arduino-esp32 #639: SPIFFS does not install with update to repositor

bill-orange, 2017-09-16:
### Hardware:
Board:	Lolin32						
Core Installation/update date:	9/16/12		
IDE name:							?Arduino IDE? 1.8.4
Flash Frequency:					80 Mhz
Upload Speed:						115200

### Description:
I updated my arduino-esp32 installation without difficulty many time following the Windows instructions.  If I run git gui nothing shows shows needing updating at present.  This makes sense since I just updates it one or two days ago.  However, the SPIFFS folder is not found with my other ESP32 libraries.  If I copy and past it in manually to the appropriate folder in hardware, I am still in trouble because the RTOS library is not found.  It is reported to not be a valid library when I let the arduino IDE try to install it..  

Obviously there is something that I am not understanding.  I would very much like to use SPIFFS on the EPS32 platform.

lbernstone, 2017-09-17:
It is located at <arduino>/hardware/espressif/esp32/libraries/SPIFFS on my system, and I can browse to it in the github master. (the esp32 directory is my git root). It sounds like your local git repo did not update properly.

bill-orange, 2017-09-17:
Yes, that is probably the case.  I can fix it (more or less) by just manually adding the SPIFFS folder into the location you mentioned (mine is the same).  That, however, does not address the missing "esp_spiffs.h" library item(s).  If you #include SPIFFS in a sketch, are you getting the error that esp_spiffs.h" is missing?

How do I fix the bad repo update or will it fix itself given a bit of time?

clowrey, 2017-09-17:
I can never seem to get my ESP32 install to update via GIT GUI on windows. It works the first time but updating never pulls new files.  So every time I just delete the whole EDP32 folder inside hardware and start from scratch and that has worked to get the latest version.

bill-orange, 2017-09-17:
Thanks, clowrey.  That's what I will do.  I am still probably left with the "esp_spiffs.h" problem.   Does anyone else get this error ( missing "esp_spiffs.h" ) when trying to #include SPIFFS?

me-no-dev, 2017-09-18:
to update on windows you can open the IDF folder, right click and open git-bash there. Then run ```git checkout master && git pull && git submodule update --init```. That should update your installation to the latest code and pull any necessary submodules (currently only SPIFFS)

bill-orange, 2017-09-18:
I have git-bash on my windows machine but not in that folder.  I ran git-bash and navigated to the esp32 master then ran the command you suggested.  The result was:
________________________________________________________________________
Already on 'master'
remote: Counting objects: 18, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 18 (delta 12), reused 18 (delta 12), pack-reused 0
Unpacking objects: 100% (18/18), done.
From https://github.com/espressif/arduino-esp32
   79ee46e..f648ad7  master     -> origin/master
There is no tracking information for the current branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.
_________________________________________________________________________
This takes us back to the local git repo did not update properly, I think?

ddchesser, 2017-09-19:
Yes that's true. I always have do the same. Delete the whole ESP32 folder and do the whole inital procedure again to be sure everything new is downloaded.

me-no-dev, 2017-09-19:
I have 3 copies of the repository on my computer. I have had the need to redownload one of them once due to some error on my side, other than that it's all good. Are you guys using forks or something?

bill-orange, 2017-09-19:
I can not speak for the others, but I am running off of the master.  I installed the repository by following the Windows instructions provided precisely.

ddchesser, 2017-09-19:
Me too, I also run of the master! I also followed axactly the installation instructions for Windows.  This always works. The "How to update to the latest code" procedure (GITGUI-fetchfrom-origin) of the existing repository show success for something new . But after finally finish get.exe from the tools-directory I never saw any changes. But this problem I had from the very beginning (about June).

Two PCs: One with Windows 10 32bit, one with Windows 10 64 bit

clowrey, 2017-09-19:
Yes that is the same thing for me. The update doesn't work with the Windows instructions on GitHub, I have to delete the folder inside arduino/hardware and start from scratch each time and then it works fine..

sansillusion, 2017-09-20:
I am on Windows 10 since June too and never had to delete and restart. updates seem to work, many things have changed since and they always seem ok on my side after updates.

me-no-dev, 2017-09-22:
yeah... why the hell it's not the same for all (BTW I have tested Win and Lin a few months back)

everslick, 2018-04-19:
Is this issue still unsolved for you? @bill-orange @clowrey @ddchesser

bill-orange, 2018-04-19:
Works fine now.  I don't remember thee resolution.

closed 2018-04-19 by everslick