espressif/arduino-esp32 #641: pinMode function not working @esp32 devmodule

Akskaypohankar, 2017-09-17:
hello guy's i am trying to hook ultrasonic sensor to esp32 programming in Arduino but it is not working things happens like this.
1) I developed code to read ultrasonic sensor in Arduino uploaded to Uno worked very well 100% perfect 
2) I made necessary changes for esp 32 and up loaded to esp WROOM 32 dev kit. but it gave me error like this
Guru Meditation Error of type LoadProhibited occurred on core 1. Exception was unhandled.
Register dump:
PC : 0x40080cf8 PS : 0x00060030 A0 : 0x800d0598 A1 : 0x3ffc77a0 
A2 : 0x00000006 A3 : 0x00000001 A4 : 0x00000020 A5 : 0x80000020 
A6 : 0x00000003 A7 : 0x00060023 A8 : 0x3f400d64 A9 : 0xdddddddd 
A10 : 0xdddddddd A11 : 0x00000060 A12 : 0x00060023 A13 : 0x3ffc3350 
A14 : 0x00000003 A15 : 0x00060023 SAR : 0x00000019 EXCCAUSE: 0x0000001c 
EXCVADDR: 0xdddddddd LBEG : 0x00000000 LEND : 0x00000000 LCOUNT : 0x00000000
Backtrace: 0x40080cf8:0x3ffc77a0 0x400d0598:0x3ffc77c0 0x400d7c46:0x3ffc77e0
CPU halted.
3) then i removed pin mode instructions from code and reuploaded, it started working but reading is always zero
can any one help me 
code used
#define trigPin_L 7 
#define echoPin_L 6
void setup() {
pinMode(trigPin_L, OUTPUT);
pinMode(echoPin_L, INPUT);
//pinMode(trigPin_R, OUTPUT);
//pinMode(echoPin_R, INPUT);
Serial.begin(115200);
}
void loop(){
long duration, distance;
digitalWrite(trigPin_L, LOW);
delayMicroseconds(2);
digitalWrite(trigPin_L, HIGH);
delayMicroseconds(10);
digitalWrite(trigPin_L, LOW);
duration = pulseIn(echoPin_L, HIGH);
distance = (duration/2) / 29.1;
Serial.print("Distance LEFT=");
Serial.println(distance);
delay(50); }

later I uploaded same code to SparkFun things(esp32board) and it worked without any error
but I need this to work on my esp 32 dev kit please help.

Raienryu97, 2017-09-17:
Variables duration and distance are defined as `long` and you are dividing them by 2 ,29.1 in `distance = (duration/2) / 29.1;` . It could be that you are actually getting an answer but since it is not of `float` type, it is just showing 0 instead of 0.xyz 
Try casting it to float or defining them as float. It might help.

Akskaypohankar, 2017-09-17:
I don't think that could be the problem because same code is working perfectly for Arduino UNO and Sparkfun esp32 things.

Raienryu97, 2017-09-17:
Could you pass your backtrace through the [Esp Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder) ? It would help for further analysis maybe.
 ( PS:  I did read your statement regarding it working on Uno and sparkfun, but thought that suggestion could be worth a shot)

Akskaypohankar, 2017-09-17:
I will try both and respond ASAP

lbernstone, 2017-09-17:
Why are you using GPIO 6 & 7?  Those are used by the flash SPI.  There are about 20 other pins to choose from :)

Akskaypohankar, 2017-09-17:
I tried on different pins it was Priti much same result.
BUT boards file and varients file in libraries, I realized I had made mistake in choosing the correct board. instead of choosing esp32 devkit I choose nodemcu32s and it solved PINMODE function problem but for ultrasonic sensor, it is giving the only first correct result only once after every reset and then all readings zero :-(

me-no-dev, 2017-09-18:
there is no reason why code would work on Sparkfun Thing and not on any other ESP32 based board, given that you have chosen the correct pins. They are all basically the same thing ;) ESP32 chip on all of them.

sansillusion, 2017-09-18:
After last Commit I recompiled my code and it did not set the selected pins HIGH when supposed to ?
Re-compiled/Flashed two more times to no avail.
After seeing this post I changed board from nodemcu-32s to node32s and it worked ?
I then changed back to nodemcu-32s and it worked, weird !
I can no longer reproduce the issue but it did occur without any changes to my code.

me-no-dev, 2017-09-18:
so you did not rebuild after updating to the latest code and just reflashed the board? The IDE might not have picked up the changes, thus flashing something wrong, but when you changed the board it actually recompiled everything and that is why it worked. ArduinoIDE is a bit weird :)

sansillusion, 2017-09-18:
I did rebuild but as you say Arduino is a bit weird sometimes, might have been some cache issue or something anyway it works !

Akskaypohankar, 2017-09-18:
Hmm after all the suggestions and few Trial and error I got these sensors to work but except any of flash pins so to use ultrasonic sensors I cannot use any of flash pins gpio6,7,8,9,10,11 and input only pin other than that they are working.

Akskaypohankar, 2017-09-18:
is there any way I can use this flash pins as GPIO pins?

me-no-dev, 2017-09-18:
not really ;) you might be able to free two of them if you set mode to DIO or DOUT (9 and 10 I think) but I really do not recommend you to attach anything to the flash pins or you might break everything ;) Did you run out of pins?

Akskaypohankar, 2017-09-18:
yes I am running out of pins

lonerzzz, 2017-09-20:
GPIO expanders are your friend in this situation. I am used them in multiple projects and they work with SPI or I2C to give you the extra pins you need. There are several that aren't too expensive.

lonerzzz, 2017-09-28:
@Akskaypohankar IS there any update on this or can we close the issue?

Akskaypohankar, 2017-10-03:
let's close I got other alternatives

closed 2017-10-03 by Akskaypohankar