espressif/arduino-esp32 #642: [feature request]Silent Start (no out out message)

luc-github, 2017-09-18:
Hi,
First thanks for this great project to guys rocks 😄 

I know ESP32 is still in devt so having a verbose start can be useful for debug
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:732
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
```
but  could it be possible to have a silent boot ? like using a flag when compiling to ask bootloader not to display the output or any other way

I already meet some issue with ESP8266 but because it use uncommon  baudrate, output is limited, unlike ESP32 which use a common used 115200, so output is harder to ignore.

Why such request ? it is because I connect the ESP to a 3D printer and such output randomly create problems: printer wrongly interpret the output a bunch of command and it is disturbed and can even do unexpected things

So having a possibility to get silent boot would be a great thing

copercini, 2017-09-18:
It's not what you are asking for, but you can use Serial1 or Serial2 to communicate with the 3D printer, so this debug output will not be displayed

luc-github, 2017-09-18:
Yes I agree but this decrease the number of pin or availability of another serial output.
 I know we can always find a workaround but before going that way I prefere to ask if there is chance to go to a clean way 😄 , better to ask than regret not asking

me-no-dev, 2017-09-19:
Pull GPIO15 low :) it will not show those messages ;) they come from the ROM and can not be controlled/silenced otherwise

me-no-dev, 2017-09-19:
please report and close on success :)

lonerzzz, 2017-09-29:
@luc-github Have you had any chance to try the suggestion?

luc-github, 2017-09-29:
Hi not yet because I missed this post Sorry @me-no-dev 
Thanks @lonerzzz for poping up

Yes it works so I close issue - thanks a lot sorry for the delay

closed 2017-09-29 by luc-github

Electrozest, 2018-01-23:
I have the same issue as I want to run deep sleep every few seconds and don't want lots of USB serial messages following every exit from deep sleep mode.
With GPIO15 pulled low on reset, the following rom debug output is suppressed (This currently appears after every exit from DEEPSLEEP_RESET):

rst:0x5 (DEEPSLEEP_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28

I recompiled ESP-IDF and used my own compiled bootloader and above was still present if GPIO not pulled low on power up. Checking all source code, I realised that the above has to be coming from the ROM code which is run prior to the custom bootloader.

What's interesting is that if GPIO15 is either pulled high, or left open circuit, the debug remains silent on each DEEPSLEEP_RESET. This indicates that the ESP32 ROM is only checking once on power reset and then setting a flag somewhere in RAM.
It would be useful if this flag could be changed after power reset, so that debug output appears only once, and then is suppressed for subsequent deep sleep cycles.
I assume only Espressif have source code for the ESP32 ROM, so only they would be able to easily determine which RAM location needs a bit toggling to disable the ROM debug output after power reset. This would be very useful for deep sleep applications.

probonopd, 2019-03-11:
Is it possible to switch the RX and TX pins in software with each other?

ic3man5, 2019-04-03:
I would like to note this issue is causing a hardware revision spin / hardware hack because we don't have a pulldown on GPIO15 and are trying to use I2C here. I would like to have a ROM with console output suppressed by default without external influence or at least better documentation saying this UART TXD0/RXD0 needs a pulldown on GPIO15 to completely disable UART on ROM power reset.

ic3man5, 2019-04-03:
Also what is the weakest pull-down I can use here without issue?