espressif/arduino-esp32 #643: Update.end() and Update.isFinished() == true, reboot still boots old firmware

tetious, 2017-09-19:
### Hardware:
Board:							Banggood WROOM-32 based Core V2 clone 
Core Installation/update date:			PlatformIO non-stage, 1.2.0
IDE name:							PlatformIO
Flash Frequency:					40Mhz/default?

### Description:
I'm having an odd problem with the Update class. I wrote some very simple code that uses an HTTP file upload (via ESPAsyncWebServer, based on the sample there) to update the firmware. It works perfectly, for a while. 

After about 15-20 updates, everything **appears** to work in that Update.end() and Update.isFinished() return true, and Update.hasError() returns false, but rebooting runs the old firmware. Once this starts happening, nothing except a serial flash via esp_tool will start it working again. Once I flash via esp_tool, the Update class works as expected again for a while. After about 15-20 updates, it will resume the broken behavior.

I dug a bit into the code, and it seems very straightforward, though it isn't using the IDF helper functions like I expected (for the ota write, at least), so I'm assuming the bug lies in Update, not in the IDF esp_ota* functions, however nothing jumped out at me.

Note that I can't presently run PIO stage due to this IDF bug: https://github.com/espressif/esp-idf/issues/994.

Any help/advice/things to try/ideas are appreciated! :)

me-no-dev, 2017-09-19:
This is rather strange. What you explain though is not probably at all related to actually flashing the partition but actually i setting the correct partition to boot. Still none of the code has anything like counter that could trip something else... rather really strange

tetious, 2017-09-19:
Thanks @me-no-dev, you gave me a clue that led to find some more information. Could the flash write be failing, causing the boot partition to be unbootable and it to automatically fallback? I'm not sure how that works.

More info:

I started logging the value of `esp_ota_get_boot_partition()->address` and `esp_ota_get_running_partition()->address` before and after the update, as well as in startup.

In normal cases, it switches between them as expected, and running == boot at startup. They seem to swap between 10000 and 150000, which I'm assuming is the position of app0 and app1.

Once it begins failing, boot and running are pointing different places at startup. In my current failure state, boot == 10000, and current == 150000. I'm not sure if app0 vs app1 matters here, as it does work fine for many attempts, exercising each slot many times over. 

I think this means that it is able to set the correct boot partition, but for some reason it isn't respecting that. Which makes me wonder if it is falling back or something. I don't see any messages about fallback and I do have verbose logging on.

Thanks again for any help you can provide. I might experiment next with using the IDF functions directly, unless something jumps out at you. :)

me-no-dev, 2017-09-20:
@tetious can you please give me your board configuration? More precisely what flash mode you are using (DIO/QIO)? I will rebuild a bootloader with verbosity ON for you that could give us some clue as to what is going on and why this happens.

tetious, 2017-09-20:
@me-no-dev Thanks for the idea to use a verbose bootloader. I didn't know about that and built and flashed it, and it works. The IDF documentation is excellent! (the page going over the boot process was especially helpful)

Once it starts to fail, it seems to be trying to boot into a non-existent partition:
```
D (597) boot: Both OTA sequence valid, using OTA slot 16
D (614) boot: Trying partition index 16...
D (627) boot: part offs 0x0 size 0x0
D (638) boot: Can't boot from zero-length partition
D (653) boot: Trying partition index 15...
```

That OTA slot 16 thing seems suspect? 

You can find a full dump of a working and failed boot here: https://gist.github.com/tetious/3e1148fb246bf57acabaa3012470a5a7

They also have all the details about my flash config, as the verbose bootloader seems to helpfully log all that. :)

projectgus, 2017-09-20:
Hi @tetious ,

During the V3.0-dev lifecycle there were some changes in the IDF bootloader to do with OTA partition selection. The platform.io ESP-IDF commit you have (969f1bb) doesn't have any known bugs, but the bootloader OTA partition selection has been changed in IDF master since that revision.

> That OTA slot 16 thing seems suspect?

The logging message is bad (since fixed) but the code itself should be testing OTA slot 16 modulo number of OTA slots (2), so that should be OTA slot 0 in that case I think. It's very weird that it thinks that partition is invalid (offset 0x0 size 0x0) though, as it logs the offset and size correctly a few lines above.

Are you able to share a flash dump of "broken" flash contents with us? You can read out the entire flash contents via `esptool.py read_flash --port PORT --baud 921600 0 0x400000 broken_readback.bin`. If you're able to also capture a working pre-broken flash, that could be useful as well.

If you don't want to attach them here publically, you can email them to angus at espressif.com. If you'd rather not share the app code with us at all, this is also OK - I can give you a different command that just dumps the partition table and OTA data - which is most likely enough to diagnose this problem.

tetious, 2017-09-21:
Hi @projectgus,

I selected 969f1bb at it was the commit used by the Arduino library currently on non-stage PlatformIO, which is what I'm using. 

When reproducing it again to send to you guys, I noticed it failed on the same offset of 16, which is interesting. The message is identical to the one I posted above. 

I've attached a good and broken readback as requested. 

[readbacks.zip](https://github.com/espressif/arduino-esp32/files/1319642/readbacks.zip)

Seriously guys, thanks for the awesome support! :)

me-no-dev, 2017-09-21:
@tetious should I consider this issue as resolved?

projectgus, 2017-09-21:
Unfortunately, yes. For a brief period in the V3.0 development cycle, this check was made against the raw OTA "slot" value, not the value modulo the number of OTA app slots:
https://github.com/espressif/esp-idf/blob/969f1bb9be1ce02f37115359111852a9bd2c9197/components/bootloader/subproject/main/bootloader_start.c#L265

So it starts failing when the counter hits 16.

The window of effected IDF commits is espressif/esp-idf@969f1bb to espressif/esp-idf@11a87ca8. Will put the word out to make sure noone deploys with these versions.

tetious, 2017-09-21:
Yep, resolved.

My thanks to both of you!

closed 2017-09-21 by tetious