espressif/arduino-esp32 #644: Failed to read a valid elf header, unpack requires a string argument of length 52

emilekm2142, 2017-09-19:
I used to get the following error (esp32_final is the name of the project):
`A fatal error occurred: Failed to read a valid ELF header from C:\Users\Emile\AppData\Local\Temp\VMBuilds\esp32_final\espressif_esp32\Debug/esp32_final.ino.elf: unpack requires a string argument of length 52` every 3 or 4 compilation try, but from today I cannot compile anything. I use vMicro for Visual Studio, however arduino IDE shows just an compilation error without any message. I have tried clean build but it did not work :/

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

cybergogo, 2018-10-08:
> I used to get the following error (esp32_final is the name of the project):
> `A fatal error occurred: Failed to read a valid ELF header from C:\Users\Emile\AppData\Local\Temp\VMBuilds\esp32_final\espressif_esp32\Debug/esp32_final.ino.elf: unpack requires a string argument of length 52` every 3 or 4 compilation try, but from today I cannot compile anything. The error repeats every single time. I use vMicro for Visual Studio, however arduino IDE shows just an compilation error without any message. I have tried clean build but it did not work :/
> 
> Edit: After 5 hours it started to work as earlier. Is there a way to fix that or the timespan of my coding will depend on ESPs mood :)? The issue happens randomly

Hi Man, had the same issue.
try to delete everything inside your pioenvs folder.
It will recompile the whole project and work again.
Take care.