espressif/arduino-esp32 #647: UART ISR

Volvox0815, 2017-09-20:
Hello together,

I am trying to write a DMX receiver, so I will have to write my own uart isr to detect breaks and frame errors for resetting buffers etc.

So to the probable issue I detected:
I started Arduino as a component of IDF:
```
extern "C" void app_main(void){
initArduino();
Serial.begin(115200);
...
```
If I step through I detect that when opening the "Serial" the _uart_isr(void *arg) in esp32-hal-uart.c
will be registered for the "ETS_UART_NUM"
```
static void IRAM_ATTR _uart_isr(void *arg)
{
    uint8_t i, c;
    BaseType_t xHigherPriorityTaskWoken;
    uart_t* uart;
    for(i=0;i<3;i++){
        uart = &_uart_bus_array[i];
        uart->dev->int_clr.rxfifo_full = 1;
        uart->dev->int_clr.frm_err = 1;
        uart->dev->int_clr.rxfifo_tout = 1;
        while(uart->dev->status.rxfifo_cnt) {
            c = uart->dev->fifo.rw_byte;
            if(uart->queue != NULL && !xQueueIsQueueFullFromISR(uart->queue)) {
                xQueueSendFromISR(uart->queue, &c, &xHigherPriorityTaskWoken);
            }
        }
    }

    if (xHigherPriorityTaskWoken) {
        portYIELD_FROM_ISR();
    }
}

void uartEnableGlobalInterrupt()
{
    xt_set_interrupt_handler(ETS_UART_INUM, _uart_isr, NULL);
    ESP_INTR_ENABLE(ETS_UART_INUM);
}
```

So the problem I think (correct me if I am wrong):
If I have one uart (usually uart 0) registered (with the "Serial.begin()"), the isr will handle and modify ALL 3 Uarts ("for" loop in ISR). So I am not able to detect correct interrupt flags on the other uarts any more because it may have been cleared already by this isr.

me-no-dev, 2017-09-22:
nope :) it clears only the uart that it has selected :) anyway there was a bug in uart ;) pull latest and try again please :)

copercini, 2017-10-20:
@Volvox0815 Is it working with last version of core?

Volvox0815, 2017-10-23:
for the moment I implemented an own uart callback isr, because for dmx I need to be able to detect the FE and BRK signals of uart modules intr_status reg. But the next weeks I will use another UART as Serial interface. I will try it again if everything works as expected.

everslick, 2018-04-19:
@Volvox0815 Can this be closed?

closed 2019-08-15 by stale[bot]