espressif/arduino-esp32 #649: Strange values from uartGetBaudRate

luc-github, 2017-09-20:
Hi,
 ESP32 miss one function to get current baudrate unlike ESP8266
so I added it in HardwareSerial
```
uint32_t  HardwareSerial::baudRate()

{
	return uartGetBaudRate(_uart);
}
```
But I got inconsistent values when using it:

Setting 57000 give 57000 ->Ok
Setting **115200** give **115201** ->**Different**
Setting **230400** give  **230423** ->**Different**
Setting 250000 give  250000 ->Ok

There is not such issue on ESP8266

here test sketch which display results at 115200
```
String result;
void setup() {
	result="Set 57000 get ";
	Serial.begin(57000);
	delay(1000);
	result+=String(Serial.baudRate());
	result+="\nSet 115200 get ";
	Serial.begin(115200);
	delay(1000);
	result+=String(Serial.baudRate());
	result+="\nSet 230400 get ";
	Serial.begin(230400);
	delay(1000);
	result+=String(Serial.baudRate());
	result+="\nSet 250000 get ";
	Serial.begin(250000);
	delay(1000);
	result+=String(Serial.baudRate());
	Serial.begin(115200);
}

void loop() {
	Serial.println(result);

	delay(3000);

}
```

Is that normal ?

me-no-dev, 2017-09-21:
Yup :) it's a integer math error. Here it is with PHP:
Baud: 57000, res: 57000
Baud: 115200, res: 115201
Baud: 230400, res: 230423
Baud: 250000, res: 250000

luc-github, 2017-09-21:
Thank you for confirming I just have one unit on hand, so was not sure was only me
Ok, so if you change UART driver it may fix the issue ?
I guess I can do the PR for baudRate function API won't change, right?

me-no-dev, 2017-09-21:
yes :) I will implement it on top of IDF's driver but that will be done in the esp32-hal-uart, so API for both will not change.

luc-github, 2017-09-21:
Thank you PR (https://github.com/espressif/arduino-esp32/pull/651) done

me-no-dev, 2017-09-21:
Thanks :) will merge as soon as the tests are done

me-no-dev, 2017-09-21:
maybe we can close this now?

luc-github, 2017-09-21:
sure thank you

closed 2017-09-21 by luc-github