espressif/arduino-esp32 #650: Strange issue with Serial2

viktor1970, 2017-09-20:
### Hardware:
Board:							DOIT DevKit V1
Core Installation/update date:			20/sep/2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
Hallo, I've a strange issue with Serial2: there's a lot of garbage sent.
Here's the [screenshot](http://it.tinypic.com/r/331iud1/9)

On 1st loop the data (Nextion display codes) are sent correctly ("n1.val=5" and "dim=99", both with 3 0xFF as terminator); starting from second loop there's only garbage (exceps when val=17, but there's garbage too).

Then I made only this change:
`HardwareSerial Seriale(1);`
Leaving same pins and (of course) same code, and it works perfectly.

There's some issue with Serial2?

Best regards

### Sketch:
```cpp

#include <Wire.h>
#include <Nextion.h>

HardwareSerial Seriale(2);
Nextion nex(Seriale, true);

unsigned long tm1,tm2=5;;
char bf[20];

void setup() {
  Seriale.begin(38400,SERIAL_8N1,26,27);
  Serial.begin(115200);
  tm1=millis();
}

void loop() {
 if((millis()-tm1)>2000)
 {
  Serial.printf("num: %d",tm2);
  sprintf(bf,"n1.val=%d",tm2++);
  nex.sendCommand(bf);
  if(tm2%2) Seriale.print("dim=2"); else Seriale.print("dim=99");
  Seriale.write(0xFF);
  Seriale.write(0xFF);
  Seriale.write(0xFF);
  tm1=millis();
 }
}```

### Debug Messages:
```
No crashes
```

me-no-dev, 2017-09-21:
There is... it's not quite clear to me what, but I am about to change the UART driver  so it should get better

me-no-dev, 2017-09-27:
Please pull the latest changes and report :) Thanks!

viktor1970, 2017-09-27:
Hallo, I installed the lastest changes but nothing changed:
This is what happens when using Serial(1):
[image]
The correct data "n1.val=5" plus 3 0xFF are sent and 01 FF FF FF code received.

When using Serial(2):
[image]

Only garbage data on TX, nothing on RX. I tried to use native pins too, same result

Regards

viktor1970, 2017-09-27:
I tried another ESP32, both with native and other pins. No changes.

Regards

everslick, 2018-04-19:
@viktor1970 Is this issue still unresolved for you? Can we close this?

EUNGJU-S, 2018-04-29:
Hello, @everslick 
I have same problem is this issue resolved ?
When I using Hardware Serial (2) sometimes It does not works correctly.
And I just changed Serial (2) to Serial (1) It is fine.
But I hope using more Serial port on ESP32 stably.

kolive360, 2018-10-05:
I try whit the Serial1 and work, but is fool

closed 2019-08-15 by stale[bot]