espressif/arduino-esp32 #653: WiFi Auto Reconnect Still Not Working?  Is there a best practice to guaranty connection?

vseven, 2017-09-21:
Board:							ModeMCU ESP32 Dev Module
Core Installation/update date:			15/Sep/2017
IDE name:							Arduino IDE

It seems there is no auto reconnect logic allowing the ESP32 to reconnect when the connection drops or if there is its not working properly.  I have had completely random results with my ESP32 board, sometimes its up for as little as 30 minutes and at most 18 - 20 hours.  WiFi is dropping and doesn't want to reconnect.  

The library I am using for WiFi is watching the WiFi event and on a disconnect its trying to reconnect which helps but sometimes it fails the reconnect and then that's it...I have to reboot.  There is no sleep/low power mode at all involved and the device is constantly powered by 3.3v in.   There was a case opened regarding this, https://github.com/espressif/arduino-esp32/issues/353, but its closed without any real answer (says auto connect is kind of implemented...not sure what that means).

I guess the question comes down to is auto reconnect implemented in the code itself, reliably, and if not what is the best Arduino code way of making sure it is connected?

-Allan

me-no-dev, 2017-09-21:
can you please enable debug in the board menu and see why it does not want to reconnect when you tell it to do so?

vseven, 2017-09-21:
I can try that tonight.  The Arduino code is pretty basic, watching the WiFi event and then has this:

```
		case SYSTEM_EVENT_STA_DISCONNECTED:
			Serial.println("WiFi lost connection.  Attempting to reconnect...");
			WiFi.reconnect();
			break;
```
But shouldn't there be something that can be called to make this happen on it's own?

nepeee, 2017-09-21:
Just for a quick fix:
Start a timer in the SYSTEM_EVENT_STA_DISCONNECTED that calls the WiFi.reconnect(); function and stop it in the SYSTEM_EVENT_STA_CONNECTED. The main problem is with your idea is if the device cant connect than the SYSTEM_EVENT_STA_DISCONNECTED event is not called so its never try to reconnect again.

vseven, 2017-09-21:
Unfortunately I'm adapting someone else library and I'm not very familiar with C++ but I'll see what I can do about adding a timer (I know VB.net...I can at least figure out the logical parts and hopefully adapt).  

But I think you are right about it not reconnecting or better yet only trying on a disconnect which then if it fails just once it give up.  Enabling debug might show me that so I'll start there.

But back to the original question and the previously closed issue:  Is this functionality not built-in?  And if not why was the previous issue closed as "kind of" working?  Also if not will it be?

vseven, 2017-09-22:
I switched the debug level to debug and opened serial monitor and got this over and over:

```
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
```

I checked the last time it sent data to another controller and that was over 24 hours ago so I'm assuming its stuck in this mode.  So I powered it down and back up, it connected to WiFi just fine and transmitted data how it should for about 2 hours this time.  Then without any other message it disconnected and started with the exact same message over and over.  I let it go for a hour, rebooted the controller, and its connected again but I'm sure its just a matter of time before it starts over.

So the first question is how do I fix it.  Second this still leads me to my previous questions about auto reconnect

me-no-dev, 2017-09-22:
Guys please enable debug so you can see more verbose WiFi state output:
[image]

everslick, 2017-09-22:
I use the following code to check if my device is still connected to the STA (call this from loop()). You get the idea? wifi_is_connected is a global that gets set and reset in the wifi event callback.

```
static void poll_watchdog_sta(void) {
  static uint32_t ms = millis();
  static uint8_t watchdog = 0;

  // watch out for STA disconnects and reboot if
  // connection cannot be reestablished after
  // X minutes

  // this watchdog MUST NOT reboot the device if
  // wifi is not enabled or never was connected
  // since last reboot

  if (wifi_is_enabled && watchdog_enabled && ((millis() - ms) > 1000 * 6)) {
    ms = millis();

    if (!wifi_is_connected) {
      if (++watchdog < watchdog_timeout * 10) { // timeout in minutes ( * 10 )
        if (watchdog == 1) {
          log_print(F("WIFI: arming network watchdog (reboot in %i min.)"),
            watchdog_timeout
          );
        }
      } else {
        log_print(F("WIFI: still not connected, triggering reboot ..."));
        system_reboot();
      }
    } else {
      if (watchdog) {
        log_print(F("WIFI: network is back, disarming watchdog"));
        watchdog = 0;
      }
    }
  }
}
```

vseven, 2017-09-22:
Yeah, I don't see a Loop() in the library I'm using for WiFi (again, third party) but I understand what you are doing and can run with it. 

Going to turn on the verbose logging and see if I get any more info first before changing anything.

Also the WiFi Events that are being checked and acted on are SYSTEM_EVENT_STA_GOT_IP, SYSTEM_EVENT_STA_DISCONNECTED, SYSTEM_EVENT_STA_START, and SYSTEM_EVENT_STA_CONNECTED.   There are more however at https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_event.h so I added a default: to the switch that should print out the event to see if the library is getting something but just not acting on it.

I saw in the library a 	//WiFi.setAutoReconnect(true);  so I'm assuming the original author tried to use auto reconnect but it wasn't working so commented it out?

vseven, 2017-09-22:
@me-no-dev  - Verbose gave a much better explanation:

WiFi lost connection.  Attempting to reconnect...
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5

And this repeats every 5 or so seconds.

vseven, 2017-09-25:
Based on that first error that repeats over and over I found this:  https://github.com/espressif/arduino-esp32/issues/180.  In it you said "Try flushing the client or reading all available data. ".  How would I go about doing that?  I'm looking at the library I'm using and this is what they have to catch the WiFi Event:

```
//**************************************************************************************
/// Event Handler for ESP32 WiFi Events (needed to implement reconnect logic for now...)
//**************************************************************************************
	void MyESP32WiFi::WiFiEvent(WiFiEvent_t event)
	{
		Serial.printf("[WiFi-event] event: %d\n", event);

		switch (event) {
		case SYSTEM_EVENT_STA_GOT_IP:
			Serial.println("WiFi connected");
			Serial.println("IP address: ");
			Serial.println(WiFi.localIP());
			break;
		case SYSTEM_EVENT_STA_DISCONNECTED:
			Serial.println("WiFi lost connection.  Attempting to reconnect...");
			WiFi.reconnect();
			break;
		case SYSTEM_EVENT_STA_START:
			Serial.println("ESP32 station start");
			break;
		case SYSTEM_EVENT_STA_CONNECTED:
			Serial.println("ESP32 station connected to AP");
			break;
		default:			
			Serial.println("Unhandled WiFi Event raised.");
			break;
		}

	}
```

Based on the above code I'm getting the "5" which is translating to the WiFi connection lost and then the code is trying to execute "WiFi.reconnect();" but its obviously not working.  Should I have something else there?

me-no-dev, 2017-09-25:
Try calling WiFi.begin() instead of reconnect and report back :)

vseven, 2017-09-26:
@me-no-dev  - I changed it to WiFi.begin() instead of WiFi.reconnect() but it didn't help.  Worked for about a hour then got the lwip_connect_r:113 line, each one about a minute or so apart followed by the disconnect every 5 or so seconds:

```
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 7 - NOT_ASSOCED
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
```

What else can I try?

me-no-dev, 2017-09-26:
hmmm.... this seems like an issue either in the lower stack or in your router.
you can see the first reason being NOT_ASSOCED and then you get AUTH_EXPIRE.
could you maybe turn off the STA and then turn it back on? The disconnect cause makes me think that something in the re-assoc request is either not changed or your router does not accept, so it refuses you to connect back. I'll see if I can raise an internal issue about this. What is the model router and maybe firmware version that you are using?

vseven, 2017-09-26:
Its a Netgear Nighthawk R8000.  Currently on the latest firmware V1.0.3.54_1.1.37 although it did it on the last version also.  I have 20 - 25 devices connected at any given time without issues including the previously mentioned two ESP8266's.  

I will attempt to replace the Wi-Fi begin/reconnect line with the STA off and on and see what happens.

me-no-dev, 2017-09-26:
I have also alerted the wifi team :) we will see if anything comes out of this

vseven, 2017-09-26:
Thanks.  Should I be trying to do the STA off and on?  If so can you guide me how or point me in the right direction?

I also saw this:  https://github.com/espressif/esp-idf/issues/738#issuecomment-311626685  which indicated that I can't call the WiFi.connect or WiFi.Begin from the event handler?  Could that also be my issue?

vseven, 2017-09-28:
So I bought a second ESP32, same one, to try and make sure it isn't a hardware issue.  I recompiled the firmware yesterday night off a fresh pull from github (https://nodemcu.readthedocs.io/en/dev-esp32/en/build/) and flashed both the new chip and the old one.  I also flashed both with the exact same sketch.  The original is in my garage, a little ways from my router, and the other test one is in my living room where the WiFi is a lot stronger.  Both have serial monitor running on a connected PC with verbose set.   I'll report back anything I find.  

With the above said is there anything else I can try?  How do I do the STA off and on?  Can that be done in the STA_Disconnect part where the WiFi.reconnect is?

vseven, 2017-09-29:
Well that didn't last long....I'm having the same issue with the new one.  First the original one started with the same "[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113" error and not sending data after barely 30 minutes online.  Usually got that error two at a time with about 5 or 6 seconds between then then 45 - 50 seconds between the next set but again kinda random.  Did that for around 10 minutes then went into the same disconnect pattern:

```

[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113
Data: Sending: temperature1 -49.02
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113
Data: Sending: humidity1 -5.02
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 7 - NOT_ASSOCED
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 202 - ASSOC_FAIL
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
```

And repeated the last 4 lines over and over.  At one point it did do something different...the error message changed slightly:

```
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 118
Data: Sending: temperature2 -1.00
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 118
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 118
Data: Sending: humidity2 -1.00
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 118
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 118
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
```

Notice the error switched to ending in 118 as opposed to 113.  But from that point on it just repeated the 118 error randomly along with the STA_DISCONNECTED and AUTH_EXPIRED over and over.

Then the new ESP32 started doing the exact same thing.  mostly 113 error then the same pattern of  NOT_ASSOCED, ASSOC_FAIL, and AUTH_EXPIRE followed by just the disconnect and AUTH_EXPIRE over and over.

They failed at different times, 25 minutes apart, while they were started at the exact same time.  WiFi signal doesn't seem to be a factor nor does any certain timing.

me-no-dev, 2017-09-29:
Still waiting on response from the WiFi team. Will make some more noise as this is annoying to me as well. Issue is somewhere way below Arduino and in the closed source wifi lib.

vseven, 2017-09-29:
Thanks.  I'm going to attempt to add a counter to the STA_DISCONNECTED event switch statement that I keep getting and after 5 try to re-init the wifi and after 10 reboot the controller and on a connect reset the counter to 0, see if that at least keeps the controller up.

If there is anything else I can provide you to help get this fixed or code to try please let me know.  If it matters the ESP32 I'm using is being used for two voltage inputs, one binary input, and a DTH22 (temp + humidity) then that data is being sent to a controller through HTTP.

liuzfesp, 2017-09-29:
Hi @vseven, I just can't reproduce this issue myself, could you let me know the firmware version of WiFi?  You can get if from the startup log, it looks like "I (588) wifi: wifi firmware version: 2cd69aa",  then I can provide a debug version wifi lib to you to debug this issue.

liuzfesp, 2017-09-29:
@vseven could you let me know the WiFi reconnect logic? I means when to call esp_wifi_connect()? It will be helpful if you can paste the implementation logic.

me-no-dev, 2017-09-29:
@liuzfesp wifi logging is disabled in Arduino.
@vseven comment [this line](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L91) to get it to show

vseven, 2017-09-29:
Ok, I commented out that line and recompiled.  I'll report back.  As a side note I tried to add a counter to reconnect by recalling my init() but it didn't do anything....it was like the init() was never called.  But the ESP.restart defiantly was:
```

	void SmartThingsESP32WiFi::WiFiEvent(WiFiEvent_t event)
	{
		Serial.printf("[WiFi-event] event: %d\n", event);

		switch (event) {
		case SYSTEM_EVENT_STA_GOT_IP:
			Serial.println("WiFi connected");
			Serial.println("IP address: ");
			Serial.println(WiFi.localIP());
			break;
		case SYSTEM_EVENT_STA_DISCONNECTED:
			Serial.println("WiFi lost connection.  Attempting to reconnect...");
			WiFi.reconnect();
			disconnectCounter++;
			if (disconnectCounter > 5) {
				Serial.println("We have recieved the STA_DISCONNECTED event 5 times now.  Re-init...");
				void init();
			}
			if (disconnectCounter > 10) {
				Serial.println("We have recieved the STA_DISCONNECTED event 10 times now.  Reboot...");
				ESP.restart();
			}

			break;
		case SYSTEM_EVENT_STA_START:
			Serial.println("ESP32 station start");
			break;
		case SYSTEM_EVENT_STA_CONNECTED:
			Serial.println("ESP32 station connected to AP");
			disconnectCounter = 0;
			break;
		}
	}
```

vseven, 2017-09-30:
@liuzfesp - So it did the same thing with the debug enabled, hopefully something in here will tell you something.  It ran fine for a hour or so then started throwing the 113 error randomly.  After about 15 minutes of random 113 errors in between the actual data sending routines it did this:

```
I (5427111) wifi: state: run -> auth (7c0)
I (5427111) wifi: pm stop, total sleep time: 0/1119534452

I (5427111) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 7 - NOT_ASSOCED
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
I (5427133) wifi: state: auth -> init (0)
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 202 - ASSOC_FAIL
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
I (5427276) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5427276) wifi: state: init -> auth (b0)
I (5428276) wifi: state: auth -> init (2)
I (5428277) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
I (5428413) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5428414) wifi: state: init -> auth (b0)
I (5429414) wifi: state: auth -> init (2)
I (5429414) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
I (5429551) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5429551) wifi: state: init -> auth (b0)
I (5430551) wifi: state: auth -> init (2)
I (5430551) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
I (5430688) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5430688) wifi: state: init -> auth (b0)
I (5431689) wifi: state: auth -> init (2)
I (5431689) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
We have recieved the STA_DISCONNECTED event 5 times now.  Re-init...
I (5431826) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5431826) wifi: state: init -> auth (b0)
I (5432826) wifi: state: auth -> init (2)
I (5432826) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
We have recieved the STA_DISCONNECTED event 5 times now.  Re-init...
I (5432963) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5432963) wifi: state: init -> auth (b0)
I (5433964) wifi: state: auth -> init (2)
I (5433964) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
We have recieved the STA_DISCONNECTED event 5 times now.  Re-init...
I (5434100) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5434101) wifi: state: init -> auth (b0)
I (5435101) wifi: state: auth -> init (2)
I (5435101) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
We have recieved the STA_DISCONNECTED event 5 times now.  Re-init...
I (5435238) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5435238) wifi: state: init -> auth (b0)
I (5436238) wifi: state: auth -> init (2)
I (5436239) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
We have recieved the STA_DISCONNECTED event 5 times now.  Re-init...
I (5436375) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5436376) wifi: state: init -> auth (b0)
I (5437376) wifi: state: auth -> init (2)
I (5437376) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
We have recieved the STA_DISCONNECTED event 5 times now.  Re-init...
We have recieved the STA_DISCONNECTED event 10 times now.  Reboot...
I (5437405) wifi: flush txq
I (5437407) wifi: stop sw txq
I (5437410) wifi: lmac stop hw txq
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

```

So for each STA_DISCONNECT it tried calling WiFi.reconnect with no apparent success then it tried running init() after more then 5 disconnects which also didn't help then once it hit 10 disconnects it rebooted which did reboot and reconnect.

vseven, 2017-10-02:
I kept serial monitor running on my ESP32 all weekend and it rebooted about 10 times between 6pm Friday night and 9pm Sunday night.  Each time it was exactly like the above logs.

@me-no-dev  - can you guide me in a way to just turn off and back on the STA to see if that also works for reconnecting and I can try that after 5 failed attempts?  The reboot after 10 works but I plan on using this to control RGB lighting and I really don't want to have lights turning off and on.

@liuzfesp  - Is there anything else I can provide you with for debugging?  Or anything that stands out I can try?

vseven, 2017-10-06:
Any updates?  Still have constant disconnects with both devices.

me-no-dev, 2017-10-10:
@liuzfesp any news here?

vseven, 2017-10-16:
I see there are some code changes coming with bluetooth and WiFi.   Not sure if any of that will help me or not.  I still continue to have the same issues and the reboot is working to keep my data being passed but I haven't been able to use it for RGB light strip control since the lights kept turning off on a reboot.  @liuzfesp - Is there anything in the logs that help or anything else I can provide or try?

vseven, 2017-10-25:
I cleaned up some things, made sure in my code there was nothing doing delays or blocking, and I now have less errors (don't see the 113 and 118 errors) but still having the same disconnect issue and it seems directly related to the WiFi trying to authenticate (renew?) and failing.  Here is a fresh capture from this weekend where I had it reboot about 5 times in a 24 hour period.  Again, same thing on two different ESP32s, I look for the disconnect (event 5) and try to reconnect with a WiFi.reconnect and a WiFi.begin both with no luck so after 10 tries it reboots and connects fine afterward:

```
I (1903652) wifi: active cnt: 5
I (1913652) wifi: active cnt: 5
I (1923652) wifi: send null to keep active
I (1933652) wifi: send null to keep active
I (1943653) wifi: active cnt: 1
I (1953653) wifi: active cnt: 2
I (1963653) wifi: send null to keep active
I (1973653) wifi: send null to keep active
I (1983653) wifi: send null to keep active
I (1993653) wifi: active cnt: 1
I (2003654) wifi: send null to keep active
I (2013654) wifi: send null to keep active
I (2023654) wifi: send null to keep active
I (2033654) wifi: send null to keep active
I (2043654) wifi: send null to keep active
I (2053654) wifi: send null to keep active
I (2063654) wifi: send null to keep active
I (2073655) wifi: send null to keep active
I (2083655) wifi: active cnt: 1
I (2093655) wifi: send null to keep active
I (2103655) wifi: send null to keep active
I (2113655) wifi: send null to keep active
I (2123655) wifi: send null to keep active
I (2133655) wifi: send null to keep active
I (2143656) wifi: send null to keep active
I (2153656) wifi: send null to keep active
I (2163656) wifi: send null to keep active
I (2173656) wifi: send null to keep active
I (2183656) wifi: send null to keep active
I (2193656) wifi: send null to keep active
I (2203657) wifi: send null to keep active
I (2213657) wifi: send null to keep active
I (2223657) wifi: send null to keep active
I (2233657) wifi: send null to keep active
I (2243657) wifi: send null to keep active
I (2253657) wifi: send null to keep active
I (2263657) wifi: send null to keep active
I (2273658) wifi: active cnt: 2
I (2283658) wifi: send null to keep active
I (2293658) wifi: send null to keep active
I (2303658) wifi: send null to keep active
I (2313658) wifi: send null to keep active
I (2323658) wifi: send null to keep active
I (2333658) wifi: send null to keep active
I (2343659) wifi: send null to keep active
I (2353659) wifi: send null to keep active
I (2363659) wifi: send null to keep active
I (2373659) wifi: send null to keep active
I (2383659) wifi: send null to keep active
I (2393659) wifi: send null to keep active
I (2403660) wifi: send null to keep active
I (2413660) wifi: send null to keep active
I (2423660) wifi: send null to keep active
I (2433660) wifi: send null to keep active
I (2443660) wifi: send null to keep active
I (2453660) wifi: send null to keep active
I (2463660) wifi: send null to keep active
I (2473661) wifi: send null to keep active
I (2483661) wifi: send null to keep active
I (2483663) wifi: state: run -> auth (7c0)
I (2483663) wifi: pm stop, total sleep time: 0/-1824912959

I (2483663) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[WiFi-event] event: 5
WiFi lost connection.  Attempting to reconnect...
I (2483674) wifi: state: auth -> init (0)
[WiFi-event] event: 5
[... 86 lines cut ...]
Disabling ESP32 WiFi Access Point

I (1816) wifi: Init Ampdu: 1 tx baw=6 rx baw=6
I (1816) wifi: mode : sta (30:ae:a4:07:f1:98)
[WiFi-event] event: 2

Initializing ESP32 WiFi network.  Please be patient...
ESP32 station start
Attempting to connect to WPA SSID: (SSIDREMOVED)
.I (2943) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
.I (3600) wifi: state: init -> auth (b0)
I (3602) wifi: state: auth -> assoc (0)
I (3606) wifi: state: assoc -> run (10)
I (3635) wifi: connected with (SSIDREMOVED), channel 1
[WiFi-event] event: 4
ESP32 station connected to AP
[WiFi-event] event: 7
WiFi connected
IP address: 
192.168.1.142
```

@liuzfesp / @me-no-dev   - Is there any other information I can provide?  Is this happening to anyone else or only with my two HiLetGo ModeMCU ESP32s?  Is there any custom firmware that I can try?  

-Allan

Markusenz, 2017-11-23:
I have the exact same behaviour on my WROOM-32 module.
Every 3rd or 4th reboot, it hangs in the STA_DISCONNECTED -> AUTH_EXPIRED loop.
Checked a second module of same type, showing same behaviour.
My Router is a TL-WR841 (TPLINK) but happens as well with a ASUS RT-AC87U.

When will there be a solution, its useless if there is need of a reboot in such situations...

vseven, 2017-11-23:
I haven't gotten a response in a month although, no offense, I'm happy it's happening to someone else so I know I'm not alone.

@liuzfesp / @me-no-dev  - Has there been any progress on this or anything we can try to get this to stay connected reliably to WiFi.

me-no-dev, 2017-11-23:
poked @liuzfesp again... this is as much as I can really do. I do not have access to the WiFi lib source

copercini, 2017-11-23:
@igrr

me-no-dev, 2017-11-23:
@copercini why?

igrr, 2017-11-24:
Since i was tagged here... @vseven, is there any chance you could perform packet capture using Wireshark and a WiFi adapter in monitor mode? Of particular interest is the time when reconnect process starts failing. Since we don't know what is happening here, and we can't reproduce similar issue in our CI environment, the packet capture might reveal some anomaly, giving a clue about the situation you are facing.

Another thought (just to reduce the number of variables), does this issue also happen if you connect to an open WiFi network?

Markusenz, 2017-11-28:
currently, all reconnects fail with my WROOM32s with following debug output:

[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 2 - STA_START
....[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 4 - STA_CONNECTED
..[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 7 - STA_GOT_IP
WiFi connected
... then I block antenna reception or press RESET at my AP, to force a disconnect ...
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 200 - NO_AP_FOUND
... when reception should be ok again, it takes about 2 minutes until it says:

[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 8 - STA_WPS_ER_SUCCESS

However it remains disconnected from network forever in that state. 
WPS is turned off at my AP. Do not know what  the debug message means.
Same behaviour with two different APs and two different WROOM32 modules, using the wificlient example sketch. No matter if WPA2 used or open network.
Can send Wireshark log in monitor mode if needed.

sidoh, 2017-12-14:
First of all, thanks everyone for your hard work on the SDK.  I've had a ton of fun developing with it, and I'm very grateful for all of the effort that's gone into it.

I'm seeing this issue too.  I think I may have managed to narrow it down a little bit.  I'm using this really small sketch that should work on both ESP8266 and ESP32:

https://github.com/sidoh/esp32_reconnect_demo

I turned off auto reconnect so that I had more control of the variables.

## Setup

1. Testing on both ESP8266 and ESP32
2. Using a mobile hotspot network.  Tried both WPA2 PSK and open networks
3. Allow MCU to connect to network and run for a few seconds.  I added an NTP client to verify connectivity.

Then I tried two different things:

#### (A) Tear down network, immediately recreate

Under these circumstances, the network is usually recreated by the time the MCU notices it's disconnected.

#### (B) Tear down network, wait for MCU to notice it's disconnected, recreate network

## Observations

1. Behavior is the same with both WPA2 and open networks.
1. The ESP8266 successfully reconnects with both (A) and (B).
1. ESP32 reconnects in experiment (A), but gets stuck in (B).
1. ESP32 **does successfully reconnect** in both (A) and (B) if: `WiFi.disconnect(true);` _and_ `WiFi.begin(WIFI_SSID, WIFI_PASSWD);` are called.  Seems like both are necessary.  
  \
  The boolean `wifioff` parameter to `disconnect` being set to true also seems necessary.  Since `WiFi.disconnect(true);` appears to clear WiFi settings, it's necessary to re-specify the ssid and password in the call to `begin`.

## Logs

All of these are with an open network.  I'm omitting the ESP8266 logs since the behavior was as expected, but can provide if they'd be useful.

#### ESP32, experiment (A)

```
I (25) wifi: wifi firmware version: 708a055
I (26) wifi: config NVS flash: enabled
I (26) wifi: config nano formating: disabled
I (31) wifi: Init dynamic tx buffer num: 32
I (32) wifi: Init data frame dynamic rx buffer num: 64
I (32) wifi: Init management frame dynamic rx buffer num: 64
I (35) wifi: wifi driver task: 3ffd46dc, prio:23, stack:4096
I (40) wifi: Init static rx buffer num: 10
I (44) wifi: Init dynamic rx buffer num: 0
I (48) wifi: wifi power manager task: 0x3ffd9418 prio: 21 stack: 2560
I (425) wifi: mode : sta (30:ae:a4:04:42:c8)
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 2 - STA_START
I (2836) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (2836) wifi: state: init -> auth (b0)
I (2853) wifi: state: auth -> assoc (0)
I (2872) wifi: state: assoc -> run (10)
I (2872) wifi: connected with glowyrectangle, channel 6
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 7 - STA_GOT_IP
Setup completed
[4] ssid=glowyrectangle
[5] ssid=glowyrectangle
I (5872) wifi: pm start, type:0

[6] ssid=glowyrectangle
[7] ssid=glowyrectangle
[8] ssid=glowyrectangle
[9] ssid=glowyrectangle
[10] ssid=glowyrectangle
I (10388) wifi: state: run -> auth (7c0)
I (10389) wifi: pm stop, total sleep time: 0/4516399

[... 201 lines cut ...]
I (30312) wifi: state: init -> auth (b0)
I (30318) wifi: state: auth -> assoc (0)
I (30339) wifi: state: assoc -> run (10)
I (30340) wifi: connected with glowyrectangle, channel 6
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 7 - STA_GOT_IP
Connection result: 3
[1513219831] ssid=glowyrectangle
[1513219832] ssid=glowyrectangle
I (33340) wifi: pm start, type:0

[1513219833] ssid=glowyrectangle
[1513219834] ssid=glowyrectangle
[1513219835] ssid=glowyrectangle
[1513219835] ssid=glowyrectangle
[1513219836] ssid=glowyrectangle
```

## Wrapup

I've done some digging around in the SDK, but don't see anything obvious.  I may dig around a bit, but I hope the above information is useful.  If there's any other info I can provide please let me know; I'd be more than happy to oblige.  

I tried getting some pcaps in monitor mode, but they weren't very interesting.  Just seemed that the ESP32 wasn't sending anything after the disconnect.

I'm happy that I've at least found a workaround (disconnect(true) + begin(ssid, passwd)).

zhangyanjiaoesp, 2017-12-14:
@vseven  @Markusenz @sidoh
Sorry to reply later, and thanks for your hard work on the SDK.
The attachment is the debug version of wifi lib,you can use it to have a test, then we can analysis your logs to solve the problem. 
[wifi_lib.tar.gz](https://github.com/espressif/arduino-esp32/files/1559164/wifi_lib.tar.gz)

vseven, 2017-12-14:
@igrr  - I have tried both open and secure with the same results.  I can try to get you a packet capture but the issue is extremely random, it could do it 4 times in a hour or twice in a day so capturing might be very tedious.  Now that multiple people are having the same issue however I will try the debug version that @zhangyanjiaoesp  posted and if that still doesn't lead you to the solution I'll try the packet capture.

Will auto reconnect be fixed to work as intended?  It sounds like @sidoh is getting around it which I have not had much luck with.

I'll get this recompiled and loaded this weekend.

sidoh, 2017-12-14:
I think the most interesting thing I noticed was that:

```c++
WiFi.disconnect(true);
WiFi.begin(SSID, PASSWORD);
```

always eventually reconnects for me (the `true` param to `disconnect` is important).  I think `WiFi.disconnect(true)` turns off the radio.  My guess is that under certain circumstances some state about the network the ESP32 is supposed to be connecting to gets cleared, and the above resets it.

Also interesting is that the issue appears to be 100% reproducible by turning off the wifi network, waiting for the ESP32 to notice it has disconnected, and turning the network back on.  If the network reappears before the ESP32 notices, it's sometimes able to reconnect successfully.  From the logs, the difference appears to be that a `NO_AP_FOUND` event is triggered in the former case, but not the latter.

zhangyanjiaoesp, 2017-12-15:
@vseven 
[wifi_debug_lib.zip](https://github.com/espressif/arduino-esp32/files/1561384/wifi_debug_lib.zip)

sidoh, 2017-12-15:
Re-ran the code linked with the debug libs.  Logs here:

* [fast_ap_recreate.log](https://gist.github.com/sidoh/bfb3612256f80242024e5970cdc58e1a#file-fast_ap_recreate-log) - experiment (A)
* [slow_ap_recreate.log](https://gist.github.com/sidoh/bfb3612256f80242024e5970cdc58e1a#file-slow_ap_recreate-log) - experiment (B)
* [slow_ap_recreate_with_disconnect_true.log](https://gist.github.com/sidoh/bfb3612256f80242024e5970cdc58e1a#file-slow_ap_recreate_with_disconnect_true-log) - same as experiment (B) with `WiFi.disconnect(true);` and `WiFI.begin(SSID)`.

This was all on an open network.

vseven, 2018-01-02:
@zhangyanjiaoesp  - Any news on this?  I haven't tried the debug code because @sidoh already did and provided lots of logs already but then again it's been a couple more weeks and no real answers.

zhangyanjiaoesp, 2018-01-04:
@sidoh @vseven 
Can you use Omnipeek to catch the wifi pkt ?  Or can you provide a demo for us to reproduce your issue?

zhangyanjiaoesp, 2018-01-04:
@sidoh 
In the `slow_ap_recreate.log` ,we find there are print like 

> [1513308152]ssid=  
Attempting to reconnect...
it seems like your ssid is empty, so after scanning all channels, return no ap found.
so you should check your code to see whether the ssid is set correctly.
[image]

sidoh, 2018-01-05:
Hi @zhangyanjiaoesp,

I think this is probably the meat of the issue.  In the `slow_ap_recreate.log`, the reconnection code is just doing (full sketch [here](https://github.com/sidoh/esp32_reconnect_demo/blob/master/src/main.cpp)):

```c++
WiFi.begin();
```

so it's not supplying an AP.  My understanding of the SDK is that it should just reconnect to the last known AP.  Perhaps if it can't find the last used SSID, it clears the config?  Is that expected?

Using this instead seems to always work:

```c++
WiFi.disconnect(true);
WiFi.begin(ssid, passwd);
```

zhangyanjiaoesp, 2018-01-08:
@sidoh Yes, you are right. This is expected behavior, you can workaround it by your method:

> WiFi.disconnect(true);
WiFi.begin(ssid, passwd);

>

sidoh, 2018-01-08:
I see, this makes sense.  I'll try to reproduce without the `WiFi.begin()` in the next couple of days.  Think the right thing is to just have the SDK handle reconnection.  

Where is reconnection handled in the SDK?  Only reference I can find is [here](https://github.com/espressif/arduino-esp32/blob/0c038b406394f6ccbcb2cbc0e11f5bff899dad48/libraries/WiFi/src/WiFiGeneric.cpp#L277-L280), which seems to use `WiFi.begin()`.  Is that expected?

zhangyanjiaoesp, 2018-01-09:
yes, you are right

sidoh, 2018-01-09:
Doesn't that mean that the SDK code handling reconnections would have the same issue my demo has?  If it tries to reconnect after the AP has disappeared, the SSID will get cleared and it won't be able to reconnect unless the AP info is re-supplied?

everslick, 2018-01-09:
definitly the SDK should try to reconnect automotically. That is the whole point of having WiFi.setAutoReconnect(true).

vseven, 2018-01-09:
@everslick  - I agree completely, it doesn't seem to work which is why I posted this originally 3+ months ago.  The WiFi.setAutoReconnect(true) doesn't seem to make it reconnect which is ironic to say the least.  The only thing that works is to reboot the controller and let it go back through its normal connect (which isn't going to work for me when controlling LEDs as they turn off) or doing to full disconnect and reconnect.  

@zhangyanjiaoesp  - So you are saying WiFi.setAutoReconnect(true) simply doesn't work?  And to work around we have to call the disconnect and reconnect with the SSID and password again?  If this is true will it be fixed or at least the documentation updated to say it doesn't work?

rrobinet, 2018-01-09:
I have a similar issue,
After reseting the by software using the following sequence:
  WiFi.disconnect();                          //Ensure current  ssid/password is erased
  delay (1000);
  ESP.restart();
The ESP reboots but never connects to the AP until an hardware reset.
The function Wifi.disconnect () is also not working correctly.
Robert

rrobinet, 2018-01-09:
The hard way is to connect a GPIO pin to the EN one and force a System reset (GPIO Low) in case reboot/restart  or wifi (re)connection is needed.

vseven, 2018-01-09:
Don't do the disconnect.  Just restart works for me.  But again...shouldn't have to restart if WiFi.setAutoReconnect(true) worked properly.

rrobinet, 2018-01-09:
It should be nice to have the Wifi.disconnect or autoreconnect working, but what's the point, if there is a Wifi connection issue, this is a major one in case of ESP32 (all subsequent applications probably relies on Wifi), so the ultimate solution is to try a hardware reset, if this one fails ... bye bye ESP32 usage.

vseven, 2018-01-09:
Rebooting is **NOT** a solution.  I'm using this for RGB LED control and rebooting resets the lights to off.  If there is a function that tells the ESP that if it loses WiFi it should keep retrying to connect then that function should actually work as documented.

rrobinet, 2018-01-09:
OK, I have tried to reset the ESP through a GPIO, but it looks that ESP doesn't like suicide, so it doesn't work without an external device. So I agreed the only solution that left is that the WiFi disconnect, the ESP restart or the Wifi auto reconnect are actually working.
Another issue for which we have to wait for a functional ESP module (I note that as far I could test it, the I2C protocol is another one for which I am waiting for a solution).
ESP32 is far to be reliable until now ....
Robert

rrobinet, 2018-01-09:
and I didn't mention the SPI interface ...

[98 more replies]

closed 2018-11-05 by vseven