espressif/arduino-esp32 #654: Can't get back into DHCP mode, from static IP,MASK, GATEWAY,DNS

euquiq, 2017-09-21:
Hi There !

On the ESP8266, if I was using static IP,GATEWAY, MASK and DNS addresses but wanted to go back into using DHCP, I issued the following line of code:

`WiFi.config({ 0,0,0,0 }, { 0,0,0,0 }, { 0,0,0,0 });`

... and thus, the ESP8266 was forced to get those values from DHCP.

Now, on ESP32, whenever that line of code is executed, I get the following error into my serial port:

`E (2807) event: invalid static ip`

... And the esp32 fails to get a Wifi Connection.

Probably there is a better way to accomplish "back to DHCP" on the ESP32, but just in case I wanted to place it as an issue.

Regards,

Enrique

me-no-dev, 2017-09-22:
Done :) can you please give this a go?

closed 2017-09-22 by me-no-dev

euquiq, 2017-09-22:
Thanks, me-no-dev, now it works perfectly ok!