espressif/arduino-esp32 #655: SPI MSB_FIRST not working

nepeee, 2017-09-21:
SPI.begin(12, 0, 13, 14);

SPISettings spiSettings = SPISettings(50000, SPI_MSBFIRST, SPI_MODE3);
SPI.beginTransaction(spiSettings);
SPI.transferBits(0xFFFFFFFF, NULL, 3);
SPI.endTransaction();

I use an oscilloscope to inspect the MOSI pin(13), the output is always low.
When i change to SPI_LSBFIRST i got the HIGH output for three clock cycles as i except.
I also tried with 0xFF, 0xFFFF as data and with the default pin mapping.

nepeee, 2017-09-21:
The main problem i think is the mask part of the code. If i remove that, the MOIS is high for three clock cycles but when i try to send 0x0D no data again:
SPI.transferBits(0xD0, NULL, 4); // not working

Only works if i put the data int the int32 MSByte like this:
SPI.transferBits(0xD0000000, NULL, 4); // working

me-no-dev, 2017-09-22:
since transfer bits takes 32 bit number, MSB of that is exactly what you say is working :)

me-no-dev, 2017-09-22:
btw MSB in this case is Most Significant Bit (not byte)

nepeee, 2017-09-22:
Yeah that's why i wrote MSBytefor for the Most significant byte in the uint32 (0xD0000000).

nepeee, 2017-09-22:
So if i want to send 1-7 bits i need to put it in the Most significant byte of the uint32. It's not so logical...

me-no-dev, 2017-09-22:
It makes sense once you start understanding that ESP32 is 32 bits and so is it's SPI bus. AVR for example has 8bit SPI

nepeee, 2017-09-22:
Ah ok, than just the mask is what causing the problem in the tansferBits function.

nepeee, 2017-09-22:
also the hardware chip select is toggled between the transferxy function calls :(

example code:
  SPI.setHwCs(true);

  SPI.beginTransaction(spiSettings);
  while(1)
    SPI.write32(0x80229018); //after every write cs goes high
  SPI.endTransaction();

I'm try to communicate with a ht1621 based lcd and for the segment data write i need to send 3 bit command and 6 bit address then the segment bits. So now i need to realign all the bit shifting stuff(for one f bit..) in my ported code (it has 96 segments) :S

me-no-dev, 2017-09-25:
Yes that is the expected behavior. What is troubling with CS being toggled? beginTransaction and endTransaction are software routines, so if you want CS toggled the whole transaction, you need to do it in software as well.

nepeee, 2017-09-25:
The protocol of this lcd requires to send a 3 bit command + 6 bit address(9 bit) and after that n*4bit data for the pixel ram. If the CS set to high between the two part the internal state is resetted so the lcd requires the command and the address again. I can do the CS by software but how can i know when the spi hardware finished with the data transmission ?

What i want to do:
  SPI.beginTransaction(spiSettings);
  SPI.transferBits(0xA480, NULL, 9);
  SPI.writeBytes(lcdArray, 12);
  SPI.endTransaction();

me-no-dev, 2017-09-26:
you do after begin transaction and before end transaction. that way you lock the bus, pull cs, execute, release cs and unlock the bus

nepeee, 2017-09-27:
I solved the problem by toggling the CS pin before the transaction and kept the device always selected. So the only problem in the spi code for me is the mask part in the esp32-hal-spi.c spiTransferBitsNL that i fixed by removing the hole mask part of the code.

closed 2019-08-15 by stale[bot]