espressif/arduino-esp32 #660: What release of ESP-IDF does the any current version of Arduino-ESP32 use?

nkolban, 2017-09-23:
It appears that the Arduino-ESP32 package supplies a copy of the ESP-IDF for building ESP32 Arduino applications.  My question is:

If I follow the published instructions to install the Arduino-ESP32 library in an Arduino IDE what version of the ESP-IDF will be found?

Here is the background.  As of the date of this post, the current release of ESP-IDF is v2.1.  The moment a release is cut, future development marches on within Github master and what ever code changes are made in there will be released in a future release.

Does the Arduino-ESP32 project/library leverage ESP-IDF releases or does it leverage "intermediate releases" such as those pulled from Git master post an ESP-IDF release but prior to the next formal release?

To add more color to this story, it appears that ESP-IDF master has made breaking changes to the APIs provided for BLE support.  This means that applications that were written to the BLE APIs in 2.1 will no longer compile or run in the 3.0 release (in the future).  What I am finding is that the current Arduino-ESP32 package seems to have pulled in the post 2.1, pre-3.0 ESP-IDF which would seem to imply that the Arduino-ESP32 package is leveraging "best so far" ESP-IDF.

I'm not taking a position on this concept as this is the first time I have run into a puzzle in this area ... however what I am asking is clarification on the facts of what ESP-IDF is actually used by the Arduino-ESP32 package at any given time.

See also:
* https://esp32.com/viewtopic.php?f=13&t=3116
* https://github.com/nkolban/esp32-snippets/issues/82

me-no-dev, 2017-09-23:
I am about to implement packager for Arduino which will allow such versioning. Currently it tracks the latest IDF. I understand that it broke things for you but at the same time I would get gazillion of other complains (why this does not compile under IDF, that is fixed in IDF but still broken in Arduino and so on) overall there will always be someone unhappy. Versioning and package manager should  help in most conditions, though surely there will still be someone unhappy here and there ;)

me-no-dev, 2017-09-23:
@nkolban I add in the update commit to Arduino to which commit in IDF it coresponds

nkolban, 2017-09-23:
Thank you sir.  It was clarification of the release levels that I was looking for and the notion that Arduino-ESP32 uses (today) the latest ESP-IDF from Git is fine by me.  99% of the time this will fix bugs ... its that small 1% that requires user code changes that will cause visibility.

I'm also assuming that the breaking code changes in the ESP-IDF are never done "arbitrarily" but rather because a better solution has presented itself.  I'm still digging into what has changed in the API that broke the existing code but on first sniff, the new APIs appear to be much better and cleaner than the old APIs.  I liken it to the transition from cassette tape to CD for music.  There may be grumbles when a new format becomes available but once we are "over the hill" everyone is happy.  New users never saw cassette tape and all is good, old users have to take the hit and move to the new standard .. which may cost them time and money ... but they can't ever argue that the new standard is "worse" than the old.

closed 2017-09-23 by nkolban

eldadwasserman, 2017-10-26:
Hi 
what you suggest me to use for a high level commercial development ? 
Arduino IDE or ESP-IDF?

Thanks

me-no-dev, 2017-10-30:
Arduino as IDF component :) best of both worlds ;)

etcetc, 2018-08-02:
Hi, on this topic I see that the package history indicates what version of IDF was used but I'm having a hard time figuring out which commit of this package I should use for the "released" 3.0 version of the IDF, which seems to have been created Nov 6 and continues to evolve.  There aren't that many commits so I can keep rolling back until I get a clean compile, but wondering if there is a "correct" way of determining this.