espressif/arduino-esp32 #663: WiFi Promiscuous filter breaks w/ WIFI_PROMIS_FILTER_MASK_ALL

SoftwareSamurai, 2017-09-24:
After calling esp_wifi_set_promiscuous_filter() with the filter set to WIFI_PROMIS_FILTER_MASK_ALL and starting up the WiFi, I randomly get this output:

"E (51478) wifi: wdev.c wdevProcessRxSucDataAll 1008"

When I get that, the Promiscuous callback appears to never get called. But it doesn't happen all the time. In fact, I can reset the ESP32 via the button (not even re-uploading my program) and sometimes it happens and sometimes it runs fine. 

--Details--
Board: ESP32 Dev Module, Rev 0.
IDE: PlatformIO (Also tried this on Arduino IDE - same results)
Core: Latest arduino-esp32 branch.

me-no-dev, 2017-09-25:
Can I please ask you to submit this in https://github.com/espressif/esp-idf ? those functions are out of the scope of Arduino. :)

SoftwareSamurai, 2017-09-25:
Done. I'll close this one then. Thanks!

closed 2017-09-25 by SoftwareSamurai