espressif/arduino-esp32 #664: Doesn't receive the result of HCI Command

wakwak-koba, 2017-09-25:
I'm going to operate the legacy bluetooth on arduino-esp32.
On probation I'll send the HCI RESET Command to VHCI, but response doesn't come.
`
#include "bt.h"
#include "Arduino.h"

static bool btSetup()
{
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){
        esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
        esp_bt_controller_init(&cfg);
        while(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE);
    }
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_INITED){
        if (esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT)) {
            Serial.println("BT Enable failed");
            return false;
        }
    }
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED){
        return true;
    }
    Serial.println("BT Setup failed");
}

static void _on_tx_ready(void)
{
    Serial.println("_on_tx_ready");
}

static int _on_rx_data(uint8_t *data, uint16_t len)
{
    Serial.print("_on_rx_data:");
    for(uint16_t i=0; i<len; i++)
        Serial.printf("%02x ", data[i]);
    Serial.println("");
}

static esp_vhci_host_callback_t vhci_host_cb = {
    _on_tx_ready,
    _on_rx_data
  };
  
void setup()
{
    Serial.begin(115200);
    Serial.print("ESP32 SDK: ");
    Serial.println(ESP.getSdkVersion());

    if(btSetup())
    {
        esp_vhci_host_register_callback(&vhci_host_cb);
        while (!esp_vhci_host_check_send_available());
        Serial.println("Started");
        delay(1000);

        uint8_t buf[4];

        // HCI RESET
        buf[0] = 0x01;
        buf[1] = 0x03;
        buf[2] = 0x0c;
        buf[3] = 0x00;
        esp_vhci_host_send_packet(buf, 4);
        esp_vhci_host_register_callback(&vhci_host_cb);
        Serial.println("esp_vhci_host_send_packet");
    }
}

void loop()
{

}
`

Doesn't fire __on_rx_data(uint8_t *data, uint16_t len)_
Why?

me-no-dev, 2017-09-25:
Please submit this over at https://github.com/espressif/esp-idf :)

wakwak-koba, 2017-09-26:
My problem has been solved.
When using bluetooth operations, It seems I have to use Core 0, not Core 1

lonerzzz, 2017-09-28:
Closing as the author indicated resolved.

closed 2017-09-28 by lonerzzz