espressif/arduino-esp32 #665: Hardware serial 2 does not work

emilekm2142, 2017-09-25:
```
HardwareSerial bt(2);
bt.begin(9600,SERIAL_8N1,35,34);
....
void loop(){
if (bt.available() > 0) {
		char r = bt.read();
		Serial.print(r);
		bt.println("got");
	}
}
```
Works in about 40% of the times. Even if it does, I can only retreive data and sometimes it halts on bt.available() when I write something to the serial. Printing never worked. The module works perfectly on uno with software serial. That is a issue also with the default `Serial` object, but in this case I can only print.

jestebanes, 2017-09-25:
Have you tried with native UART2 pins 16 RXD & 17 TXD, the highest pins only can be used as inputs.

Best regards

driverblock, 2017-09-25:
I was just about to post an issue on this myself.  I've been working with the Adafruit Feather ESP32 and an Adafruit GPS Featherwing.  With the Adafruit GPS_HardwareSerial_Parsing example sketch, I find that the combination of Serial at 115200 and UART 2 at 9600 does not work. the available function hangs, the read function hangs, and the delay function hangs.  Reducing the Serial baud rate to 57600 or lower allows the UART 2 functions to work, but it typically takes several processor resets before the sketch will come up working.  Otherwise, it continues to hang.  Lowering the Serial baud rate further below 57600 has no effect on the problem.

https://github.com/adafruit/Adafruit_GPS

I can also demonstrate the problem with a simple loopback jumper between the uart 2 pins. I will post demo code this afternoon.

driverblock, 2017-09-25:
@jestebanes:  I have been using pins 16 and 17.  The problem occurs with these pins.  I haven't tested UART1, but I suspect the problem also occurs there.

Regarding the loopback jumper, there is no problem if the jumper is not connected. That is, uart 2 does not exhibit any problems if there is no input connected.

kurtow, 2017-09-25:
I have the some problem and I use pins 16 RXD & 17 TXD

Best regards

emilekm2142, 2017-09-25:
Good to hear that it is not my fault at least.

jestebanes, 2017-09-25:
Another thing, have you tried not to use
Serial.begin(115200);
To avoid conflicts I'm only using
HardwareSerial Serial2(2);
UART0 is initialized by default to 115200 8N1 in sdkconfig.h

Best regards

emilekm2142, 2017-09-25:
Nothing changed.

jestebanes, 2017-09-25:
Pull up resistor in RXD pin?

kurtow, 2017-09-25:
same Problem with Pull up resistor 10k

emilekm2142, 2017-09-25:
In my case Serial 0 works partially too, so I'm sure it is a software problem.

jestebanes, 2017-09-25:
I am running out of options
Maybe is a concurrent problem if you are running your process so fast, OS can't read in the buffer, try a delay

void loop(){
       if (bt.available() > 0) {
		char r = bt.read();
		Serial.print(r);
		bt.println("got");
	}
        delay(1);
}

emilekm2142, 2017-09-25:
nope :/

jestebanes, 2017-09-25:
I give up, I am going to wait to the experts

UART2 works for me most of the time with the current release, but I have some performance issues related to the HardwareSerial driver that I have posted in the past, waiting for a solution.

Best regards

copercini, 2017-09-25:
As far I know, @me-no-dev will replace the current Serial driver by the IDF one (next week maybe), so it intends to fix a lot of bugs like this one.... 

https://github.com/espressif/arduino-esp32/issues/645#issuecomment-331768771

driverblock, 2017-09-25:
Some more observations on the Adafruit Feather. The attached code will work _sometimes_, using 115200 for Serial and 9600 for uart 2.  Different baudrates on Serial make no difference.

Run the code and be ready to keep hitting reset until it works.  Once it's working, it will most likely fail after the next reset.
`HardwareSerial Serial1(2);

void setup() {
  while(!Serial);
  while(!Serial1);
  Serial.begin(115200);
  Serial1.begin(9600);
  Serial.println("Uart 2 test");
}

void loop() {
  char tc, rc;

  //Loop to transfer 'A' through 'Z'
  for (tc ='A'; tc <= 'Z'; tc++)
  {
    Serial1.write(tc);
    
    while (!Serial1.available());
    
    rc = Serial1.read();
    
    if (rc == tc) {
      Serial.print(rc);
    }
    else {
      Serial.println();
      Serial.print("Error: sent '"); Serial.print(tc); Serial.print("', got '"); Serial.print(rc); Serial.println("'");

      while (Serial1.available())  //try to re-synchronize
        Serial.print(Serial1.read());
    }
  }
  Serial.println();
}`

emilekm2142, 2017-09-25:
I hope the driver change will fix that. After all, serial connection is a core functionality

me-no-dev, 2017-09-25:
i kinda really want to know what is causing this. serial is nothing spectacular and nothing major changed in idf, or at least i patched what i saw was changed. the fact that sometimes works is way more disturbing than if it did not work at all... @igrr @Spritetm @projectgus you have any thoughts on this? Peripherals are enabled in dport

jan-skold, 2017-09-26:
I have the same problem as described and even the same issue with Feather MO and Feather nRF52 Bluefruit LE. Anybody that has found a feather that works with GPS feather wing on Serial2?

driverblock, 2017-09-26:
The Feather nRF52 will work with the GPS, but you won't be able to program it with the GPS attached. The Feather M0 works fine.  Make sure you're using Serial1, not Serial2.

ladyada, 2017-09-26:
@me-no-dev just to add, i verified and even this basic sketch crashes

```
int led = A1;

HardwareSerial Serial1(2);

// the setup routine runs once when you press reset:
void setup() {
  Serial.begin(115200);              
  // initialize the digital pin as an output.
  pinMode(led, OUTPUT);     

  printf("serial1:\n");
  Serial1.begin(9600);              
  printf("\nserial1 done\n");
}

// the loop routine runs over and over again forever:
void loop() {
  Serial.println("blinx");
  Serial1.println("whee");
  Serial.write(Serial1.read());
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(500);               // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(500);               // wait for a second
}
```

icatstudio, 2017-09-27:
I have the very similar issue going on with a ESP32 DOIT board. I want to share things I encounter with the hope that it helps to those who are involved with the development process. 

I do coding both in Arduino IDE and Sloeber Eclipse in parallel. My Arduino IDE ESP32 development environment was setup quite some time ago (say 4-5 months) and not updated since. My Sloeber environment on the contrary, is just a few days old, setup with the most recent ESP32-IDF (with 21 Sep Update). 

Regarding the Serial.available() function, the code uploaded to the board via Arduino IDE functions without any problems. BUT the very same code uploaded stops responding IF I send something to the serial. 

As a side note I would mention that I run two communication processes as two separate tasks. A TCP communication the main loop and a Serial communication in a task. 

After initial boot I can send and receive on the TCP com BUT once I try to send something to the Serial every thing halts.

I guess what ever the problem is it is a global issue affecting both cores on the ESP32. 

These are my experiences regarding this issue. I hope it helps and I am looking for a firm diagnostic and solution we all benefit. 

Thanx for the effort guys

me-no-dev, 2017-09-27:
Please pull the latest changes and report :) Thanks!

kurtow, 2017-09-27:
@me-no-dev 
Seems to be okay now!
I restarted about 200 Times from DeepSleep and pressed 'reset' knob about 40 times
No more problems with Serial2. 

Thank you very much for solving the problem !!

me-no-dev, 2017-09-27:
YEAH! :D I'm looking forward closing this :)

emilekm2142, 2017-09-27:
Now Arduino JSON library stopped working :/
```
Decoded trace:
0x400da77b: __cxa_guard_release at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/guard.cc line 400
0x400d1155: ArduinoJson::JsonObject::invalid() at C:\Users\Emile\AppData\Local\Temp\VMBuilds\esp32_final\espressif_esp32\Release/esp32_final.cpp line 1160
0x400d2214: ArduinoJson::Internals::JsonParser  ::Reader, ArduinoJson::Internals::StringWriter  >::parseObject() at C:\Users\Emile\AppData\Local\Temp\VMBuilds\esp32_final\espressif_esp32\Release/esp32_final.cpp line 1160
0x400d2237: ArduinoJson::JsonObject& ArduinoJson::JsonBufferBase   >::parseObject (char*, unsigned char) at C:\Users\Emile\AppData\Local\Temp\VMBuilds\esp32_final\espressif_esp32\Release/esp32_final.cpp line 1160
0x400d3f22: _GLOBAL__sub_I_bt at C:\Users\Emile\AppData\Local\Temp\VMBuilds\esp32_final\espressif_esp32\Release/esp32_final.cpp line 220
:  (inlined by) _GLOBAL__sub_I_bt at C:\Users\Emile\AppData\Local\Temp\VMBuilds\esp32_final\espressif_esp32\Release/esp32_final.cpp line 1222
0x400e337f: do_global_ctors at /Users/ficeto/Espressif/ESP32/esp-idf/components/esp32/./cpu_start.c line 376 (discriminator 3)
0x40081239: start_cpu0_default at /Users/ficeto/Espressif/ESP32/esp-idf/components/esp32/./cpu_start.c line 313
0x40081409: call_start_cpu0 at /Users/ficeto/Espressif/ESP32/esp-idf/components/esp32/./cpu_start.c line 207
```
line 1160: `replaceProperty(actualRootTemplateRootObject, "text", "text", data["text"].as<const char*>());`
220: `DynamicJsonBuffer topBarBuffer(JSON_OBJECT_SIZE(2) + 14200);`
1222 is just an enter before the closing `}` of `void loop()`

me-no-dev, 2017-09-27:
line numbers are not correct in stack traces. And this is a whole other issue ;) Maybe show more of your code ;) and I suggest to open another issue, as this is not at all connected to Serial.

ladyada, 2017-09-27:
yep it is fixed!

emilekm2142, 2017-09-27:
I fixed that bug with JSON too. Serial works <3

closed 2017-09-28 by me-no-dev

icatstudio, 2017-09-28:
Great job :) huge relief :) thanks for the effort

CarlP66, 2019-02-27:
I know that this is an old thread, but I am currently having this exact problem with my Mega 2560. Serial0 works fine 100% of the time for communication with a Raspberry Pi using CmdMessenger / PyCmdMessenger  at 9600 baud, but when I also communicate with a TFMini Lidar unit on Serial1 at 115200, the Serial0 connection to the Pi intermittently gets screwed up - it works for a few transfers, then crashes the Python code on the Raspberry Pi. 

I believe my IDE and environment is up to date, but can someone be specific about what was updated back in Sept 2017 to fix this. 

I have confirmed that my Serial0 and Python code work fine when I comment out the Serial1 reads, and just set t2 & t4 to integer values (ie t1 = 5;  t4 = 10;)

The entire code is quite long, but the declarations and  offending function is below:

================ Declarations ================

#include <NewPing.h>
#include <Servo.h>
#include "CmdMessenger.h"

#define CENTER_TRIGGER_PIN  10
#define CENTER_ECHO_PIN     11
#define LEFT_TRIGGER_PIN  8
#define LEFT_ECHO_PIN     9
#define RIGHT_TRIGGER_PIN  12
#define RIGHT_ECHO_PIN     13
#define MAX_DISTANCE 200

NewPing centerSonar(CENTER_TRIGGER_PIN, CENTER_ECHO_PIN, MAX_DISTANCE);
NewPing leftSonar(LEFT_TRIGGER_PIN, LEFT_ECHO_PIN, MAX_DISTANCE);
NewPing rightSonar(RIGHT_TRIGGER_PIN, RIGHT_ECHO_PIN, MAX_DISTANCE);
Servo mytiltservo;  // create tilt servo object to control a servo
Servo mypanservo;  // create pan servo object to control a servo

const int leftForward = 2;
const int leftBackward = 4;
const int leftEnable =5;
const int rightForward = 3;
const int rightBackward = 7;
const int rightEnable = 6;

int tiltpos = 90;    // variable to store the servo position
int panpos =90;
int centerDistance = 100;
int leftDistance = 100;
int rightDistance = 100;

int t1; // byte 3 from lidar
int t2; // byte 4 from lidar
int t3; // byte 5 from lidar
int t4; // byte 6 from lidar

enum {
    spin,
    anticlockwise,
    anticlockwise_until,
    clockwise,
    clockwise_until,
    backup,
    forward,
    forward_until,
    ping_all,
    pan_tilt_test,
    lidar_read,
    pings,
    lidar_val,
    error,
};

/* Initialize CmdMessenger -- this should match PyCmdMessenger instance */
const int BAUD_RATE = 9600;
const long LIDAR_BAUD_RATE = 115200;
CmdMessenger c = CmdMessenger(Serial,',',';','/');

void setup(){
  pinMode(leftForward , OUTPUT);
  pinMode(leftBackward , OUTPUT);
  pinMode(rightForward , OUTPUT);
  pinMode(rightBackward , OUTPUT);
  Serial.begin(BAUD_RATE);
  Serial1.begin(LIDAR_BAUD_RATE);

  attach_callbacks();  
}

void attach_callbacks(void) { 
    c.attach(spin,on_spin);
    c.attach(anticlockwise,on_anticlockwise);    
    c.attach(anticlockwise_until,on_anticlockwise_until_free);      
    c.attach(clockwise,on_clockwise);          
    c.attach(clockwise_until,on_clockwise_until_free);      
    c.attach(backup,on_backup);    
    c.attach(forward,on_forward);
    c.attach(forward_until,on_forward_until_object);    
    c.attach(ping_all,on_ping_all);
    c.attach(pan_tilt_test,on_pan_tilt_test);  
    c.attach(lidar_read,on_lidar_read);           
    c.attach(on_unknown_command);
}

=====================  Offending Function =========================

void on_lidar_read()
{
// flush lidar buffer
  while(Serial1.available() > 0) 
  {
    char t = Serial1.read();
  }  

// wait for 9 new bytes - ie one full message
  while(Serial1.available()<9)
  {
      delay(5);
  }

// read message & return to Raspberry Pi
  if((0x59 == Serial1.read()) && (0x59 == Serial1.read())) //Byte 1 & Byte 2
  {
        t1 = Serial1.read(); //Byte 3
        t2 = Serial1.read(); //Byte 4
        t3 = Serial1.read(); //Byte 5
        t4 = Serial1.read(); //Byte 6
        for(int i=0; i<3; i++) 
        { 
            Serial1.read(); ////Byte 7,8,9
        }

        t2 <<= 8;
        t2 += t1;
        t4 <<= 8;
        t4 += t3;

        c.sendCmdStart(lidar_val);
        c.sendCmdBinArg(t2);
        c.sendCmdBinArg(t4);
        c.sendCmdEnd();
  }   
}

Thanks.

demym, 2021-03-12:
Sorry to resume a so old topic. I am using an Heltec WiFi Kit 32 (ESP32) in Arduino 1.8.3. 

My sketch takes input from Bluetooth Device and tries to send to Serial2 HardwareSerial (on pins 17 and 17). This is for sending MIDI upon receiving a BT flooarboard press. As soon as I do MySerial.write, the sketch crashed and ESP32 restarts.

I have tried several fixes read here and there, but not able to get the sketch not crashing.

I also tried to reduce Serial to 9600, or to remove it completely (for Serial.prints)...

Can someone give me an help ?

Thanks

chegewara, 2021-03-12:
Did you try change pins?
On that board pin IO16 is display reset, it may be some conflict.

ramanraja, 2025-01-27:
Two things helped:
1. Reducing the baud rate to 38400.
2. Explicitly mentioning the Tx and Rx pins during initialization: 
`SerialAT.begin(38400, SERIAL_8N1, 16, 17);` instead of just `SerialAT.begin(38400);`