espressif/arduino-esp32 #666: SPI bus inconsistency

reaper7, 2017-09-25:
I looked at the SPI lib source files
and I found some inconsistency:
https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.h#L54
set default spi bus to HSPI

but inside cpp file:
https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.cpp#L278
spi bus is set to VSPI

Is it correct?

nepeee, 2017-09-25:
In the first file, the HSPI is just a default value for the constructor so the VSPI is used by default.

me-no-dev, 2017-09-26:
And i will probably need to move both to HSPI because os PSRAM support, which was not in the picture when the driver was written :)

reaper7, 2017-09-26:
@nepeee thanks for the explanation ;)
@me-no-dev 👍

closed 2017-10-13 by copercini