espressif/arduino-esp32 #668: SDA/SCL connections not working with TM1637

jwktje, 2017-09-26:
### Hardware:
Board:							Wemos Lolin32 Lite
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I'm having no luck getting a TM1637 screen to work. It needs SDA and SCL pins which (if i'm right) are pins 21 and 22 by default. The Lolin32 Lite only exposes pin 22. I found Wemos commenting on a commit that pins are able to be remapped freely but I can find no info on this. (https://github.com/espressif/arduino-esp32/pull/35/files/46423c86972b999bdb9a81ea9fe6e1ba85851d5d#r86942155)

Software libraries to drive the screen aren't working for me either. I just want to show some numbers on a screen. I might just be an idiot but whatever I try, it's not working. 

Everything else works great!

### Sketch:
```
#include "SevenSegmentTM1637.h"

const byte PIN_CLK = 22;
const byte PIN_DIO = 4;
SevenSegmentTM1637 display(PIN_CLK, PIN_DIO);

void setup() {
  display.begin();
  display.setBacklight(7);
  display.print("INIT");
}
void loop() {
  
}
```

me-no-dev, 2017-09-26:
where is the source of SevenSegmentTM1637?

me-no-dev, 2017-09-26:
If it uses Wire, you can redefine with Wire.begin(SDA, SCL)

jwktje, 2017-09-26:
Sadly it doesn't use Wire. I've found a couple sources saying that Wire is supported for something called GPIO Matrix, so when using Wire, it should automatically work. 

The library I've tested the most with is this one:
https://github.com/bremme/arduino-tm1637

This library was advertised as being platform independent, so it sounded like the best bet. No luck with any library sadly. Screen stays off. 

I've tested the screen with an Arduino Uno to check if it's not a hardware issue. And it wasn't. The screen works there.

(I'm so sorry for sounding like a massive Noob. Senior webdev here, I'm getting into embedded electronics as a hobby)

MarkyAD, 2017-09-26:
Most of those displays aren't i2c, they just use a bit-banged protocol, so any 2 digital pins should work. Try using a different pin for clock.

jwktje, 2017-09-26:
After some more testing I can confirm you are right. This is just a bit-bang protocol.
But still no progress really. The sketch I pasted runs fine on the Arduino Uno, and exact same code on my ESP32 doesn't work. I tested multiple pins but just to be sure I'll test them all in a row right now.

What else could be the cause?

MarkyAD, 2017-09-26:
display.setBacklight(100); // set the brightness to 100 %

jwktje, 2017-09-26:
Just verified that this also works on Uno but not on ESP32:
`#include "SevenSegmentTM1637.h"

const byte PIN_CLK = 13;
const byte PIN_DIO = 12;
SevenSegmentTM1637    display(PIN_CLK, PIN_DIO);

void setup() {
  display.begin();
  display.print("INIT");
}
void loop() {

}`

MarkyAD, 2017-09-26:
Try using display.write() instead of display.print(). If no luck there, try feeding one of the ascii character bit patterns defined in SevenSegmentTM1637.h to display.printRaw() and see if you can at least get a character to display.

jwktje, 2017-09-26:
Just tried both and no change. Just for a full picture. This was the sketch I tested:
```
#include "SevenSegmentTM1637.h"

const byte PIN_CLK = 5;
const byte PIN_DIO = 18;
SevenSegmentTM1637    display(PIN_CLK, PIN_DIO);

byte rawData = B11100011;

void setup() {
  display.begin();
  display.printRaw(rawData,9);
}
void loop() {

}
```

And this is what the serial output shows after flashing or reset (Maybe that helps?):
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:572
load:0x40078000,len:0
load:0x40078000,len:9880
entry 0x400789d8
```

MarkyAD, 2017-09-26:
Try any of the other tm1637 libraries? There seems to be quite a few around. Although the driver ic is i2c, its simple enough that all of them use bit-banging, so I'm not sure why there would be a problem specific to esp. The only one of these I have currently is in a 328 project which I've already assembled.

jwktje, 2017-09-26:
I agree with you, I'd think it would be weird if this is a ESP32 specific issue but I can't exclude it at this stage. At this stage I tried around 7 different libraries. With 2 exceptions that didn't run on the UNO, the situation is the same. Uno runs it and the screen works, but ESP32 doesn't.

Could this be a clock speed issue? Maybe bitbanging could be off because of different cycle speeds?

MarkyAD, 2017-09-26:
Doubt clock speed is an issue since they all use e.g. delay(), delayMicroseconds(), etc which are clock speed independent. The only potential stumbling blocks for the bremme library I saw are that it inherits from the print class for display.print() which is why I suggested display.write() as a lower-level test, and that it uses progmem for storing the character bit patterns in flash. Might be worth it to try the example from the bremme library in examples/LowLevel/ and see if any luck there. Also try to stick with using pins that arent used for something else already (i2c, spi, etc).

jwktje, 2017-09-26:
I tried the LowLevel example and it works fine on Arduino, on any pin. But still no luck on the ESP32.
I get this feeling that I'm overlooking something staring me in the face. Here you can see the screen working with the LowLevel example. 

I honestly don't quite know what could be the issue here. 

[image]

MarkyAD, 2017-09-26:
Overlooking the obvious, perhaps I was as well, how about... the display ic is a 5volt part and wont work at 3.3v logic levels. The only datasheet I have on it is in chinese but it plainly shows it as a 5v part.

MarkyAD, 2017-09-26:
Overlooking the obvious, perhaps I was as well... how about, the display ic is a 5volt part and can be problematic (or not work at all) at 3.3volt logic levels.

me-no-dev, 2017-09-27:
if it just writes to this display and not read anything from it, you can probably powered it with 5V and still use 3.3V logic levels.

jwktje, 2017-09-27:
I tested the same sketch on an ESP8266 with 3.3v volt and it runs. But this board (Feather Huzzah) has a lower clockspeed than the Wemos too. 

I'm gonna test with a higher clock delay and see if that changes anything. This might just be a clockspeed issue after all.

jwktje, 2017-09-27:
Increased clock delay but no change. The LowLevel example showed me that the write to the display isn't being "Acknowledged" as it's called in the code.

Which basically means that programatically setting the data pin to INPUT_PULLUP fails I think. That's what I gather from the code. Is this pinmode something that fundamentally fails when using Arduino on the ESP32?

me-no-dev, 2017-09-28:
can’t be really that... though if you can put pullups on sda and scl that in effect will pull up the lines. can you try?

jwktje, 2017-09-28:
@me-no-dev I will try tonight and report back! Thanks for the tip

jwktje, 2017-09-29:
No difference sadly @me-no-dev

me-no-dev, 2017-09-30:
I looked more into the library code. It's just digital read and digital write. I can not possibly figure out why it would not work. It all looks so basic that it should have been fine on basically any Arduino platform...

lonerzzz, 2017-09-30:
Is this issue really about I2C or should we rename it? Only glancing at the text, it looks like a different two wire protocol is being used.

me-no-dev, 2017-09-30:
yup :) not I2C

jwktje, 2017-09-30:
@me-no-dev I understand man. I'm as stumped as anyone. I honestly think I've tested everything already. I also have an issue open with the library author. 
If it's not the pins I use, not the clockspeed, not the voltage, not defective hardware, not the library, not the sketch, then what could it be? I'll keep testing and update. If anyone has any tips they are greatly appreciated

everslick, 2017-10-21:
I can share some experience by myself. Bitbanging is MUCH less reliable on ESP32 then it is on ESP8266. I have software I2C and SPI working 100% stable on ESP8266, the same code on ESP32 produces bit errors (sometimes) for SPI, or refuses to work at all for I2C (yet). I think that timing stability is key here. Even though Arduino core is supposed to run on the APP_CPU exclusively, I guess we cannot expect uninterrupted execution. RAM and flash are still shared between cores, so one core will suspend the other from time to time.

What strikes me is, why protocols that rely on a dedicated CLOCK signal still give troubles, because any jitter should just not matter. But that is maybe dependent on the periphery that is connected.

jwktje, 2017-10-21:
Just an update on my specific situation. I switched to using the Lolin32 (so not the lite version) which has the hardware I2C pins exposed, and an I2C OLED screen.

Simply because I couldn't get the bitbang to work whatever I tried.

me-no-dev, 2017-10-21:
But there are no hardware i2c pins on esp32. They are all defined in the software and mapped in the signal matrix

me-no-dev, 2017-10-21:
@everslick i will look into that software i2c issue you mention. By all means it should have worked...

jwktje, 2017-10-21:
@me-no-dev Ah right, well the Lolin32 indicates 2 pins as SDA/SCL. That's why I thought those would be dedicated pins for that.

thijstriemstra, 2018-06-12:
@jwktje I'm using pins 15 and 2 for SDA/SCL on Lolin32 Lite without any issues:

[image]

froussel, 2018-06-16:
@thijstriemstra
I tear off the (rare) hair with I2C and lolin32 lite
Could I have a piece of code that works with the hookups please

thijstriemstra, 2018-06-16:
I'm using micropython and a different device but I can confirm I2C works on any free pins on a lolin32 lite.

stickbreaker, 2018-06-16:
@thijstriemstra don't use the main branch, use the [stickbreaker-i2c](https://github.com/espressif/arduino-esp32/tree/stickbreaker-i2c?files=1). The main branch i2c subsystem is unreliable.

Chuck.

froussel, 2018-06-16:
https://github.com/espressif/arduino-esp32/tree/stickbreaker-i2c?files=1   .... Made

arduino-esp32-stickbreaker-V0.2.0.zip    installed

[code]

/*
 * https://github.com/asukiaaa/MPU9250_asukiaaa/blob/master/examples/GetData/GetData.ino
 * 
 * 
 */

#include <MPU9250_asukiaaa.h>

#ifdef _ESP32_HAL_I2C_H_
#define SDA_PIN 15
#define SCL_PIN 2
#endif

MPU9250 mySensor;

uint8_t sensorId;
float aX, aY, aZ, aSqrt, gX, gY, gZ, mDirection, mX, mY, mZ;

void setup() {
  while(!Serial);
  Serial.begin(115200);
  Serial.println("started");

#ifdef _ESP32_HAL_I2C_H_ // For ESP32
  Wire.begin(SDA_PIN, SCL_PIN); // SDA, SCL
#else
  Wire.begin();
#endif

  mySensor.setWire(&Wire);
  mySensor.beginAccel();
  mySensor.beginGyro();
  mySensor.beginMag();

  // You can set your own offset for mag values
  // mySensor.magXOffset = -50;
  // mySensor.magYOffset = -55;
  // mySensor.magZOffset = -10;

  sensorId = mySensor.readId();
}

void loop() {
  Serial.println("sensorId: " + String(sensorId));

  mySensor.accelUpdate();
  aX = mySensor.accelX();
  aY = mySensor.accelY();
  aZ = mySensor.accelZ();
  aSqrt = mySensor.accelSqrt();
  Serial.println("accelX: " + String(aX));
  Serial.println("accelY: " + String(aY));
  Serial.println("accelZ: " + String(aZ));
  Serial.println("accelSqrt: " + String(aSqrt));

  mySensor.gyroUpdate();
  gX = mySensor.gyroX();
  gY = mySensor.gyroY();
  gZ = mySensor.gyroZ();
  Serial.println("gyroX: " + String(gX));
  Serial.println("gyroY: " + String(gY));
  Serial.println("gyroZ: " + String(gZ));

  mySensor.magUpdate();
  mX = mySensor.magX();
  mY = mySensor.magY();
  mZ = mySensor.magZ();
  mDirection = mySensor.magHorizDirection();
  Serial.println("magX: " + String(mX));
  Serial.println("maxY: " + String(mY));
  Serial.println("magZ: " + String(mZ));
  Serial.println("horizontal direction: " + String(mDirection));

  Serial.println("at " + String(millis()) + "ms");
  Serial.println(""); // Add an empty line
  delay(2000);
}

[/code]

All data of mpu9250 are 0

froussel, 2018-06-17:
Hello
I have 3K pullups.
With your script : Found 0 i2c Devices.

<<<Also, you can set the core debug level above None, ERROR will just show problems, DEBUG will show most non normal conditions.   
I don't understand  How ?

I replace the files existing before Stickbreaker-i2c
And : Found 0 i2c Devices.

I will test with a ESP8266 my MPU9250

copercini, 2018-07-01:
The I2C core was changed to @stickbreaker code officially after https://github.com/espressif/arduino-esp32/commit/13dcfe5c13bc882f8923d137ed604aa65b6db057 (thanks @stickbreaker), Let's close issues with old code and if have problems with the new code, open new issues =)

closed 2018-07-01 by copercini