espressif/arduino-esp32 #670: SPIFFS in Platformio

holema, 2017-09-27:
### Hardware:
Board:							NODE32S
Core Installation/update date:			27.09.2017
IDE name:						Arduino IDE

### Description:
On the ESP6288 in the Platformio IDE it was possible to create a SPIFFS file very easy and upload it to the ESP. I`m able to use the SPIFFS and the FS Library and to store values in a filesystem but I also need to create some files and send them to the ESP. Is there a way to get compile the data-folder?

me-no-dev, 2017-09-27:
it is a bit more complex task on ESP32 and I am thinking/working on it. The issue is that the partition scheme needs to be read and decoded in order for the proper size of the partition to be obtained and used for creating the SPIFFS image.

holema, 2017-09-27:
Thanks for your fast reply. Is there a Milestone for this feature?

me-no-dev, 2017-09-27:
nope, but I will look into it soon ;) I already have a branch that includes the needed mkspiffs in this repo. It's a matter of figuring out the proper way to get the config and add the code to the upload sketch tool.

roboticboyer, 2017-09-30:
@me-no-dev could be used [esp8266fs-plugin](https://github.com/esp8266/arduino-esp8266fs-plugin/releases/download/0.3.0/ESP8266FS-0.3.0.zip) also for ESP32?
This plugin creates the data folder/directory inside the SPIFFS memory

me-no-dev, 2017-09-30:
@roboticboyer please read my answer above :)

holema, 2017-10-16:
Hello me-no-dev,
I saw hat it is now possible to generate a SPIFFS file in the platformIO. But is this also uploaded to the device, or do I need a further step for this?

copercini, 2017-10-16:
@roboticboyer me-no-dev's released a version for ESP32 here: https://github.com/me-no-dev/arduino-esp32fs-plugin

me-no-dev, 2017-10-16:
@holema you should ask the platformio guys about that :) I am not familiar with their tools and procedures

holema, 2017-10-16:
@me-no-dev and @copercini  thanks a lot. 
@me-no-dev : does your asyncwebserver works with the SPIFFS System that is implemented up to now?

me-no-dev, 2017-10-16:
@holema yes

closed 2018-04-19 by everslick