espressif/arduino-esp32 #671: Error compiling

Teerawut101, 2017-09-27:
exec: "C:\\Arduino18\\hardware\\espressif\\esp32/tools/gen_esp32part.exe": file does not exist  
 
"gen_esp32part.exe" ro "gen_esp32part.py"

me-no-dev, 2017-09-27:
File is in the repo: https://github.com/espressif/arduino-esp32/blob/master/tools/gen_esp32part.exe

Maybe your clone failed?

euquiq, 2017-09-27:
Windows Defender detects a Trojan:Win32/Azden.A!cl  in  \Documents\Arduino\hardware\espressif\esp32\tools\esptool.exe ...

I did Allow this file and reinstalled the Github repository. Now everything is working fine. I only hope that esptool.exe is NOT actually infected !

copercini, 2017-09-27:
@me-no-dev Windows Defender come active by default in all windows installations, so I think it's a big problem, but looks like just last version of esptool and gen_esp32part are affected

me-no-dev, 2017-09-27:
Not the first time i see this ;) all exes i prepare on an isolated virtual machine

me-no-dev, 2017-09-27:
Though i used the one on my laptop since i am in shanghai. I usually do it on a carbon copy of it on my desktop. There is no other difference

copercini, 2017-09-27:
Probably it's not infected, but can be some new feature that is similar to a virus comportment, like download another .exe from internet, or whatever

The problem is Windows Defender is present in all windows computers, so all windows users will have this problem with updates

me-no-dev, 2017-09-27:
So what would be the safe course? Make them install python, pip and all necessary packages?

copercini, 2017-09-27:
Revert a version and investigate what changed that is triggering Windows defender?

copercini, 2017-09-27:
I'm testing these files: https://github.com/espressif/arduino-esp32/commit/d27d29719579cd765ef032b76e9a50b71bcf2a84

https://dl.espressif.com/dl/esptool-a420774-windows.zip (works fine)
https://dl.espressif.com/dl/esptool-4dab24e-windows.zip (deleted by windows defender in zip extraction)

https://github.com/espressif/arduino-esp32/blob/e2bd93c0920346cfbe3c792e82b0b3f99f4354a0/tools/gen_esp32part.exe (works fine, from one commit before https://github.com/espressif/arduino-esp32/commit/d27d29719579cd765ef032b76e9a50b71bcf2a84)
https://github.com/espressif/arduino-esp32/blob/master/tools/gen_esp32part.exe (blocked after download)

copercini, 2017-09-27:
@me-no-dev it's probably detected as virus due the process name in both files is `"<Input Sample>"` and lot of virus have the same name...

(yeah, I know, It's not me who program the antivirus software to have conclusions like this.... )

me-no-dev, 2017-09-28:
there must be something different between my virtual machines to cause this... i have no clue what though... maybe they are at different updates or something... i really have not used the laptop one in ages. It is certainly not a virus. if anything the one i used is much more cleaner than the other one

bill-orange, 2017-09-28:
How annoying.  As of this morning Windows Defender is detecting esptool as a virus and deleting it.  I kept putting it back and when I tried to upload, it disappeared again.  I will post to the windows 10 forum.  I can upload okay with Defender off.

bill-orange, 2017-09-28:
Result of submission to MS of esptool:

[image]

me-no-dev, 2017-09-29:
yeah... as I said, but this is still something I would like to get away from. I can recompile the EXEs once I get back home (in 10 days) but not before that. And would actually be able to compile them on  any of my machines... so what to do... crap

vseven, 2017-09-29:
Just a FYI - NodeMCU-PyFlasher-2.0.exe (https://github.com/marcelstoer/nodemcu-pyflasher/releases) is also getting flagged.  I used it a couple weeks ago to reflash my ESP32 and it was just sitting on my Windows 10 desktop.  I got a notice 4 or 5 days ago that Defender found and fixed some malware...I didn't notice it was gone until I tried to reflash the firmware a couple days ago.  Had to add a exception in Defender to get it to redownload.

I mention this because NodeMCU-PyFlasher-2.0.exe hasn't changed in over two months yet it just started getting flagged and removed 4 or 5 days ago.  Flagged for the same thing also.

bill-orange, 2017-09-29:
You probably should submit that file to Microsoft for re-analysis.  

https://www.microsoft.com/en-us/wdsi/filesubmission

dtbarn, 2017-09-30:
Thanks for this discussion.  I had been operating for several days without problems until this file disappeared.  To solve the problem on my Win 7 64-bit running MS Security Essentials (not Windows Defender), I had to unzip the file in a separate directory and copy the gen_esp32part.exe file to the tools directory.  Running get.exe or get.py must have gotten tangle with Windows and removed the file.

bill-orange, 2017-09-30:
Dan, I have submitted the offending files to MS for reclassification. I don’t know how long it takes them to revise the Definition file. We will see.

copercini, 2017-10-09:
Here it stopped detecting the file as virus =)

bill-orange, 2017-10-09:
Good news. Thanks.

copercini, 2017-10-12:
I think the files were re-evaluated by Microsoft and now the last version of Windows Defender doesn’t flag it anymore. Thanks @bill-orange for submitting to reclassification!

closed 2017-10-12 by copercini

dunnleaddress, 2018-01-17:
Go to:

Sketch -> include libraries -> manage library  and search all type all topic for "esptool" then click install

Done.

this package origin is https://github.com/jandrassy/UnoWiFiDevEdSerial1

astellingwerf, 2020-12-29:
SentinelOne still considers these files a thread. Is there a known workaround, or a proper fix maybe?