espressif/arduino-esp32 #672: Virus detected.

daytona16520, 2017-09-27:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			27/sep/2017
IDE name:							Arduino IDE 1.8.4
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Hi,
File gen_esp32part.exe gets detected as a Trojan:Win32/Azden.A!cl virus by Windows Defender and the IDE also reports it as virus when compiling a sketch.

So, I decided to use an older copy of the file from previous installation and that seemed to work but compiling now fails with the following message:
Arduino: 1.8.4 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

Archiving built core (caching) in: C:\Users\tom00\AppData\Local\Temp\arduino_cache_302453\core\core_espressif_esp32_esp32_FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_cc8d03a82408ab89806eebc3d4cf6de4.a
Multiple libraries were found for "WiFi.h"
 Used: E:\OneDrive\Arduino\My Projects\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exec: "E:\\OneDrive\\Arduino\\My Projects\\hardware\\espressif\\esp32/tools/esptool.exe": file does not exist
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

me-no-dev, 2017-09-27:
Surely not a virus;) make sure you allow it or you will not be able to compile

closed 2017-09-27 by me-no-dev

daytona16520, 2017-09-27:
Thanks for the quick response.

However, as I mentioned, even if I allow it in Windows Defender, the Arduino IDE still refuses to build the sketch with an error message complaining that esptool.exe does not exist.

Cheers.

me-no-dev, 2017-09-27:
Oh did you run get.py?

me-no-dev, 2017-09-27:
Sorry... get.exe :) check the readme

daytona16520, 2017-09-27:
Yes, I ran get.exe (Windows 10).  I will run it again.

me-no-dev, 2017-09-27:
Get.exe is the one that downloads the tools (esptool.exe and the toolchain) and could have also triggered your defender

daytona16520, 2017-09-27:
OK.  It's working now.
 When unzipping esptool-4dab24e-windows.zip Windows Defender falsely detected the gen_esp32part.exe as a virus and aborted the unzipping prematurely.

Temporarily disabling the Defender did the trick.

Thanks for your help.  Sorry to have caused some concern, but at least now if you get a similar complaint, you'll know what it's about.  :-)

Cheers,

daytona16520, 2017-09-27:
FYI, esptool.exe also triggers Defender.

me-no-dev, 2017-09-27:
All exes there are just compiled versions of the python files, so we do not make users install python and modules required. It’s totally normal if one triggers the defender then all to do it :)