espressif/arduino-esp32 #674: WiFiSmartConfig does not work anymore.

CelliesProjects, 2017-09-27:
### Hardware:
Board:							MH-ET LIVE ESP32MiniKit
Core Installation/update date:			v3.0-dev-745-gc4e65d6a updated on 22 Sept 2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
I noticed that in my project the WiFi smart config option did not work anymore.
The code I use is based on the example code below and written 2 months ago.
 
Then tried to compile and run the 'WifiSmartConfig.ino' example sketch.
But the 'WiFiSmartConfig' example sketch does also not work / connect anymore.

### Sketch:
```#include "WiFi.h"

void setup() {
  Serial.begin(115200);

  //Init WiFi as Station, start SmartConfig
  WiFi.mode(WIFI_AP_STA);
  WiFi.beginSmartConfig();

  //Wait for SmartConfig packet from mobile
  Serial.println("Waiting for SmartConfig.");
  while (!WiFi.smartConfigDone()) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("SmartConfig received.");

  //Wait for WiFi to connect to AP
  Serial.println("Waiting for WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("WiFi Connected.");

  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
Waiting for SmartConfig.
........................................................ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
Waiting for SmartConfig.
..................
SmartConfig received.
Waiting for WiFi
..[W][WiFiGeneric.cpp:220] _eventCallback(): Reason: 201 - AUTH_FAIL
.......................................................
```
The code will never connect, am I missing something here?

me-no-dev, 2017-09-28:
in igeneral it indicates that password is wrong, but unfortunately i have no way to test it right now :(

CelliesProjects, 2017-09-29:
Pretty sure I filled in the correct pw. Just wiped the pw from my phone to be sure.
My phone still connects with the pw but the esp does not.
Used the v0.3.4.3 version of the espressif app.

CelliesProjects, 2017-09-29:
Maybe anyone else can confirm this behaviour?

dmosberger, 2017-10-01:
Not sure if this is related, but WPS isn't working anymore either.  I just confirmed that with the September 24 snapshot of esp-idf, WPS works fine (latest commit 9a26296a0e88a4c3ae27e9c848be970946fff87e).  With today's version (latest commit 9a26296a0e88a4c3ae27e9c848be970946fff87e), WPS fails to connect.

CelliesProjects, 2017-10-04:
Just installed Arduino core snapshot 1407654c52dcf473b28df513a94afd294e3ca819 on my laptop and with that version the above sketch works again like expected.

There is something gone wrong in the WiFi code somewhere after that commit.

@me-no-dev Could you please take a look and preferably fix it?

me-no-dev, 2017-10-04:
I can not myself fix it, but our WiFi team is aware of the issue, hough China is in national holiday this week, so maybe next week this will be fixed.

CelliesProjects, 2017-10-04:
@me-no-dev Thanks for the feedback. 
Will keep an eye on the commits.

beegee-tokyo, 2017-10-06:
Can confirm that SmartConfig has a problem.
Using PlatformIO.
If I use `platform = espressif32` (Platform **_Espressif 32_** Version **_0.10.0_**) SmartConfig works as expected.
If I use `platform = espressif32_stage` (Platform **_Espressif 32 (Stage)_** Version **_aba1d7d_**) SmartConfig fails. The device seems to receive the SmartConfig data, but then fails to connect.

The error is as well visible on the Android app. If ESP32 staging version is used, the Android app shows _`Esptouch fail`_ else it shows _`Esptouch success, bssid = ...., InetAddress = 192.168....`_

Log from Android (in case SmartConfig succeeds:
```
10-06 18:05:22.423 22125-22572/tk.giesecke.esptouchforandroid I/EsptouchTask: Welcome Esptouch v0.3.4.6
10-06 18:05:22.423 22125-22572/tk.giesecke.esptouchforandroid D/UDPSocketServer: mServerSocket is created, socket read timeout: 60000, port: 18266
10-06 18:05:22.423 22125-22572/tk.giesecke.esptouchforandroid D/EsptouchTask: execute()
10-06 18:05:22.423 22125-22572/tk.giesecke.esptouchforandroid I/EsptouchTask: localInetAddress: /192.168.0.10
10-06 18:05:22.433 22125-22572/tk.giesecke.esptouchforandroid D/EsptouchTask: send gc code 
10-06 18:05:22.433 22125-21758/tk.giesecke.esptouchforandroid D/EsptouchTask: __listenAsyn() start
10-06 18:05:22.433 22125-21758/tk.giesecke.esptouchforandroid I/EsptouchTask: expectOneByte: 23
10-06 18:05:22.433 22125-21758/tk.giesecke.esptouchforandroid D/UDPSocketServer: receiveSpecLenBytes() entrance: len = 11
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid D/UDPSocketServer: received len : 11
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[0]:23
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[1]:36
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[2]:10
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[3]:-60
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[4]:-127
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[5]:-50
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[6]:-100
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[7]:-64
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[8]:-88
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[9]:0
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: recDatas[10]:105
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: receiveSpecLenBytes: $
                                                                                 āΜ(��i
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid I/EsptouchTask: receive correct broadcast
10-06 18:05:26.283 22125-21758/tk.giesecke.esptouchforandroid I/EsptouchTask: mSocketServer's new timeout is 56144 milliseconds
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid I/EsptouchTask: receive correct broadcast
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid I/System.out: 24
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid I/System.out: 0a
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid I/System.out: c4
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid I/System.out: 81
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid I/System.out: ce
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid I/System.out: 9c
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid D/EsptouchTask: __putEsptouchResult(): count = 1
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid D/EsptouchTask: __putEsptouchResult(): put one more result
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid I/UDPSocketClient: USPSocketClient is interrupt
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid I/UDPSocketServer: USPSocketServer is interrupt
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid E/UDPSocketServer: mServerSocket is closed
10-06 18:05:26.293 22125-21758/tk.giesecke.esptouchforandroid D/EsptouchTask: __listenAsyn() finish
```
Log from Android if SmartConfig fails:
```
10-06 18:10:25.633 29321-30171/tk.giesecke.esptouchforandroid I/EsptouchTask: Welcome Esptouch v0.3.4.6
10-06 18:10:25.633 29321-30171/tk.giesecke.esptouchforandroid D/UDPSocketServer: mServerSocket is created, socket read timeout: 60000, port: 18266
10-06 18:10:25.633 29321-30171/tk.giesecke.esptouchforandroid D/EsptouchTask: execute()
10-06 18:10:25.633 29321-30171/tk.giesecke.esptouchforandroid I/EsptouchTask: localInetAddress: /192.168.0.10
10-06 18:10:25.653 29321-30171/tk.giesecke.esptouchforandroid D/EsptouchTask: send gc code 
10-06 18:10:25.653 29321-30966/tk.giesecke.esptouchforandroid D/EsptouchTask: __listenAsyn() start
10-06 18:10:25.653 29321-30966/tk.giesecke.esptouchforandroid I/EsptouchTask: expectOneByte: 23
10-06 18:10:25.653 29321-30966/tk.giesecke.esptouchforandroid D/UDPSocketServer: receiveSpecLenBytes() entrance: len = 11
10-06 18:10:31.663 29321-30171/tk.giesecke.esptouchforandroid D/EsptouchTask: send gc code 
10-06 18:10:37.693 29321-30171/tk.giesecke.esptouchforandroid D/EsptouchTask: send gc code 
10-06 18:10:43.723 29321-30171/tk.giesecke.esptouchforandroid D/EsptouchTask: send gc code 
[... cut ...]
10-06 18:11:25.673 29321-30966/tk.giesecke.esptouchforandroid W/System.err:     at tk.giesecke.esptouchforandroid.task.__EsptouchTask$1.run(__EsptouchTask.java:175)
10-06 18:11:25.673 29321-30966/tk.giesecke.esptouchforandroid I/EsptouchTask: receive rubbish message, just ignore
10-06 18:11:25.683 29321-30966/tk.giesecke.esptouchforandroid D/EsptouchTask: __listenAsyn() finish
```
Looks like the ESP32 fails to respond over UDP after it received the SmartConfig data because the Android app fails with a UDP receive timeout `java.net.SocketTimeoutException`.

CelliesProjects, 2017-10-17:
Just updated ESP32 Arduino core to: v3.0-dev-899-g3a271a4a

Still not working.
Now the smartconfig reboots continually after the first try.
```
.....ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:956
load:0x40078000,len:0
load:0x40078000,len:11856
entry 0x40078a34
ESP32 SDK: v3.0-dev-899-g3a271a4a

Waiting for SmartConfig.
..........................................................................
SmartConfig received.
Waiting for WiFi
............................................ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:956
load:0x40078000,len:0
load:0x40078000,len:11856
entry 0x40078a34
ESP32 SDK: v3.0-dev-899-g3a271a4a

Waiting for SmartConfig.
.........Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4010ffd9  PS      : 0x00060730  A0      : 0x800fc0a9  A1      : 0x3ffd3ba0  
A2      : 0x00000000  A3      : 0x3ffd9e9c  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x0000000a  A7      : 0x3ffca300  A8      : 0x8010ffb1  A9      : 0x3ffd3b90  
A10     : 0x3ffd9e9c  A11     : 0x3ffc1c3a  A12     : 0x00000006  A13     : 0x3ffd9ea2  
A14     : 0x00001313  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4010ffd9:0x3ffd3ba0 0x400fc0a6:0x3ffd3be0 0x400fc2be:0x3ffd3c30 0x400fc2d3:0x3ffd3c70 0x400fdf57:0x3ffd3c90 0x400fe046:0x3ffd3cb0 0x4010a107:0x3ffd3cd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:956
load:0x40078000,len:0
load:0x40078000,len:11856
entry 0x40078a34
ESP32 SDK: v3.0-dev-899-g3a271a4a

Waiting for SmartConfig.
.........Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4010ffd9  PS      : 0x00060730  A0      : 0x800fc0a9  A1      : 0x3ffd3ba0  
A2      : 0x00000000  A3      : 0x3ffd9e9c  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x0000000a  A7      : 0x3ffca300  A8      : 0x8010ffb1  A9      : 0x3ffd3b90  
A10     : 0x3ffd9e9c  A11     : 0x3ffc1c3a  A12     : 0x00000006  A13     : 0x3ffd9ea2  
A14     : 0x00001313  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4010ffd9:0x3ffd3ba0 0x400fc0a6:0x3ffd3be0 0x400fc2be:0x3ffd3c30 0x400fc2d3:0x3ffd3c70 0x400fdf57:0x3ffd3c90 0x400fe046:0x3ffd3cb0 0x4010a107:0x3ffd3cd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:956
load:0x40078000,len:0
load:0x40078000,len:11856
entry 0x40078a34
ESP32 SDK: v3.0-dev-899-g3a271a4a

Waiting for SmartConfig.
.........Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4010ffd9  PS      : 0x00060730  A0      : 0x800fc0a9  A1      : 0x3ffd3c30  
A2      : 0x00000000  A3      : 0x3ffd9f2c  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x0000000a  A7      : 0x3ffca300  A8      : 0x8010ffb1  A9      : 0x3ffd3c20  
A10     : 0x3ffd9f2c  A11     : 0x3ffc1c3a  A12     : 0x00000006  A13     : 0x3ffd9f32  
A14     : 0x00001313  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4010ffd9:0x3ffd3c30 0x400fc0a6:0x3ffd3c70 0x400fc2be:0x3ffd3cc0 0x400fc2d3:0x3ffd3d00 0x400fdf57:0x3ffd3d20 0x400fe046:0x3ffd3d40 0x4010a107:0x3ffd3d60

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:956
load:0x40078000,len:0
load:0x40078000,len:11856
entry 0x40078a34
ESP32 SDK: v3.0-dev-899-g3a271a4a

Waiting for SmartConfig.
.........Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4010ffd9  PS      : 0x00060730  A0      : 0x800fc0a9  A1      : 0x3ffd3c40  
A2      : 0x00000000  A3      : 0x3ffd9f40  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x0000000a  A7      : 0x3ffca300  A8      : 0x8010ffb1  A9      : 0x3ffd3c30  
A10     : 0x3ffd9f40  A11     : 0x3ffc1c3a  A12     : 0x00000006  A13     : 0x3ffd9f46  
A14     : 0x00001313  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4010ffd9:0x3ffd3c40 0x400fc0a6:0x3ffd3c80 0x400fc2be:0x3ffd3cd0 0x400fc2d3:0x3ffd3d10 0x400fdf57:0x3ffd3d30 0x400fe046:0x3ffd3d50 0x4010a107:0x3ffd3d70

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:956
load:0x40078000,len:0
load:0x40078000,len:11856
entry 0x40078a34
ESP32 SDK: v3.0-dev-899-g3a271a4a

Waiting for SmartConfig.
...
```
@me-no-dev Any news from the WiFi team?

aronipani, 2017-11-02:
Hi,
I changed my wifi password on the router and after that I am not able to connect anymore with the new password using smartconfig. All devices connect except for smartconfig. I switched the password back and it worked again. Any idea what needs to be done? 
Another question, is there a way to change the esppressif hostname , again if I chnage the hostname using WiFi.setHostname("new-hostname") , the device does not connect and gets stuck at configuring smartconfig..
Thanks for your help.

allex1978, 2017-11-04:
SmartConfig received.
Waiting for WiFi
..[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:220] _eventCallback(): Reason: 201 - AUTH_FAIL
............................................................................

Hi, 
I fixed this. And sended the pull request. I'm a newbie on github.

me-no-dev, 2017-11-06:
your PR zeroes the bssid, which even if kept, should be the same.

allex1978, 2017-11-06:
Not the same. For test i comment my line  and added   log_d("bssid_set=%d",sta_conf->bssid_set); for debug.  And the result on my ESP32 is:

[D][WiFiSTA.cpp:621] _smartConfigCallback(): Status: FIND_CHANNEL
[D][WiFiSTA.cpp:621] _smartConfigCallback(): Status: GETTING_SSID_PSWD
[D][WiFiSTA.cpp:624] _smartConfigCallback(): Type: ESPTOUCH
[D][WiFiSTA.cpp:621] _smartConfigCallback(): Status: LINK
[D][WiFiSTA.cpp:627] _smartConfigCallback(): SSID: Vesper
[D][WiFiSTA.cpp:630] _smartConfigCallback(): bssid_set=231
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL

me-no-dev, 2017-11-06:
ok your log convinced me. Can you also dumb bssid and see if that is correct as data (no need to post it here)

allex1978, 2017-11-06:
Another AP (SSID=Elisabet) ... 2 different projects: 

        log_d("bssid_set: %d", sta_conf->bssid_set);
        log_d("BSSID: %x", sta_conf->bssid);

Project1 debug:
.[D][WiFiSTA.cpp:621] _smartConfigCallback(): Status: GETTING_SSID_PSWD
[D][WiFiSTA.cpp:624] _smartConfigCallback(): Type: ESPTOUCH
................[D][WiFiSTA.cpp:621] _smartConfigCallback(): Status: LINK
[D][WiFiSTA.cpp:627] _smartConfigCallback(): SSID: Elisabet
[D][WiFiSTA.cpp:629] _smartConfigCallback(): bssid_set: 182
[D][WiFiSTA.cpp:630] _smartConfigCallback(): BSSID: 3ffc6c0d

Project2 debug:
[D][WiFiSTA.cpp:621] _smartConfigCallback(): Status: GETTING_SSID_PSWD
[D][WiFiSTA.cpp:624] _smartConfigCallback(): Type: ESPTOUCH
[D][WiFiSTA.cpp:621] _smartConfigCallback(): Status: LINK
[D][WiFiSTA.cpp:627] _smartConfigCallback(): SSID: Elisabet
[D][WiFiSTA.cpp:629] _smartConfigCallback(): bssid_set: 182
[D][WiFiSTA.cpp:630] _smartConfigCallback(): BSSID: 3ffc8565

Not sure, but looks SmartConfig get only SSID and psk ...  bssid and bssid_set values is some "trash" from ram.  and with non zero bssid_set esp32 is trying to connect to wrong bssid.

CelliesProjects, 2017-11-08:
@allex1978 47cdfff4756876ac952832c1bcec3adecd982a49 seems to have solved the problem.
Thanks for your effort.
Closing this issue.

CelliesProjects, 2018-01-05:
@me-no-dev There is again a problem with SmartConfig.

The example sketch will not finish SC, but will show this output and wait forever:
````
load:0x40078000,len:13076
entry 0x40078ad0
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 14 - AP_STACONNECTED
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 14 - AP_STACONNECTED
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 14 - AP_STACONNECTED
Waiting for SmartConfig.
.........[D][WiFiSTA.cpp:624] _smartConfigCallback(): Status: FIND_CHANNEL
...............................................................................
````
Seems a different problem than the one above.

CelliesProjects, 2018-01-05:
Updated to 'ESP32 SDK: v3.0-dev-1325-ge5b2c1cf' and all is working again...

CelliesProjects, 2018-01-12:
@me-no-dev 
I spoke too soon. Seems like the problem with SC has re-appeared in a slightly different form.

It now only works (sometimes) if I restart WiFi on my phone and when it works it does it only one time.

For debug output when it does not work see the example output I posted before.
Is Espressif aware of the issue?

ctruter, 2018-02-01:
@CelliesProjects I also have the same issue. It also happens in my case without the Smart Config, as stated in [this](https://github.com/espressif/esp-idf/issues/1366) issue on the IDF forum.

I have tried 3 different ESP modules and multiple APs, with no success.

CelliesProjects, 2018-04-15:
Closing this issue as 69f72eca849a44688e2da38028eb9aa707e91f08 solved it.

closed 2018-04-15 by CelliesProjects