espressif/arduino-esp32 #676: ERROR : dport_access.c

birdtechstep, 2017-09-28:
assertion "res == pdTRUE" failed: file "/Users/ficeto/Espressif/ESP32/esp-idf/components/esp32/./dport_access.c", line 184, function: esp_dport_access_int_init
abort() was called at PC 0x400dddc7 on core 1
Rebooting...

lonerzzz, 2017-09-28:
@birdtechstep More information is needed in order to help you. Can you provide the core dump associated with this issue? Also, is there some specific code that you have which can be used to reproduce it?

birdtechstep, 2017-09-29:
### Hardware:
Board: ESP32 Devkit v.1
Core Installation/update date:	     lastest update
IDE name:					Arduino IDE 1.8.4
Flash Frequency:				80Mhz
Upload Speed:					921600

### Description:
Describe your problem here

//Change the code below by your sketch
#include <Arduino.h>
#include <SPI.h>

void setup() {
}

void loop() {
}

Other is OK.
but use SPI error on serial monitor.

assertion "res == pdTRUE" failed: file "/Users/ficeto/Espressif/ESP32/esp-idf/components/esp32/./dport_access.c", line 184, function: esp_dport_access_int_init
abort() was called at PC 0x400e58e3 on core 1

Backtrace: 0x40086a1c:0x3ffe7ce0 0x40086b1b:0x3ffe7d00 0x400e58e3:0x3ffe7d20 0x400ddaaa:0x3ffe7d50 0x400812a0:0x3ffe7d80 0x400812e1:0x3ffe7da0 0x40007c31:0x3ffe7dc0 0x4000073d:0x3ffe7e30

Rebooting...

lonerzzz, 2017-09-29:
You have to paste the Backtrace into the ESP Exception Decoder which you have to add to the Arduino IDE in order for the back trace to be meaningful to someone else.

The exception decoder is here:
https://github.com/me-no-dev/EspExceptionDecoder

You are going to need more code than what you have provided for anything to work in any case. The program has nothing to do and will complain so try doing the setup of the SPI, etc and see what you get.

carbocation, 2017-11-21:
I seem to get the same error when I request slightly "too much" RAM, and a different error when I request a little more (both being well below 192k). I have a [fairly large sketch](https://github.com/carbocation/soulmate/tree/master/jpp/first) for controlling LED strips. When I try to set N_LEDS (the number of LEDs, allocated as an array) to ~1800, I start to get this same error. The sketch is using 676559 bytes of program storage space and the global variables use 119488 bytes of dynamic memory. This is the backtrace I get:

`assertion "res == pdTRUE" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./dport_access.c", line 187, function: esp_dport_access_int_init
abort() was called at PC 0x400dd65b on core 0`

`Backtrace: 0x40087b98:0x3ffe3b60 0x40087c97:0x3ffe3b80 0x400dd65b:0x3ffe3ba0 0x400eaed2:0x3ffe3bd0 0x4008134b:0x3ffe3c00 0x4008152c:0x3ffe3c30 0x40078856:0x3ffe3c50 0x400788bd:0x3ffe3c80 0x400789f9:0x3ffe3cb0 0x40078b1f:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20`

Which decodes to:
`0x40087b98: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x40087c97: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553`

`0x400dd65b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)`

`0x400eaed2: esp_dport_access_int_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./dport_access.c line 187 (discriminator 1)`

`0x4008134b: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 346`

`0x4008152c: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 210`

Interestingly, if I set N_LEDS to a higher value (around 2400), with the program using 676503 bytes of program storage space and 144688 bytes of dynamic memory, I start to get the following:

`abort() was called at PC 0x40081455 on core 0`

`Backtrace: 0x40087b98:0x3ffe3bf0 0x40087c97:0x3ffe3c10 0x40081455:0x3ffe3c30 0x40078856:0x3ffe3c50 0x400788bd:0x3ffe3c80 0x400789f9:0x3ffe3cb0 0x40078b1f:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20`

This decodes to:
`0x40087b98: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x40087c97: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x40081455: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 143`

Up to 1522 N_LEDS, the program compiles and runs without issue, with the program using 676539 bytes of program space and 107816 bytes of dynamic memory.

closed 2018-04-19 by everslick