espressif/arduino-esp32 #677: Remote event logging for ESP32

edautz, 2017-09-28:
### Hardware:
Board:							 DOIT ESP32 DEVKIT V1
Core Installation/update date:                 15/sep/2017
IDE name:					         Arduino IDE 1.6.12
Flash Frequency:					 80Mhz
Upload Speed:				         OTA

### Description:
I run a ESP32 webserver in the main loop, and in seperate tasks SSD1306 oled test code, ntp and a blink sketch. But my setup isn't stable. After a few hour I got reboots. 

I looking for a way to log critical events to syslog and send them to a syslog server.
I want to use a library like this https://github.com/arcao/Syslog to implement a syslog client.

But I can't figure out how to setup the code to redirect critical events to syslog. 
Does someone accomplished that? And provide a sample code to get it working?

lonerzzz, 2017-09-28:
@edautz The easiest thing might be simply to setup a UART to a terminal on your PC. Then provide a huge buffer for the terminal. Subsequently, you can add printouts and any core dumps or startup logging will go to the UART. I put a UART on every project I make for this specific purpose.

edautz, 2017-09-29:
@Ionerzzz I agree, this would be the easist way, but I intent to put several ESP32 in production for IOT/homedomotica purposes. For that I want to build a central remote logging solution so I can monitor and act on events while those ESP are running.

me-no-dev, 2017-09-29:
One way is to store the error in RTC and after reboot and connect to send it to that syslog.

edautz, 2017-09-30:
@me-no-dev. Do you have example code how to do this?

closed 2017-10-27 by edautz

NareshDealer, 2018-04-15:
@edautz Did you found any working example.. i m on the same boat... Really appreciate if you can share your achievement here.

PotatoSauceVFX, 2019-07-31:
Yeah, I could really use an example sketch for my IOT project.

edautz, 2019-07-31:
What I finally did, is to make my own remote syslog based events. E.g. When the WiFi signal becomes to bad, I send a syslog trap to my Raspberry based syslog server, also after a reboot etc.