espressif/arduino-esp32 #679: On Fatfs time zone is ignored on file time

luc-github, 2017-09-29:
Hi when using fatfs function f_stat to get last access date/time I found time zone is ignored and only UTC time is given

I did a script to show issue on SD : https://gist.github.com/luc-github/9124cbb2b1e0ed0b3696b4e6223b8e6e

Be noted the time zone is currently upside down as reported here #36

luc-github, 2017-09-29:
After investigation it seems the function to get timestamp in libfatfs : `get_fattime`
use `gmtime` instead of `localtime_r` but this part is in IDF

me-no-dev, 2017-09-30:
you can submit an issue or pr there :) those are looked at and taken care of as well :)

luc-github, 2017-09-30:
Yes I am currently setting up the environnement to compile IDF to test first - so I will be able to do PR 😄

me-no-dev, 2017-09-30:
put arduino as component and set up under sloeber/eclipse :) you might like it

luc-github, 2017-09-30:
Ok will try it thank you- I get used to geany (http://www.geany.org/)  which is  a light cross platform  code editor

me-no-dev, 2017-09-30:
sure :D whatever floats your boat :P

luc-github, 2017-09-30:
I cannot for some unknown reason compile using arduino-esp32 as ESP-IDF component but was able to get modified libfatfs.a and replaced  arduino-esp32 one
and it works so fix is confirmed I will do PR on IDF later

luc-github, 2017-09-30:
PR done

luc-github, 2018-10-04:
wow seems this issue is still open - I close it sorry for the delay

closed 2018-10-04 by luc-github