espressif/arduino-esp32 #681: SPIFFS can open nonexisting file for reading?

codemasterPL, 2017-09-29:
Device: ESP32-WROOM32
IDE: Arduino

It seems like SPIFFS can open (read) a file, that is not existing..

`
#include "FS.h"
#include "SPIFFS.h"

void setup() {

 Serial.begin(9600);
 delay(1000);
 
 Serial.println("Mounting FS...");
  
  while (!SPIFFS.begin(true))
  {
    Serial.println("Failed to mount file system, formatted");
    delay(5000);
  }  

  SPIFFS.remove("/somefile.txt");
  delay(1000);
  
  File configFile = SPIFFS.open("/somefile.txt","r");
  
  if (!configFile) 
  {
    Serial.println("Failed to open config file");
  }
  else
  {
    Serial.println("File Opened");
  }

}

void loop() {
  // put your main code here, to run repeatedly:

}
`

Output:

> Mounting FS...
> File Opened
>

me-no-dev, 2017-09-29:
FS::open opens files and directories. It's no longer enough to check if open returned true. Also need to check that it is not a folder. SPIFFS itself returns true every time (no notion of folders and for some reason the developer decided that it should return true). So...
```cpp
if(configFile && !configFile.isDirectory()){
  //file exists
}
```

closed 2017-09-29 by me-no-dev

codemasterPL, 2017-09-29:
Hold on,

Looking at the example:

```
File file = fs.open(path);
    if(!file){
        Serial.println("Failed to open file for reading");
        return;
    }

```

SPIFFS should return false if a file cannot be open right (is not existing for reading) ?

So when i do:

`File configFile = SPIFFS.open("/somefile.txt","r");
`  

On a file that doesn't exist, what is returned?

me-no-dev, 2017-09-29:
example needs to be updated ;) this is a different FS than ESP8266 and as I mentioned SPIFFS itself always returns true. You can take my word on it or argue :D your choice.
SPIFFS itself does not understand folders, so files like /blah/smth.ext have actually filename /blah/smth.ext and not smth.ext inside folder /blah. FS at the same time returnes true if it had found folder or file with that name, but for that to work in SPIFFS, we have to loop through all files and see if one starts with the folder name (looping takes much time).
I hope I made myself clear

codemasterPL, 2017-09-29:
Thanks i don't want to argue ;)

I just want to understand, how do i actually check if a file exists :)
Thx, i think i got all to move fwd with the project now.

Thx again

me-no-dev, 2017-09-30:
great :)

fri2, 2019-07-18:
> 
> 
> As i found the source of my troubles here, i want to put in my .02.
> I firmly believe POSIX behavior for file systems should be the standard.
> 
> i found this post by ESPfri2: https://esp32.com/viewtopic.php?t=10451 describing the issue perfectly.
> I hope this will be corrected, but am not holding my breath :-)

me too !!!

louislewis2, 2019-10-29:
Terrible bug to have around, I dont think there is a question of whether it is or is not a bug..
Had me going for ages till I found this post. Opening a non existent file in read mode, should not return.

I ask that this be re-opened and looked at please.

ArsineMan, 2020-07-07:
Where was this post 4 hours ago? Too stubborn to google is a flaw.  Knocking my head against a wall on this. Finally, decided there was some sort of race condition or priority of execution.  I still don't know why this exists but its a pain.  My only recourse was to place a non-blocking delay between SPIFFS.exists() and SPIFFS.open(). Unfortunate solution. To bad there isn't at least a callback function when executing anything against SPIFFS. Good luck all.

anothermouse, 2020-11-17:
Another workaround that I use for this annoying problem:
`
  File myfile = SPIFFS.open("/index.html", "r");
  Serial.println(String(page.size()));
  if (myfile.size() > 0) {
  //do something
  }
`

Of course there's also the edge case where the file _might be_ zero bytes in size, but it solves some problem cases.